ros-navigation/navigation2 #3567: Navigation2 broken on Humble

ZealousGinger, 2023-05-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Jammy 22.04 LTS 
- ROS2 Version:
  - ROS2 Humble
- Version or commit hash:
  - ros-humble-navigation2	1.1.6-1jammy.20230426.153445
- DDS implementation:
  - Default
#### Steps to reproduce issue

Hello I'm quite new to ROS2 and Navigation2 so I'm not sure if this is the right repo to ask this question, any help is greatly appreciated thanks! I'm not sure if the problem lies with Navigation2 or ROS2 or something else. Last week around 2nd-5th may after running sudo apt update && upgrade nav2 seems to stop working. 
I am using linorobot2 a repository similar to ROS2 Turtlebot3, along with running the robot in simulation in gazebo.

#### Expected behavior
In the program in rviz2 , when a nav2 goal is set the robot normally navigates to the point. Then stops once it reaches the point.
#### Actual behavior
Now, the navigation path towards the goal (green line) is shown, the robot starts to move towards the goal, however, the rainbow outline map or costmap keeps on rotating as the robot moves, it disorientates the robot. This causes the robot and the costmap to keep jumping into random places. Here are some screenshots to show the problem:
[image]
[image]
Some of the errors and warnings shown in the terminal are:
```
[global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Lookup would require extrapolation into the past.  Requested time 8.981000 but the earliest data is at time 9.601000, when looking up transform from frame [base_link] to frame [map]
```
```
[component_container_isolated-1] [ERROR] [1683639367.543086814] [GoalDistCritic]: None of the points of the global plan were in the local costmap.
[component_container_isolated-1] [WARN] [1683639367.543157881] [DWBLocalPlanner]: A scoring function failed to prepare
[component_container_isolated-1] [ERROR] [1683639367.543198694] [PathDistCritic]: None of the 68 first of 68 (68) points of the global plan were in the local costmap and free
[component_container_isolated-1] [WARN] [1683639367.543206012] [DWBLocalPlanner]: A scoring function failed to prepare
[component_container_isolated-1] [ERROR] [1683639367.543231657] [PathDistCritic]: None of the 68 first of 68 (68) points of the global plan were in the local costmap and free
[component_container_isolated-1] [WARN] [1683639367.543234671] [DWBLocalPlanner]: A scoring function failed to prepare
[component_container_isolated-1] [ERROR] [1683639367.543263756] [GoalDistCritic]: None of the points of the global plan were in the local costmap.
[component_container_isolated-1] [WARN] [1683639367.543266804] [DWBLocalPlanner]: A scoring function failed to prepare
[component_container_isolated-1] [ERROR] [1683639367.545871492] [DWBLocalPlanner]: No valid trajectories out of 419! 
[component_container_isolated-1] [ERROR] [1683639367.545899660] [DWBLocalPlanner]: 1.00:   PathDist/Trajectory Hits Unreachable Area.
[component_container_isolated-1] [WARN] [1683639367.546132820] [controller_server]: No valid trajectories out of 419! 
[component_container_isolated-1] [WARN] [1683639385.612778664] [local_costmap.local_costmap]: Sensor origin at (0.14, 0.00 0.12) is out of map bounds(3.12, 2.98, 0.90) to (-0.30, -3.45, 0.00). The costmap cannot raytrace for it.
```
This is the logs:
[problem.log](https://github.com/ros-planning/navigation2/files/11431769/problem.log)
Once again, any help is appreciated, thanks! 
#### Additional information

ZealousGinger, 2023-05-10:
Thanks for the reply, I have tried changing the DDS to Cyclone and the issue is still there, will try out the turtlebot3 demos soon thanks!

SteveMacenski, 2023-05-10:
Let me know how it goes!