ros-navigation/navigation2 #3568: Typo in nav2_params.yaml

MikeCoss, 2023-05-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.2 Jammy
- ROS2 Version:
  - Humble Debian packages
- Version or commit hash:
$ dpkg-query --show "ros-$ROS_DISTRO-navigation2"
ros-humble-navigation2  1.1.6-1jammy.20230221.193247
$ dpkg-query --show "ros-$ROS_DISTRO-nav2-*"
ros-humble-nav2-amcl    1.1.6-1jammy.20230221.155303
ros-humble-nav2-behavior-tree 1.1.6-1jammy.20230221.155321
ros-humble-nav2-behaviors     1.1.6-1jammy.20230221.163010
ros-humble-nav2-bringup 1.1.6-1jammy.20230221.194427
ros-humble-nav2-bt-navigator  1.1.6-1jammy.20230221.163022
ros-humble-nav2-collision-monitor   1.1.6-1jammy.20230221.161841
ros-humble-nav2-common  1.1.6-1jammy.20230221.141902
ros-humble-nav2-constrained-smoother      1.1.6-1jammy.20230221.162135
ros-humble-nav2-controller    1.1.6-1jammy.20230221.162139
ros-humble-nav2-core    1.1.6-1jammy.20230221.161908
ros-humble-nav2-costmap-2d    1.1.6-1jammy.20230221.160008
ros-humble-nav2-dwb-controller      1.1.6-1jammy.20230221.164541
ros-humble-nav2-lifecycle-manager   1.1.6-1jammy.20230221.155332
ros-humble-nav2-map-server    1.1.6-1jammy.20230221.155310
ros-humble-nav2-msgs    1.1.6-1jammy.20230221.145152
ros-humble-nav2-navfn-planner 1.1.6-1jammy.20230221.162224
ros-humble-nav2-planner 1.1.6-1jammy.20230221.162256
ros-humble-nav2-regulated-pure-pursuit-controller     1.1.6-1jammy.20230221.162324
ros-humble-nav2-rotation-shim-controller  1.1.6-1jammy.20230221.162925
ros-humble-nav2-rviz-plugins  1.1.6-1jammy.20230221.192742
ros-humble-nav2-simple-commander    1.1.6-1jammy.20230221.145801
ros-humble-nav2-smac-planner  1.1.6-1jammy.20230221.162453
ros-humble-nav2-smoother      1.1.6-1jammy.20230221.162519
ros-humble-nav2-theta-star-planner  1.1.6-1jammy.20230221.162606
ros-humble-nav2-util    1.1.6-1jammy.20230221.154439
ros-humble-nav2-velocity-smoother   1.1.6-1jammy.20230221.171226
ros-humble-nav2-voxel-grid    1.1.6-1jammy.20230221.145224
ros-humble-nav2-waypoint-follower   1.1.6-1jammy.20230221.162747

- DDS implementation:
  - Fast-RTPS
 
#### Steps to reproduce issue
There is a typo in nav2_param.yaml file.  Line of interest is https://github.com/ros-planning/navigation2/blob/49d0cdfe5b4025074b39a026493a7c41e3f3e49c/nav2_bringup/params/nav2_params.yaml#L311

Text should be global_frame: map, not global_flame: map

**Note: this is not on the humble branch but is an error introduced in the main branch**

#### Expected behavior
global_frame should be defined as a behavior parameter.

#### Actual behavior
global_flame is defined.

#### Additional information
The parameter is referenced in 2 places:
1. https://github.com/ros-planning/navigation2/blob/main/nav2_behaviors/src/behavior_server.cpp#L63
2. https://navigation.ros.org/configuration/packages/configuring-behavior-server.html#behavior-server

SteveMacenski, 2023-05-09:
Can you open a PR to fix?

SteveMacenski, 2023-05-09:
Merged

closed 2023-05-09 by SteveMacenski

SteveMacenski, 2023-05-09:
(was it me? that would be really embarrassing if it was me)

MikeCoss, 2023-05-09:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

jwallace42, so I presume is not you :)

---Michael J Coss

SteveMacenski, 2023-05-09:
oooooooooooooooooooooohhhhh Josh :wink: 

CC @jwallace42 

Its OK, I once made a regression that caused: 

[image]

I've never laughed so hard in my life at a bug

MikeCoss, 2023-05-09:
Got to love path planning gone wrong.  And I do like the frame to flame.  All good.

Again, thanks for the quick fix.

---Michael J Coss

[suboptimal]<https://user-images.githubusercontent.com/14944147/237176168-9d805cd6-42d3-4f9b-b9f8-244471672d0c.png>