ros-navigation/navigation2 #3573: Final orientation cannot be reached for Ackermann vehicles

HarunTeper, 2023-05-12:
Hello,

I am currently configuring the Nav2 stack for an ackermann vehicle and I am having issues with the planner and controller for the final orientation. Specifically, the planner always recreates a new circular path around the goal pose, while the controller is not able to reach the final goal pose orientation.

You can see the behavior in this video https://streamable.com/5r2f0a.

My nav2 configuration can be found here: https://github.com/HarunTeper/AuNa/blob/main/src/auna_nav2/config/nav2_params/nav2_params.yaml

I am using the nav2_regulated_pure_pursuit_controller::RegulatedPurePursuitControllerand the nav2_smac_planner/SmacPlannerHybrid for the navigation.

What should I change to prevent this behavior?

closed 2023-05-15 by HarunTeper

Rodrag, 2023-12-06:
Hi there, did you find a fix for this, working on the same issue atm

HarunTeper, 2023-12-07:
@Rodrag I changed the global planner to SmacPlannerHybrid, to make the car plan according to its kinematic constraints. You may also change the local planner to RegulatedPurePursuit, and add a converter node to change the output from unicycle to bicycle or Ackermann model as well.