ros-navigation/navigation2 #3574: RemovePassedGoals Plugin frame parameter inheritance

HarunTeper, 2023-05-15:
## Bug report

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Galactic
- Version or commit hash:
  - 2de3f92c0f476de4bda21d1fc5268657b499b258
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
git clone https://github.com/HarunTeper/AuNa
cd AuNa
git checkout 24c114b3aaf0cd6da65316245e08b98669ca948b

./auto_setup.sh
colcon build --symlink-install
source install/setup.bash
ros2 launch auna_scenarios scenario_single_robot_arena_navigation.launch.py

2nd Terminal:
source install/setup.bash
ros2 launch auna_mqtt mqtt_waypoint_receiver.launch.py

#### Expected behavior
The robot receives 4 waypoints to use the navigate_through_poses behavior tree to move

#### Actual behavior
The robot starts the recovery behavior and the following error occurs:
[bt_navigator-12] Warning: Invalid frame ID "base_link" passed to canTransform argument source_frame - frame does not exist
[bt_navigator-12]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.5/src/buffer_core.cpp

#### Additional information

The remove_passed_goals_action.cpp plugin does not natively use the robot_base_frame specified for the bt_navigator. Instead, it uses the frame names specified in the navigate_through_poses_bt_xml file.

## Feature request

Make the BtNavigator plugins automatically use the same parameters as the bt_navigator, specifically the "global_frame" and "robot_base_frame" parameters for this case.

SteveMacenski, 2023-05-15:
I see 4 instances where we have BT nodes with their own global and robot base frame ports https://github.com/search?q=repo%3Aros-planning%2Fnavigation2%20getInput(%22robot_base_frame%22&type=code.

The intent here is that there's a potential that we may want to perform these actions wrt other frames than those that are used for global navigation. I will admit looking over these 4, it seems unlikely that the base frame would ever change (but the global frame may wrt the distance controller and condition). 

I think what could be reasonable is to change this node from having a default value in the BT port to:
- Check if its set in the port, then use that value
- If not, then the port result should be an empty string, in which case, use the ROS 2 params API to get the BT navigator's version

That would let us automatically use the BT navigator's value when its set and not overrided by the BT for some specific behavior. Does that sound reasonable to you? Would you be open to submitting a PR to help make this change?

SteveMacenski, 2023-05-23:
Ah yes, we pass a separate node to the BT from the BT navigator so we don't lock up the BT navigator processing BT-specific callbacks or requests. There's almost definitely a better way to do this with the design of custom executors, but this is the best way with the default tools (e.g. totally isolate them).

We could use the param clients to get the parameters from the BT Navigator's node https://answers.ros.org/question/340600/how-to-get-ros2-parameter-hosted-by-another-node/, that comes with the cost of involving the network if called in the `tick()` methods. Optionally, we could just make the "basic" parameters (eg frames, timeouts) params of the internal nodes instead. 

Optionally again, we could in the BT navigator declare the frames into the new internal nodes by `get_parameter` on the main node and then `declare_parameter` on the new nodes with the output of the main node's `get` command. That would mask this problem for any type of navigator downstream if put into the Navigator's base class.

SteveMacenski, 2023-06-05:
Any word?

HarunTeper, 2023-06-06:
I am still working on it (and other things).
When I find the time, I will either relocate the parameters to the separate nodes or try your last suggestion.

HarunTeper, 2023-06-24:
https://github.com/ros-planning/navigation2/pull/3653

closed 2023-07-28 by SteveMacenski