ros-navigation/navigation2 #3575: [amcl-3] terminate called after throwing an instance of 'tf2_ros::InvalidTimerHandleException'

staredeath, 2023-05-15:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04 LTS
- ROS2 Version: ROS2 version Galactic

I'm a new user in ROS2 and Navigation 2. When I try run AMCL in navigation2, I have a problem:

#### Steps to reproduce issue

```
[amcl-3] terminate called after throwing an instance of 'tf2_ros::InvalidTimerHandleException'

[amcl-3] what(): Invalid timer handle in remove()

[ERROR] [amcl-3]: process has died [pid 15010, exit code -6, cmd '/home/rtc/robot_ws/install/nav2_amcl/lib/nav2_amcl/amcl

--ros-args -r __node:=amcl --params-file /tmp/tmpw2jrx24q -r /tf:=tf -r /tf_static:=tf_static'].

```

#### Expected behavior

#### Actual behavior

#### Additional information

Anyone can help me? Thanks you !!!!!

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2023-05-19:
I'm sorry to hear the issue, but Galactic is EOL as of Nov of last year and will not receive further open-source support https://docs.ros.org/en/galactic/Releases.html

I'd recommend upgrading to a more modern distribution.

tonynajjar, 2023-06-21:
FYI, I've seen it also happen on Humble with the planner server, I'll add information if I find any

tfoote, 2023-08-11:
I found two more reports of this.

https://answers.ros.org/question/413833/amcl-3-terminate-called-after-throwing-an-instance-of-tf2_rosinvalidtimerhandleexception/

https://robotics.stackexchange.com/questions/25094/robot-state-publisher-dont-publish-on-tf-topic#comment40398_25094

universeteng, 2023-08-17:
Have you solved your problem

tonynajjar, 2023-08-17:
Looks like we have enough independent reports to reopen this issue, I'm happy to participate in the investigation

SteveMacenski, 2023-08-17:
Finding where that exception is thrown is probably a good first bet! We may just need to wrap an error case in a try/catch, but that's really odd that would happen starting more or less out of the blue - we haven't touched AMCL meaningfully in years. I suspect it might point us to a TF2 regression or somehow we're abusing it in AMCL

tonynajjar, 2023-08-29:
looks like the fix is [here](https://github.com/ros2/geometry2/pull/474)

SteveMacenski, 2023-08-29:
Thanks for the debugging! This was on my list to look into this week, thanks for the help!

closed 2023-08-29 by SteveMacenski