ros-navigation/navigation2 #3606: multi_tb3_simulation doesn't spawn world or robots

nikkusan, 2023-06-08:
[gazebo-1] [Wrn] [ModelDatabase.cc:340] Getting models from[http://models.gazebosim.org/]. This may take a few seconds.
[spawn_entity.py-7] [ERROR] [1686202573.180098818] [spawn_entity]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[spawn_entity.py-7] [ERROR] [1686202573.180436371] [spawn_entity]: Spawn service failed. Exiting.
[spawn_entity.py-3] [ERROR] [1686202573.229332304] [spawn_entity]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[spawn_entity.py-3] [ERROR] [1686202573.229669352] [spawn_entity]: Spawn service failed. Exiting.
[ERROR] [spawn_entity.py-7]: process has died [pid 67495, exit code 1, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity robot2 -file /opt/ros/humble/share/nav2_bringup/worlds/waffle.model -robot_namespace robot2 -x 0.0 -y -0.5 -z 0.01 -R 0.0 -P 0.0 -Y 0.0 --ros-args'].
[ERROR] [spawn_entity.py-3]: process has died [pid 67465, exit code 1, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity robot1 -file /opt/ros/humble/share/nav2_bringup/worlds/waffle.model -robot_namespace robot1 -x 0.0 -y 0.5 -z 0.01 -R 0.0 -P 0.0 -Y 0.0 --ros-args'].
[gazebo-1] [Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://turtlebot3_world]
[gazebo-1] [Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_world]
[gazebo-1] Error Code 12 Msg: Unable to find uri[model://turtlebot3_world]

xkaraman, 2023-06-08:
Hey, 
Make sure you exported the gazebo model files. Check #3601 and https://navigation.ros.org/getting_started/index.html as i had almost same problems for the a single TB3.
In my case gazebo world loads correctly with 2 waffles but on RViz i can't see anything yet.

nikkusan, 2023-06-08:
I have exported it but it still does not work

SteveMacenski, 2023-06-08:
Your errors are pretty clear 

```

[gazebo-1] [Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://turtlebot3_world]
[gazebo-1] [Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_world]
[gazebo-1] Error Code 12 Msg: Unable to find uri[model://turtlebot3_world]
```

You haven't provided the correct paths. I also just went ahead and tested the multirobot setup and its fine for me :smile:

xkaraman, 2023-06-09:
@SteveMacenski Maybe this should be reopened.

After trying to bring up the nav2 stack using the default launch and param files, i noticed some errors.

Current setup: 
Docker ros:humble
Jammy
ROS 2 Humble
Nav2: ros-humble-navigation2  1.1.6-1jammy.20230426.153445 Installed from apt

First, after reading https://github.com/ros-planning/navigation2/issues/3182, i had to make sure, to add ` use_composition': 'False'`, in the launch arguments of [multi_tb3_simulation](https://github.com/ros-planning/navigation2/blob/4a7dee1c8c272f2d550b7412f74f1a15cbf25161/nav2_bringup/launch/multi_tb3_simulation_launch.py#L138). for it to load the maps correctly. It is by default `true` in the single tb3_simulation launch file.  
Maybe add a launch argument for it, until the fix appears in debians or hardcore it to false for 1.1.6 version.

After add the above line, i can see the maps in rviz as i was able in the single tb3 simulation after pressing the nav2 startup button. Yet, i can see this error
```
Configuring bt_navigator
[bt_navigator-26] [INFO] [1686312175.982201679] [robot2.bt_navigator]: Configuring
[bt_navigator-26] [ERROR] [1686312176.108448143] []: Caught exception in callback for transition 10
[bt_navigator-26] [ERROR] [1686312176.108467644] []: Original error: Could not load library: libnav2_are_error_codes_active_condition_bt_node.so: cannot open shared object file: No such file or directory
```
After inspecting tag 1.1.6, i can see that [nav2_behavior_tree condition nodes](https://github.com/ros-planning/navigation2/tree/1.1.6/nav2_behavior_tree/include/nav2_behavior_tree/plugins/condition), there is no `are_error_code_active_condition_bt_node` along with `nav2_would_a_controller_recovery_help_condition_bt_node`, `nav2_would_a_planner_recovery_help_condition_bt_node`, `nav2_would_a_smoother_recovery_help_condition_bt_node` as opposes to tag 1.2.0.

But tag 1.1.6 params for `multirobot_params_1/2.yaml` (single_tb3.yaml does not) have them as plugins 
https://github.com/ros-planning/navigation2/blob/4a7dee1c8c272f2d550b7412f74f1a15cbf25161/nav2_bringup/params/nav2_multirobot_params_1.yaml#L70-L73

so it fails to load them and can't give navigation commands. Commenting the four plugins out from the both param files, one can give navigation commands, to both robots.

I hope this helps clarifies what i encountered. Hope i didn't overlook something as you said, you did run and tested the setup on your machine.

SteveMacenski, 2023-06-09:
That error appears to be the result of https://github.com/ros-planning/navigation2/pull/3383 which was backported to Humble https://github.com/ros-planning/navigation2/commit/040c585f413732050296070d99db0d9ea362dc14

I fixed that in https://github.com/ros-planning/navigation2/commit/9b91662eacba3cc73719cdebc7ee1d409f5a3a47 so its not in the humble main branch but I haven't run a release to Humble since then. 

I was planning on doing a Humble sync today, so I can take care of that shortly here. For now, you fix is to remove those nodes from your configuration file so you can pass through -- or if you want to own the configuration file, then you can just clone and build Humble from source and it will be solved for you. 

I'll let @AlexeyMerzlyakov comment on the composition point, but that should already be in Humble given the links above

xkaraman, 2023-06-12:
I've solved it by removing the nodes currently. #3383 seems to be backported but I couldn't get it up and running using `'use_composition': 'True'`. I can investigate more on why it fails if it is indeed merged.