ros-navigation/navigation2 #3608: Local costmap not update, and the timestamp of the received footprint is the previous time point.malloc(): unsorted double linked list corrupted

lizhao112396, 2023-06-08:
## Bug report

**Required Info:**

- Operating System:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2023-06-08:
@lizhao112396 please fill in the issue template with all of the requested information, its important if you want assistance.

lizhao112396, 2023-06-09:
Ok, thank you for your reply.
Add an planner_costmap  in the controller_ server , used for local path planning, and planner_costmap is also a separate thread. Occasional planner_ Costmap or local_ Costmap is not updated and footprint remains unchanged.
The location of the problem is as follows:
void
Costmap2DROS::updateMap()
{
  RCLCPP_DEBUG(get_logger(), "Updating map...");

  if (!stop_updates_) {
    // get global pose
    geometry_msgs::msg::PoseStamped pose;
    if (getRobotPose(pose)) {
      const double & x = pose.pose.position.x;
      const double & y = pose.pose.position.y;
      const double yaw = tf2::getYaw(pose.pose.orientation);
      layered_costmap_->updateMap(x, y, yaw);

      auto footprint = std::make_unique<geometry_msgs::msg::PolygonStamped>();
      footprint->header = pose.header;
      transformFootprint(x, y, yaw, padded_footprint_, *footprint);

      RCLCPP_DEBUG(get_logger(), "Publishing footprint");
      footprint_pub_->publish(std::move(footprint));
      initialized_ = true;
    }
  }
}
In fact, the problem with calling getRobotPose() is essentially caused by the received '/tf' not being updated, but the specific reason has not been found.

SteveMacenski, 2023-06-09:
Are you saying that you modified the code and the issues are w.r.t. the modifications of your version of the code? I'm not clear what you mean here or what the specific problem is.

SteveMacenski, 2023-06-16:
@lizhao112396 any info?

SteveMacenski, 2023-06-27:
No response + lack of actionable information.

closed 2023-06-27 by SteveMacenski