ros-navigation/navigation2 #3610: nav2_common hosts some useful, nav2-agnostic functionalities

130s, 2023-06-08:
## Feature request

#### Feature description
As seen in the answer in [answers.ros.org#416270](https://answers.ros.org/question/416270/how-to-concatinate-launchconfigurations-default-value-with-string-in-ros-humble-python-launch-file/), `nav2_common` solves some common problems that all .launch.py users among the entire ros2 community may run into, but the solution is limited to `nav2_common`. Needless to say, this could result into broader ros2 community not taking advantage of already-invented tool, and more maintenance in nav2 community.

#### Implementation considerations
- Consider sending those non-nav2 specific functionalities to upstream.

SteveMacenski, 2023-06-08:
I've never been opposed to upstreaming generally useful utilities and many in the past have gone to live in core repositories. Though, nav2_common doesn't depend on anything in Nav2 https://github.com/ros-planning/navigation2/blob/main/nav2_common/package.xml#L10 so you can easily just pull that into your project and use it in non-nav2 contexts. 

@clalancette / @jacobperron I see that you two are the top contributors to launch ROS.  [Would you have interest in any of the following capabilities](https://github.com/ros-planning/navigation2/tree/main/nav2_common/nav2_common/launch)?
- ReplaceString: reads a configuration file and if a particular string is found, its replaced with another 
- HasNodeParams: Used in launch conditions to check if a particular configuration file contains parameters for a particular node or not 
-  ReWrittenYaml: Used to remap params under a launch-set namespace, remap keys in case of launch-time changes

130s, 2023-06-16:
Just found [PickNikRobotics/launch_param_builder](https://github.com/PickNikRobotics/launch_param_builder) (CC I guess you're the author @JafarAbdi). I guess apparently it's a common desire to improve launch_py.

SteveMacenski, 2023-06-16:
I mean that's kind of the point of having an extensible API.

SteveMacenski, 2023-06-27:
There doesn't appear to be a great deal of interest - but if that changes, happy to give back anything that folks think have generic utility to the broader community! For now, these will live in `nav2_common` but not that it has no other dependencies on Nav2 so you can use it in any package, even non-navigation-related without any additional overhead from Nav2.

closed 2023-06-27 by SteveMacenski