ros-navigation/navigation2 #3625: `-fconcepts` not available as flag in clang

wolfv, 2023-06-14:
## Bug report

The `-fconcepts` flag doesn't exist in clang14 and makes it harder to compile the `mppi_controller` on macOS. Is the flag really needed?

https://github.com/ros-planning/navigation2/blob/01b84136493a47c2fe6ffa9d0b1132cbf132b189/nav2_mppi_controller/CMakeLists.txt#L76

Maybe one could also switch to C++20 – that should enable concepts in clang (if necessary).

Here is a log from a build in the RoboStack project: https://github.com/prefix-dev/rattler-build/actions/runs/5255875110/jobs/9496597928?pr=158#step:17:3507

**Required Info:**

- Operating System:
  - macOS
- ROS2 Version:
  - humble
- Version or commit hash:
  - source

#### Steps to reproduce issue

Compile with clang, and it doesn't compile due to `-fconcepts` flag not being available.

SteveMacenski, 2023-06-14:
First off - Hi! We all stand on the shoulders of giants and without your great work on `xtensor`, this work would not be possible! 

REP 2000 still has ROS at C++17, which I try to follow the best I can since this falls under the umbrella of "ROS" to many. I'm however not opposed to upgrading a little early to C++20 if that helps solve a problem as long as the plan for J-turtle is to be C++20 so we're not preempting it by more than a few months until the next distribution. 

@wjwwood: is C++20 destined for J-turtle? 

---

But to answer the other part of the question, yes, it cannot simply be removed: 

I tried compiling it without that and I get problems with `auto` use in method arguments. 

```
/home/steve/Documents/navigation2_ws/src/navigation2/nav2_mppi_controller/test/utils/factory.hpp:174:3: error: use of ‘auto’ in parameter declaration only available with ‘-std=c++20’ or ‘-fconcepts’ [-Werror]
  174 |   auto costmap_ros)
```

wolfv, 2023-06-14:
Thanks for the recognition! I was super happy to see xtensor & xsimd being used, too :)

OK, I see. I guess one might be able to go back to the old-school way of using a template 

```
template<typename T>
void func(T param) { /* ... */ }
```

Or we could enable `-std=c++20` for clang, and keep `-fconcepts` for gcc?

SteveMacenski, 2023-06-14:
Some of the `auto` arguments are easy to resolve (like my example above), but others are very much so... not due to xtensor's complex types. Templating if they were all pure header methods would work, but most of our implementations are broken out into cpp files so we'd need to tell the compiler the ID of the types used. Its an all or nothing kind of thing, though.

Moving to C++20 in general doesn't seem unreasonable, but I do need to have a conversation with William or someone at Intrinsic on those tentative plans to know if the easy solution is the best solution. However, if that doesn't seem viable, having the setting to use C++20 for MacOS / clang seems very reasonable since they're both non-Tier 1 supported anyway so I don't need to make any statement about consistency. 

That could go in either https://github.com/ros-planning/navigation2/blob/main/nav2_mppi_controller/CMakeLists.txt or the nav2_common macro (https://github.com/ros-planning/navigation2/blob/main/nav2_common/cmake/nav2_package.cmake) to apply to all packages. Dealer's choice.

SteveMacenski, 2023-06-24:
I chatted with Chris, it doesn't sound like there's a plan to make C++20 the default in Jazzy. @wolfv I think the most sensible thing is to have C++20 set for only MPPI controller so that its just this one algorithm that has it set as an exception so we can remove the flag for MacOS. We could use C++17 for non-Mac users, but I think for a plugin of the framework as long as its not the framework itself, we can make an exception for one package to have it use C++20 globally.

Does that work for you?

SteveMacenski, 2023-06-27:
https://github.com/ros-planning/navigation2/pull/3660 resolves

SteveMacenski, 2023-06-27:
That did not compile, I'll look into this later if I have time. I did this late in the afternoon thinking it would be a trivial update

wolfv, 2023-06-27:
Thanks for your reply with the reasoning - makes sense! 
I think a C++20 flag definitely works for our purposes (even if only for macOS).

Looks like you are hitting a new warning / error with C++20 :(

SteveMacenski, 2023-06-27:
@wolfv I've looked into it a bit, it seems to come from `xtl` 

```
    inlined from ‘void mppi::Optimizer::integrateStateVelocities(mppi::models::Trajectories&, const mppi::models::State&) const’ at /home/steve/Documents/navigation2_ws/src/navigation2/nav2_mppi_controller/src/optimizer.cpp:313:87:
/usr/include/c++/11/bits/stl_algobase.h:431:30: error: ‘void* __builtin_memcpy(void*, const void*, long unsigned int)’ writing 32 bytes into a region of size 16 overflows the destination [-Werror=stringop-overflow=]
  431 |             __builtin_memmove(__result, __first, sizeof(_Tp) * _Num);
      |             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/xtensor/xutils.hpp:27,
                 from /usr/include/xtensor/xtensor_simd.hpp:16,
                 from /usr/include/xtensor/xstorage.hpp:23,
                 from /usr/include/xtensor/xbuffer_adaptor.hpp:22,
                 from /usr/include/xtensor/xtensor.hpp:19,
                 from /home/steve/Documents/navigation2_ws/src/navigation2/nav2_mppi_controller/include/nav2_mppi_controller/optimizer.hpp:21,
                 from /home/steve/Documents/navigation2_ws/src/navigation2/nav2_mppi_controller/src/optimizer.cpp:15:
/usr/include/xtl/xsequence.hpp: In member function ‘void mppi::Optimizer::integrateStateVelocities(mppi::models::Trajectories&, const mppi::models::State&) const’:
/usr/include/xtl/xsequence.hpp:130:19: note: destination object ‘ret’ of size 16
  130 |                 R ret;
      |                   ^~~
```

This doesn't look like something we can solve from Nav2's side but xtl needs to have this warning that's triggering `-Werror=stringop-overflow=` resolved. Once that's good to go, it should be an easy update to enable C++20 for MacOS users! Is that something you can potentially look at given your proximity to `xtensor`?

SteveMacenski, 2023-07-28:
@wolfv any thoughts?

SteveMacenski, 2023-08-30:
I'm not sure what the action is here given:
- C++20 isn't going to be the used version in ROS 2 for the foreseeable future 
- C++20 doesn't compile for us due to build warnings that are generated from `xtl` 
- C++17 + fconcepts doesn't work for MacOS, but does for all others

It seems we're in a rock and a hard place where the two options are to (1) just not support MacOS natively or (2) xtl needs updates so we can do so. The changes to `xtl` seem more in @wolfv 's wheelhouse and not entirely something actionable here. 

If we don't have some plan / method, I may have to default to (1) for the time being until there's a MacOS user willing to put in the effort to fix the `xtl` build warnings.

SteveMacenski, 2023-09-10:
Closing ticket for now. There is not support for MacOS for the time being for the reasons listed above. The fix to allow requires updates to `xtl`.

closed 2023-09-10 by SteveMacenski