ros-navigation/navigation2 #3640: BehaviorTree issue when SubTrees are used

facontidavide, 2023-06-20:
## Bug report

This problem was originally reported here by @Rid0n

https://github.com/BehaviorTree/BehaviorTree.CPP/issues/563

The issue is here.

https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server_impl.hpp#LL139C1-L144C97

The problem is that each subree has its own blackboard, unless the Subtree was created with "__shared_blackboard=true".

The `SubtreePlus / _autoremap`  works differently and, instead of using the blackboard of the parent, it remap the single ports.

But this means that Subtrees can not see the entries `node`, `server_timeout` and `bt_loop_duration`.

There are two solutions:

1. The correct one. Stop doing that :smile: . You are abusing the blackboard and expect it to do things that is not supposed to do. The solution is to follow the pattern suggested in [this tutorial](https://www.behaviortree.dev/docs/3.8/tutorial-basics/tutorial_08_additional_args). There is a more concrete example here: https://github.com/BehaviorTree/BehaviorTree.ROS2

2. The quick and dirty solution: 

Change this: https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server_impl.hpp#L203

For 

```c++
tree_ = bt_->createTreeFromFile(filename);
for(auto& blackboard: tree.blackboard_stack)
{
  blackboard->set<rclcpp::Node::SharedPtr>("node", client_node_);  
  blackboard->set<std::chrono::milliseconds>("server_timeout", default_server_timeout_);  
  blackboard->set<std::chrono::milliseconds>("bt_loop_duration", bt_loop_duration_);   
}
```

facontidavide, 2023-06-20:
My goal is to have a preliminary discussion on this thread, and submit a PR later, once we agreed how to proceed.

SteveMacenski, 2023-06-20:
That suggestion looks more or less the same to me, just moved around in a way that doesn't impact end-use. I've got no problem with it. 

I don't see how this is _abusing_ the blackboard though. The blackboard is to store information that needs to be shared with other BT nodes in the context of execution. Throwing in the shared resources like the goal pose or tools needed to obtain results seems totally in line with its intent. 

But that's perhaps here nor there.

facontidavide, 2023-06-20:
> I don't see how this is abusing the blackboard though.

Let me explain :smiley:

The recommended abstraction to use is not the blackboard, but the input/output ports.
It is at that level that things are tested, including:

- type checking, 
- conversions from/to strings 
- manual or automatic remapping.

When using the blackboard directly, **you are on your own**. Things may work or not (as in this case, they don't), I don't offer any guaranty that things will work as you may expect they should.

I am very careful in my documentation/tutorials about **not** using the blackboard directly for that reasons.
But Nav2 created a precedent that I need to address over and over again with my users.

SteveMacenski, 2023-06-20:
When nodes are passing data to each other or params specific to a particular BT node, yes I agree. And we do that broadly. 

However you **must** be able to insert information onto the blackboard when starting a BT because you need to be able to communicate to the BT template the request specifics (e.g. the goal pose requested or other task-specific request parameters). These are read into the blackboard and then used as input ports to the appropriate BT nodes that need them (e.g. `{goal_pose}`). That's a straight up requirement and I don't know how someone could argue against that need. If you have a BT doing a task template (e.g. go to pose, pick up a box, whatever), you need to give it the request specific inputs somehow and that seems to me to be the role of the blackboard.  

Separately, these items you see added are global parameters that are needed by virtually all BT nodes as a shared resource. It makes little practical sense to have every BT node have an input port of a node or timeouts -- those XML files would just get massive and it would be silly to see `node={node} server_timeout={server_timeout} ...` written over and over again on every line. I see the argument for why when people use it, it should be an input port which points to the blackboard variable. But given every single node needs this same shared resource, its a shorthand in Nav2's internal nodes that we just grab it from the blackboard because we know its there in all the BT factories. I don't think your issue is how we're putting shared resources onto the blackboard (as the previous paragraph points out, that is required for task templates no matter what), I think your issue is how the BT nodes themselves are grabbing them directly instead of using the port/blackboard grabbing with `{}`. I can appreciate that. 

> But Nav2 created a precedent, and I find myself fixing issues like these that are derived from using the "wrong" API.

I'm not asking you to. But it sounds like even if we did  `node={node} server_timeout={server_timeout} ...` wouldn't actually have avoided the issue. The issue is that we didn't throw it on all the blackboards which is an easy enough solution.

SteveMacenski, 2023-06-20:
Making the change 

```
tree_ = bt_->createTreeFromFile(filename);
for(auto& blackboard: tree.blackboard_stack)
{
  blackboard->set<rclcpp::Node::SharedPtr>("node", client_node_);  
  blackboard->set<std::chrono::milliseconds>("server_timeout", default_server_timeout_);  
  blackboard->set<std::chrono::milliseconds>("bt_loop_duration", bt_loop_duration_);   
}
```

Is an easy fix that I'm happy to make -- is there something else you wanted to change as well?

SteveMacenski, 2023-08-02:
@facontidavide how should we proceed? I don't want to leave this one hanging open if we have a solution. If that's the only change you want me to make, I'm happy to do it, but I need to know :smile:

SteveMacenski, 2023-10-28:
@facontidavide pinging -- I'm going to execute on the solution above if I don't hear otherwise

SteveMacenski, 2023-10-28:
https://github.com/ros-planning/navigation2/pull/3911

closed 2023-10-28 by SteveMacenski

facontidavide, 2023-10-30:
well done! Sorry for the high latency in responding the ping