ros-navigation/navigation2 #3661: Local costmap clearing on robot rotation

nino192, 2023-06-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04.
- ROS2 Version:
  - ROS2 Humble
- Version or commit hash:
  - 1.1.6-1jammy.20230426.090204

#### Steps to reproduce issue
Yaml of costmap params:
```
local_costmap:
  local_costmap:
    ros__parameters:
      update_frequency: 5.0
      publish_frequency: 2.0
      global_frame: odom
      robot_base_frame: base_link
      use_sim_time: True
      rolling_window: True
      width: 12
      height: 12
      resolution: 0.05
      plugins: ["obstacle_layer", "inflation_layer", "voxel_layer"]
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: True
        observation_sources: pointcloud1 pointcloud2
        footprint_clearing_enabled: True
        max_obstacle_height: 2.0
        combination_method: 1
        pointcloud1:
          topic: /camera_1/points
          obstacle_max_range: 6.0
          obstacle_min_range: 0.6
          raytrace_max_range: 6.5
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          min_obstacle_height: 0.1
          clearing: True
          marking: True
          data_type: "PointCloud2"
          inf_is_valid: false
        pointcloud2:
          topic: /camera_2/points
          obstacle_max_range: 6.0
          obstacle_min_range: 0.6
          raytrace_max_range: 6.5
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          min_obstacle_height: 0.1
          clearing: True
          marking: True
          data_type: "PointCloud2"
          inf_is_valid: false
      voxel_layer:
        plugin: "nav2_costmap_2d::VoxelLayer"
        enabled: True
        publish_voxel_map: True
        origin_z: 0.0
        z_resolution: 0.15
        z_voxels: 10
        max_obstacle_height: 2.0
        mark_threshold: 0
        observation_sources: pointcloud1 pointcloud2 pointcloud3
        pointcloud1:
          topic: /camera1/points
          obstacle_max_range: 6.0
          obstacle_min_range: 0.6
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "PointCloud2"
        pointcloud2:
          topic: /camera2/points
          obstacle_max_range: 6.0
          obstacle_min_range: 0.6
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "PointCloud2"
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        enabled: true
        inflation_radius: 0.3
        cost_scaling_factor: 1.0
        inflate_unknown: false
        inflate_around_unknown: true

```
Launching of local costmap node snippet:
```
    controller_server_node_sim = Node(condition=IfCondition(sim_gazebo),
                                  package='nav2_controller',
                                  executable='controller_server',
                                  output='screen',
                                  respawn=use_respawn,
                                  respawn_delay=2.0,
                                  parameters=[costmap_params_gazebo],
                                  arguments=['--ros-args', '--log-level', log_level])
```
URDF of gazebo simulated cameras:
```
   <!-- Camera 1 simulator -->

    <gazebo reference="rs_camera_1_link">
        <material>Gazebo/Red</material>

        <sensor name="camera_1" type="depth">
            <pose> 0 0 0 0 0 0 </pose>
            <visualize>true</visualize>
            <update_rate>10</update_rate>
            <camera>
                <horizontal_fov>1.089</horizontal_fov>
                <image>
                    <format>R8G8B8</format>
                    <width>640</width>
                    <height>480</height>
                </image>
                <clip>
                    <near>0.05</near>
                    <far>8.0</far>
                </clip>
            </camera>
            <plugin name="camera_1_controller" filename="libgazebo_ros_camera.so">
                <frame_name>camera_1_link_optical</frame_name>
            </plugin>
        </sensor>
    </gazebo>

    <!-- Camera 2 simulator -->

    <gazebo reference="rs_camera_2_link">
        <material>Gazebo/Red</material>

        <sensor name="camera_2" type="depth">
            <pose> 0 0 0 0 0 0 </pose>
            <visualize>true</visualize>
            <update_rate>10</update_rate>
            <camera>
                <horizontal_fov>1.089</horizontal_fov>
                <image>
                    <format>R8G8B8</format>
                    <width>640</width>
                    <height>480</height>
                </image>
                <clip>
                    <near>0.05</near>
                    <far>8.0</far>
                </clip>
            </camera>
            <plugin name="camera_2_controller" filename="libgazebo_ros_camera.so">
                <frame_name>camera_2_link_optical</frame_name>
            </plugin>
        </sensor>
    </gazebo>
```

#### Expected behavior
Local costmap doesen't get cleared on rotation of the robot, and random line-like obstacles don't show up when using simulated depth cameras as gazebo plugins.

#### Actual behavior
Local costmap gets cleared on robot rotation while doing teleoperation. Line like imaginary obstacles are occasionally shown in the costmap. (demonstrated on attached video, timestamp 0:25)

#### Additional information

[Screencast from 14.06.2023 11:05:38.webm](https://github.com/ros-planning/navigation2/assets/113939473/dc76915a-4168-45b0-87c1-0fb7f1c1c8a5)

nino192, 2023-06-27:
always_send_full_costmap parameter was set to false

closed 2023-06-27 by nino192

nakai-omer, 2023-08-04:
@nino192 Thank you very much, wasted hours on this!