ros-navigation/navigation2 #3666: Building nav2 from source fails, ubuntu 22.04 ROS2 Humble

cocodmdr, 2023-06-29:
## Bug report

**Required Info:**

- Operating System:

Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.2 LTS
Release:	22.04
Codename:	jammy

- ROS2 Version:

ROS_VERSION=2
ROS_PYTHON_VERSION=3
AMENT_PREFIX_PATH=/opt/ros/humble
PYTHONPATH=/opt/ros/humble/lib/python3.10/site-packages:/opt/ros/humble/local/lib/python3.10/dist-packages
LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/opt/ros/humble/opt/rviz_ogre_vendor/lib:/opt/ros/humble/lib/x86_64-linux-gnu:/opt/ros/humble/lib
ROS_LOCALHOST_ONLY=0
GAZEBO_MODEL_PATH=:/opt/ros/humble/share/turtlebot3_gazebo/models
PATH=/opt/ros/humble/bin:/home/corentin/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
ROS_DISTRO=humble

- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```
cd ~/ros2_ws/src
# clone nav2 :
git clone https://github.com/ros-planning/navigation2.git -b humble
cd ..
# install dependencies: 
rosdep install -i --from-path src --rosdistro humble -y
# build: 
colcon build
```

#### Expected behavior

no error while building

#### Actual behavior
error while building
`
--- stderr: nav2_smac_planner                                            
In file included from /home/corentin/ros2_ws/src/navigation2/nav2_smac_planner/test/test_nodelattice.cpp:23:
/opt/ros/humble/src/gtest_vendor/include/gtest/gtest.h: In instantiation of ‘testing::AssertionResult testing::internal::CmpHelperGT(const char*, const char*, const T1&, const T2&) [with T1 = long unsigned int; T2 = int]’:
/home/corentin/ros2_ws/src/navigation2/nav2_smac_planner/test/test_nodelattice.cpp:350:3:   required from here
/opt/ros/humble/src/gtest_vendor/include/gtest/gtest.h:1629:28: warning: comparison of integer expressions of different signedness: ‘const long unsigned int’ and ‘const int’ [-Wsign-compare]
 1629 | GTEST_IMPL_CMP_HELPER_(GT, >);
      |                            ^
/opt/ros/humble/src/gtest_vendor/include/gtest/gtest.h:1609:12: note: in definition of macro ‘GTEST_IMPL_CMP_HELPER_’
 1609 |   if (val1 op val2) {\
      |            ^~
---
`
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2023-06-29:
@cocodmdr can you verify that this patch resolves the issue? https://github.com/ros-planning/navigation2/pull/3667 If so, I'll backport to Humble to fix that for you as well.

cocodmdr, 2023-06-30:
Works,
I get another small error :
--- stderr: nav2_simple_commander                                                          
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---

SteveMacenski, 2023-06-30:
That's not in our control, that needs a larger update in ROS 2's handling of python packages before we can address it here. That's simply a warning that you don't need to worry about

closed 2023-06-30 by SteveMacenski