ros-navigation/navigation2 #3668: Using Simple Commander API for multi robot systems

AryanArora12, 2023-06-29:
----
## Feature request
Using [Simple Commander API](https://navigation.ros.org/commander_api/index.html) for multi robot systems. This is in continuation to [this](https://github.com/ros-planning/navigation2/issues/2415) post.  

#### Feature description
I am making a multi-robot simulation and for my purposes, I want to use the Python API to send navigation goals to different robots. As of now, I cannot see how to do that with two robots(that is add a namespace and setting up goal poses as well as set initial poses individually for each robot). Any guidance on how to do this would be helpful. 

#### Implementation considerations

I think the solution proposed in the [post](https://github.com/ros-planning/navigation2/issues/2415) mentioned earlier sounds good, if not implemented yet.

SteveMacenski, 2023-06-29:
That was outside of the scope of the original "simple" nature of the commander, but it shouldn't be prohibitively difficult to make work and it would be a great contribution back!

I'd add the namespace support and than have a different simple commander object for each of the robots. That should separate them from each other and namespaced by the appropriate services and such.

AryanArora12, 2023-06-29:
Cool! I thought about the same thing but was struggling in adding the namespace functionality. I'll work on it over the next few days and provide an update if I get any progress/or am able to implement that successfully.

SteveMacenski, 2023-06-29:
I think its mostly just adding the namespace when we create all the clients in the python API's constructor! 

OK!

SteveMacenski, 2023-07-04:
Whoops wrong button. Can you open a PR? Happy to merge it so others can use!

AryanArora12, 2023-07-05:
Yeah sure. I'll try and do a few more tests for sanity check and then open a PR by this week.

ipa-rar, 2023-07-21:
@SteveMacenski Can you check this PR #3701

SteveMacenski, 2023-09-11:
https://github.com/ros-planning/navigation2/blob/main/nav2_simple_commander/nav2_simple_commander/robot_navigator.py#L50  https://docs.ros2.org/latest/api/rclpy/api/node.html. That might just do everything for us automatically. Can you try that instead?

SteveMacenski, 2023-09-11:
https://github.com/ros-planning/navigation2/pull/3803 implements + tested

PR #3803 (2023-09-11): Using Simple Commander API for multi robot systems
Resolves #3668
closed 2023-09-11 by SteveMacenski