ros-navigation/navigation2 #3669: navfn_planner: different implementation of mapToWorld() ?

joeldushouyu, 2023-06-29:
Hello all,

I am not sure if this should be worth being an issue or not.

I was reading over the navfn_planner.cpp to get some insight and help on how should implement my own planner.
I found that the implementation of mapToWorld() in  [navfn_planner.cpp](https://github.com/ros-planning/navigation2/blob/78c693a5d29da86d696ee535e303897220be4663/nav2_navfn_planner/src/navfn_planner.cpp#L510)  is slightly different from the implementation in  [costmap_2d.cpp](https://github.com/ros-planning/navigation2/blob/78c693a5d29da86d696ee535e303897220be4663/nav2_costmap_2d/src/costmap_2d.cpp#L281C17-L281C27) by not adding the 0.5 offset.

Thus, I am wondering if is there any specific purpose for the implementation in navfn_planner. 
Or could this version of mapToWorld() happen to be outdated? 

I am sorry if this seems to be a naive problem. I am still new to ros2 and nav2.

SteveMacenski, 2023-06-29:
The +0.5 was added in the original public commit to ROS 1 Navigation, so I can't track exactly why that was done, but in my opinion (not having been there, but looking at it) it was done to put the coordinates in the middle of the cell instead of the edges. The fact that NavFn does not do this is also on the first commit to the ROS 1 stack, but I think that might just be a small mistake no one's caught until now.

joeldushouyu, 2023-06-29:
Jus to be clear, so the version with +0.5 is the correct implementation?

Moreover, do you want me to submit a PR to fix this? Or this should be just left as it is for now?

SteveMacenski, 2023-06-29:
> Jus to be clear, so the version with +0.5 is the correct implementation?

I believe so. 

> Moreover, do you want me to submit a PR to fix this? Or this should be just left as it is for now?

I'm honestly on the fence. I took a look and it appears that the other planners use the +0.5 method, so I think its safe to update. If you opened a PR, that would cause an action item for me to test it out and make sure I don't see anything funky.

joeldushouyu, 2023-07-01:
@SteveMacenski  When I was comparing the implementation between worldToMap() in the  [navfn_planner.cpp](https://github.com/ros-planning/navigation2/blob/78c693a5d29da86d696ee535e303897220be4663/nav2_navfn_planner/src/navfn_planner.cpp#L510)  and   [costmap_2d.cpp](https://github.com/ros-planning/navigation2/blob/78c693a5d29da86d696ee535e303897220be4663/nav2_costmap_2d/src/costmap_2d.cpp#L281C17-L281C27), it seems to me that the logic is exactly the same, with some addition warning log.

Thus, do you think will it be a good idea if I also remove worldToMap() and use the worldToMap() in costmap_2d.cpp instead ?

SteveMacenski, 2023-08-02:
The NavFn planner has `std::round` in its `worldToMap` modeling, which would round world coordinates to the closest cell. Since we've rounded to the nearest cell, adding 0.5 would probably make everything shift a little further from the correct locale. Costmap2D's version does NOT have that, and instead when we static_cast it truncates it down to its lowest whole value - implying that the origin is actually at the lower left corner of the cell, not in its center. I think because of this, the +0.5 in the costmap version is to deal with the coordinates being in on the edges and not the cell centers, to make the output coordinates in the cell's centers.  

The tests I see failing in your PR I traced back to essentially small variations in the coordinate reported by the start/goal positions on the path not being the requested start/goal poses -- off by half a cell. Due to this, I think that the NavFn planner is doing the right thing for how its data is being considered in the planner. This is in spite of the fact that I can't exactly point to where the coordinate systems vary between them, but clearly how the data is being handled today is creating expected exact behavior while the changes are not. 

So, in that light, I closed your PR and this ticket, I think NavFn should stay the way it is. It seems to be planning in the proper coordinate space with these conversions in how it treats the grid. Evidence points to NavFn treating the coordinates as in the cell-centers so we're correct in not adding the extra +0.5.

closed 2023-08-02 by SteveMacenski