ros-navigation/navigation2 #3670: Build failed: nav2_util

SriHasitha, 2023-06-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - ROS2 Humble
- Version or commit hash:
  - installation by git clone https://github.com/ros-planning/navigation2.git

#### Expected behavior
colcon build successful
#### Actual behavior
--- stderr: nav2_util                                
CMake Error at test/CMakeLists.txt:7 (find_package):
  By not providing "Findtest_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "test_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "test_msgs" with
  any of the following names:

    test_msgsConfig.cmake
    test_msgs-config.cmake

  Add the installation prefix of "test_msgs" to CMAKE_PREFIX_PATH or set
  "test_msgs_DIR" to a directory containing one of the above files.  If
  "test_msgs" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< nav2_util [3.49s, exited with code 1]

Summary: 6 packages finished [1min 28s]
  1 package failed: nav2_util
  1 package had stderr output: nav2_util
  31 packages not processed

maryampirmoradi, 2023-06-30:
Hello! I also have a similar problem, somehow.

I am actually trying to install Nav2 on Nvidia Jetson AGX ORIN. I have built ROS2 humble from source on it successfully. When I try to build Nav2, the installation process stops, showing this:

--- stderr: nav2_util                           
CMake Error at CMakeLists.txt:17 (find_package):
  By not providing "Findbondcpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "bondcpp", but
  CMake did not find one.

  Could not find a package configuration file provided by "bondcpp" with any
  of the following names:

    bondcppConfig.cmake
    bondcpp-config.cmake

  Add the installation prefix of "bondcpp" to CMAKE_PREFIX_PATH or set
  "bondcpp_DIR" to a directory containing one of the above files.  If
  "bondcpp" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< nav2_util [8.55s, exited with code 1]

Summary: 6 packages finished [27.8s]
  1 package failed: nav2_util
  1 package had stderr output: nav2_util
  30 packages not processed

Please let me mention that before building, I tried to install the rosdep dependencies but faced the following lines:

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
nav2_behavior_tree: No definition of [behaviortree_cpp_v3] for OS version [focal]
nav2_bt_navigator: No definition of [behaviortree_cpp_v3] for OS version [focal]
nav2_smac_planner: No definition of [angles] for OS version [focal]
nav2_mppi_controller: [xsimd] defined as "not available" for OS version [focal]
nav2_smoother: No definition of [angles] for OS version [focal]
nav2_util: No definition of [bond] for OS version [focal]
nav2_system_tests: No definition of [gazebo_ros_pkgs] for OS version [focal]
nav2_constrained_smoother: No definition of [angles] for OS version [focal]
dwb_plugins: No definition of [angles] for OS version [focal]
nav2_costmap_2d: No definition of [angles] for OS version [focal]
dwb_critics: No definition of [angles] for OS version [focal]
nav2_waypoint_follower: No definition of [cv_bridge] for OS version [focal]
nav2_rotation_shim_controller: No definition of [angles] for OS version [focal]
nav2_bringup: No definition of [turtlebot3_gazebo] for OS version [focal]
nav2_lifecycle_manager: No definition of [diagnostic_updater] for OS version [focal]
nav2_controller: No definition of [angles] for OS version [focal]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

Can anyone help me with that? Is what I am going to do actually feasible?

Thank you in advance!

SteveMacenski, 2023-06-30:
You must install the proper dependencies via `rosdep` https://navigation.ros.org/development_guides/build_docs/build_troubleshooting_guide.html?highlight=rosdep#common-nav2-dependencies-build-failures

closed 2023-06-30 by SteveMacenski

juanlofer-eprosima, 2024-09-10:
In my case, for this library to be found before building I had to source the `install.bash` created after running rosdep.