ros-navigation/navigation2 #3675: Install navigation2 for ROS2 Humble on Docker, Ubuntu 22.04

vishwaslip, 2023-06-30:
Hello, I am trying to create a create a docker container that contains ROS 2 Humble, ros1_bridge, and nav2 to communicate with another ROS 1 noetic container. I have successful built the container with ros 2 humble and the ros1_bridge, though I am having trouble building nav2. 

**Required Info:**

- Operating System:
  
  - Docker, Ubuntu 22.04
- ROS2 Version:
  
  - Humble source
- Version or commit hash:
  
  - 3bb9a95719539fcdfedf520c63339bc3157f80b4

#### Steps to reproduce issue

Here is my [Dockerfile and devcontainer.json.zip](https://github.com/ros-planning/navigation2/files/11920039/Dockerfile.and.devcontainer.json.zip) for the ROS 2 container, im using vscode. Because of [this](https://docs.ros.org/en/humble/How-To-Guides/Using-ros1_bridge-Jammy-upstream.html), my docker file is the best way I've gotten it to build successfully. After it creates the container, I do in terminal:
```
mkdir -p ~/nav2_ws/src
cd ~/nav2_ws/src
git clone https://github.com/ros-planning/navigation2.git --branch humble
cd ~/nav2_ws
source /ros2_humble/install/setup.bash && colcon build
```

#### Expected behavior
Navigation 2 builds.

#### Actual behavior
```
--- stderr: nav2_util                           
CMake Error at CMakeLists.txt:17 (find_package):
  By not providing "Findbondcpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "bondcpp", but
  CMake did not find one.

  Could not find a package configuration file provided by "bondcpp" with any
  of the following names:

    bondcppConfig.cmake
    bondcpp-config.cmake

  Add the installation prefix of "bondcpp" to CMAKE_PREFIX_PATH or set
  "bondcpp_DIR" to a directory containing one of the above files.  If
  "bondcpp" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< nav2_util [2.44s, exited with code 1]
```

#### Additional information
 I have tried the following things, 

1.

```
sudo apt update && sudo apt install ros-humble-bondcpp
```
which returns:
```
...
Selecting previously unselected package python3-catkin-pkg-modules.
(Reading database ... 102699 files and directories currently installed.)
Preparing to unpack .../00-python3-catkin-pkg-modules_0.5.2-1_all.deb ...
Unpacking python3-catkin-pkg-modules (0.5.2-1) ...
dpkg: error processing archive /tmp/apt-dpkg-install-dCs1pi/00-python3-catkin-pkg-modules_0.5.2-1_all.deb (--unpack):
 trying to overwrite '/usr/lib/python3/dist-packages/catkin_pkg/__init__.py', which is also in package python3-catkin-pkg 0.4.24-2
...
Selecting previously unselected package ros-humble-bondcpp.
Preparing to unpack .../78-ros-humble-bondcpp_3.0.2-3jammy.20230623.070535_amd64.deb ...
Unpacking ros-humble-bondcpp (3.0.2-3jammy.20230623.070535) ...
Errors were encountered while processing:
 /tmp/apt-dpkg-install-67qYqC/00-python3-catkin-pkg-modules_0.5.2-1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

2.
```
cd ~/nav2_ws && rosdep update && rosdep install -yrq --from-paths src --ignore-src --rosdistro humble
```
which returns:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
nav2_mppi_controller: No definition of [xtensor] for OS version [Jammy Jellyfish]
nav2_lifecycle_manager: No definition of [diagnostic_updater] for OS version [Jammy Jellyfish]
nav2_behavior_tree: No definition of [behaviortree_cpp_v3] for OS version [Jammy Jellyfish]
nav2_waypoint_follower: No definition of [cv_bridge] for OS version [Jammy Jellyfish]
nav2_util: No definition of [bond] for OS version [Jammy Jellyfish]
nav2_smoother: No definition of [angles] for OS version [Jammy Jellyfish]
nav2_system_tests: No definition of [gazebo_ros_pkgs] for OS version [Jammy Jellyfish]
nav2_controller: No definition of [angles] for OS version [Jammy Jellyfish]
nav2_smac_planner: No definition of [angles] for OS version [Jammy Jellyfish]
nav2_bt_navigator: No definition of [behaviortree_cpp_v3] for OS version [Jammy Jellyfish]
dwb_critics: No definition of [angles] for OS version [Jammy Jellyfish]
dwb_plugins: No definition of [angles] for OS version [Jammy Jellyfish]
nav2_rotation_shim_controller: No definition of [angles] for OS version [Jammy Jellyfish]
nav2_constrained_smoother: No definition of [angles] for OS version [Jammy Jellyfish]
nav2_bringup: No definition of [slam_toolbox] for OS version [Jammy Jellyfish]
nav2_costmap_2d: No definition of [angles] for OS version [Jammy Jellyfish]
Continuing to install resolvable dependencies...
executing command [sudo -H apt-get install -y -qq libomp-dev]
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
executing command [sudo -H apt-get install -y -qq python3-zmq]
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
executing command [sudo -H apt-get install -y -qq lcov]
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
executing command [sudo -H apt-get install -y -qq libgraphicsmagick++1-dev]
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
executing command [sudo -H apt-get install -y -qq graphicsmagick-libmagick-dev-compat]
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
executing command [sudo -H apt-get install -y -qq libceres-dev]
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y -qq libomp-dev] failed
  apt: command [sudo -H apt-get install -y -qq python3-zmq] failed
  apt: command [sudo -H apt-get install -y -qq lcov] failed
  apt: command [sudo -H apt-get install -y -qq libgraphicsmagick++1-dev] failed
  apt: command [sudo -H apt-get install -y -qq graphicsmagick-libmagick-dev-compat] failed
  apt: command [sudo -H apt-get install -y -qq libceres-dev] failed
  apt: Failed to detect successful installation of [libomp-dev]
  apt: Failed to detect successful installation of [python3-zmq]
  apt: Failed to detect successful installation of [lcov]
  apt: Failed to detect successful installation of [libgraphicsmagick++1-dev]
  apt: Failed to detect successful installation of [graphicsmagick-libmagick-dev-compat]
  apt: Failed to detect successful installation of [libceres-dev]
```

3.
```
sudo apt --fix-broken install
```
which returns:

```
The following additional packages will be installed:
  python3-catkin-pkg-modules
The following NEW packages will be installed:
  python3-catkin-pkg-modules
0 upgraded, 1 newly installed, 0 to remove and 5 not upgraded.
78 not fully installed or removed.
Need to get 43.2 kB of archives.
After this operation, 258 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://packages.ros.org/ros2/ubuntu jammy/main amd64 python3-catkin-pkg-modules all 0.5.2-1 [43.2 kB]
Fetched 43.2 kB in 0s (87.8 kB/s)                     
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 110302 files and directories currently installed.)
Preparing to unpack .../python3-catkin-pkg-modules_0.5.2-1_all.deb ...
Unpacking python3-catkin-pkg-modules (0.5.2-1) ...
dpkg: error processing archive /var/cache/apt/archives/python3-catkin-pkg-modules_0.5.2-1_all.deb (--unpack):
 trying to overwrite '/usr/lib/python3/dist-packages/catkin_pkg/__init__.py', which is also in package python3-catkin-pkg 0.4.24-2
Errors were encountered while processing:
 /var/cache/apt/archives/python3-catkin-pkg-modules_0.5.2-1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

SteveMacenski, 2023-06-30:
Please search for solutions to your dependency issues and if you cannot figure out on your own, ask on ROS Answers.

closed 2023-06-30 by SteveMacenski

vishwaslip, 2023-07-18:
Hello, I don't mean to keep bugging. Though I have tried many things, and that is why I am implying this may be a problem with Nav2, or perhaps not enough documentation on what I need to do correctly. As of right now, here are the most repeatable instructions that I feel show that this may be a Nav2 bug. 

**Required Info:**

- Operating System:
  
  - Docker, Ubuntu 22.04
- ROS2 Version:
  
  - Humble source
- Version or commit hash:
  
  - ead48a9039c80cadb18dcfbb21fff0bdd496ef03

**Steps to Reproduce Issue**
Here is my [repo](https://github.com/vishwaslip/docker-nav2-humble) where you can reproduce the steps. In the Dockerfile, from the Ubuntu Jammy base image, I download ROS2 from source exactly as stated on the official ROS2 documentation. Then once it is built, in the terminal, I proceed to attempt to install Nav 2 from source. I had to create my own .repos file, as I couldn't get it to work with the tools/underlay.repos in the humble branch, even un-commenting every line. 

1. `sudo su`, Password: password
2. `mkdir -p /opt/ros/nav2/src && cd /opt/ros/nav2/src`
3. `git clone https://github.com/ros-planning/navigation2.git --branch humble`
4. `source /opt/ros/${ROS2_DISTRO}/install/setup.bash && cd /opt/ros/nav2`
5. `vcs import src < /path/to/nav2_humble_dependencies.repos`
6. `rosdep update && rosdep install -y --ignore-src --from-paths src`
7. `colcon build --merge-install`

**Expected outcome**

Nav 2 Builds

**Actual Behavior**

nav2_smac_planner errors:
```
--- stderr: nav2_smac_planner                                                                                                                                               
CMake Error at CMakeLists.txt:75 (add_library):
  Target "nav2_smac_planner" links to target "Boost::serialization" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:75 (add_library):
  Target "nav2_smac_planner" links to target "Boost::filesystem" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:75 (add_library):
  Target "nav2_smac_planner" links to target "Boost::system" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

  ...

 CMake Error at /opt/ros/humble/install/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_lattice_node" links to target "Boost::serialization" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/install/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/humble/install/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  test/CMakeLists.txt:114 (ament_add_gtest)

CMake Error at /opt/ros/humble/install/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_lattice_node" links to target "Boost::filesystem" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/install/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/humble/install/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  test/CMakeLists.txt:114 (ament_add_gtest)

CMake Error at /opt/ros/humble/install/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_lattice_node" links to target "Boost::system" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/install/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/humble/install/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  test/CMakeLists.txt:114 (ament_add_gtest)

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< nav2_smac_planner [2min 46s, exited with code 1]
```

**Additional Information**
This setup should have libboost-all-dev version 1.74, correctly debian installed to the best of my knowledge. 

This whole process is separate from ros1_bridge, as right now I am just trying to get nav2 from source work with ros2 humble from source. It doesn't seem like a system problem because I am operating from a docker container, and following all official documentation. So I feel like this is either a Nav2, or documentation error, and should be addressed, as I feel like this process should not be this difficult.