ros-navigation/navigation2 #3682: BT_service node and BT_action node should check server at runtime not crash on creation

BriceRenaudeau, 2023-07-07:
## Bug report
- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Iron
  - from source
- DDS implementation:
  - Cyclone

#### Steps to reproduce issue

1. Start a BT action server node with a BT using bt_service_noe or bt_action_node.
2. Wait for 1s (hardcoded) 
3. The node has crashed

#### Expected behavior

The BT is created.
- If the service server is down, tick() call return failure
- If the service server is up, tick() call return normally

#### Actual behavior

The BT is created and crash after 1s

#### Additional information
Lines in bt_service_node.hpp
https://github.com/ros-planning/navigation2/blob/778c0faee41788c9e3323a420ff2c10122ba4aae/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp#L80-L88

Lines in bt_action_node.hpp
https://github.com/ros-planning/navigation2/blob/778c0faee41788c9e3323a420ff2c10122ba4aae/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L96-L103

BriceRenaudeau, 2023-07-07:
I can do a small PR to change the behavior.

SteveMacenski, 2023-07-07:
Change to what?

BriceRenaudeau, 2023-07-10:
I cannot see a way to check the bt_nodes state at construction time, except a `try{}catch{}` to catch the thrown exception.

Yes, a `FAILURE` due to a server not being connected can pass silently.

We got the issue because of some DDS latency and a race condition appears on the BT creation. We could make the timeout of the `wait_for_service(1s)` a parameter like the execution parameter?

SteveMacenski, 2023-07-10:
Do you think that 2, 5, 10 seconds would fix a problem 1 would not? 1s seems like a massive amount of time that unless its on another computer remotely connected.

BriceRenaudeau, 2023-07-11:
My opinion is that changing the timeout will not fix the issue and a throw or an infinite loop (trying forever to connect to the server) neither.

My preferred solution si a runtime `FAILURE` result. It may be hard to find but the ERROR logging should be enough help to find.

SteveMacenski, 2023-07-13:
This doesn't seem like a sound solution for reliability. We shouldn't be executing a tree if its resources are not yet available - I don't mind adjusting how the check on tree setup is accomplished, but I think failing at runtime to perform a critical behavior is a massive safety issue - especially if the failed behavior was a safety critical element like "stop robot". Since these are generic, its currently not possible to distinguish them.

Does that make sense? I agree your suggestion makes things easier, but I think that could create unsafe situations + the check for wait if server would add delays in the behavior you're interested in. Perhaps some DDS config changes and/or another way to check if the server is up (or a fallback try?) could make sense?

huemerj, 2023-07-19:
I've experienced a similar issue and a quick fix was to set the hard-coded timeout to 5s. Maybe we can use the already existing server_timeout parameter for that, similar to https://github.com/BehaviorTree/BehaviorTree.ROS2/blob/11f871c60d52c3bfedb2184828045263fa03e40e/include/behaviortree_ros2/bt_service_node.hpp#L216.

SteveMacenski, 2023-07-22:
I'm OK making that a param, but I think @BriceRenaudeau mentioned that changing that value wouldn't actually fix the issue 100% of the time necessarily. What do you think @BriceRenaudeau? Would just making a param to allow a longer time to process be a fix for you?

BriceRenaudeau, 2023-07-24:
The `bt_navigator` node creates a BT every time there is a new goal. So the crash(throw) is not a problem if it is caught.

In my case, it was a standard node that creates a BT once and for all. When it crashes, the node is definitely down.
1s to start is very long and should not have been an issue unless DDS issues...

The expected behavior may vary according to BT. In my case, returning FAILURE (+log) is enough. But in many other cases, the FAILURE will trigger other actions that shouldn't if the service server is down.

I think we can close this issue/PR considering everyone may want to handle this differently. (we can easily recreate a bt_servie_node that doesn't throw an exception).

SteveMacenski, 2023-07-24:
Ideally, I'd like to not make you have to do that :( 

When you say "crash" it doesn't actually crash, right? You mean it throws the exception when is handled and rejects the BT construction? That's a problem if that timeout's exception isn't being caught somewhere to not literally _crash_. Is your problem the crashing server or the actual fact that we check there instead of at run-time of the BT? Maybe I've misinterpreted this. We should definitely fix any literal crash. 

If the exception is handled, it should create the failure + log which you're looking for on BT construction. Would that work?

BriceRenaudeau, 2023-07-24:
W

BriceRenaudeau, 2023-07-24:
You are right, it is my node that doesn't catch the thrown exception.
I can easily catch it and retry the BT creation, that would fix my issue.

But more generally, this exception at the creation forbids the creation of `bt_service_node` if the server is not started. That may be an issue for someone. It's still much better than an exception at loop time.

SteveMacenski, 2023-07-26:
Can you give me a plausible example situation when a server would not be running when we execute a BT (e.g. a navigation request) which has a dependency on it? 

I suppose we could make it an option in the bt service node to allow for a logging failure, but I'd make that default `false` unless someone really knew what they were doing.

BriceRenaudeau, 2023-07-27:
I can only think of situations where there is a fault that makes the server not available: The navigation is not started (when you start the robot for example). The server is a lifecycle node that is activated at runtime (like mapping). 

If the BT is created at the start, and not runtime like the bt_navigator, the BT creation will fail.

SteveMacenski, 2023-07-27:
OK, so what we could do:
- Have an input port for whether to allow a failure of the server on construction and continue (default off). That would let you select at BT design time which servers you might want to allow to fail (e.g. like mapping) based on your own system-level knowledge.
- If it fails on construction and/or as a general policy, look at the BT server implementation and either (1) adjust it to add the `wait_for_server` in the tick before sending requests or (2) sending requests, but verifying that our current error handling is sufficient to handle the situation that the server isn't alive / responsive in the tick. 

I think that should meet all our needs, no? Would that be something you'd be open to implementing? The Point 1 is trivial, but Point 2 requires actually testing in that failure mode condition I have a feeling you already know how to trigger :wink: Though, a unit test would be useful here as well since services has caused us all collectively so much headache. That might be a good 'canary in the coalmine' unit test

SteveMacenski, 2023-09-11:
@BriceRenaudeau any movement here?

SteveMacenski, 2023-09-12:
From @BriceRenaudeau:

> It is not an issue anymore. We made is more robust to start the BT when everything is ready.

Sounds like the lifecycle manager ordering was wrong :wink: Yes, the BT navigator (e.g. orchestrator of all nodes) should be last to transition up so its child services are also available already. Closing.

closed 2023-09-12 by SteveMacenski