ros-navigation/navigation2 #3684: Need some delay after simulation is up before launching navigation ?

jeremyfix, 2023-07-08:
## Bug report

**Required Info:**

- Operating System:
 Ubuntu 20.04

- ROS2 Version:
 Foxy

- Version or commit hash:

navigation2 : ca482808a7a7c52ce01ae3c662dc2b980968fc16   (foxy-devel)
slam_toolbox: 4786e90c06a4dc6fa811c5057d4e88387fba3829  (foxy-devel)

- DDS implementation:

I do not know, I did not change anything, I think it is rmw_fastrtps  (got it from   ps aux | grep rtps)

#### Steps to reproduce issue

Here is the launch files I'm using 

***slam.launch.xml**
```
<launch>
	<arg name="nav" default="1"/>

	<set_env name="TURTLEBOT3_MODEL" value="burger"/>
	<include file="$(find-pkg-share turtlebot3_gazebo)/launch/turtlebot3_house.launch.py"/>

	<!-- Visualization -->
	<node pkg="rviz2" exec="rviz2" name="rviz2"  args="-d $(find-pkg-share slamnav_turtlebot)/config/rviz-slam.rviz"/>

       <!-- Nav2 navigation -->
        <group if="$(var nav)"> 
	<include file="$(find-pkg-share slamnav_turtlebot)/launch/nav.launch.xml"/>
        </group>
</launch>
```
**nav.launch.xml**
```
<launch>
    <!-- Nav2 navigation -->
    <include file="$(find-pkg-share nav2_bringup)/launch/navigation_launch.py"/>
</launch>

```

There are two conditions :

This one is failing : 
```
ros2 launch slamnav_turtlebot slam.launch.xml  nav:=1
```

This one is working :

```
ros2 launch slamnav_turtlebot slam.launch.xml  nav:=0
THEN
ros2 launch slamnav_turtlebot nav.launch.xml 
```

#### Expected behavior / Additional information

By failing, above, I mean, If I define a goal in rviz, the robot does not move. By working, it means the robot is movig. 

When it fails, I observe several things :

- the /goal_pose does not get a subscriber. While when it is working , I have bt_navigator subscribing it
- I see some bt_navigator errors in the log :

```
[bt_navigator-9] [INFO] [1688795265.440572578] [bt_navigator]: Configuring
[bt_navigator-9] [ERROR] [1688795265.478983144] []: Caught exception in callback for transition 10
[bt_navigator-9] [ERROR] [1688795265.479011936] []: Original error: Could not load library: libnav2_rotate_action_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-9] [WARN] [1688795265.479060841] []: Error occurred while doing error handling.
[bt_navigator-9] [FATAL] [1688795265.479074835] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented

```

I do not know if this is expected or not. It seems navigation2 requires some elements to be up on startup. But what puzzles me if the error on the unability to load the library `libnav2_rotate_action_bt_node.so`. I do not understand why this should be raised in one case, not the other. 

For now, I will be looking for a way to inject a delay in the nav2 launch startup or to start them separately. 

Thank you for your help.

SteveMacenski, 2023-07-08:
We don’t manage or maintain those packages. Take it up with those project or maintainers

closed 2023-07-08 by SteveMacenski

jeremyfix, 2023-07-10:
@SteveMacenski  I'm not sure to understand your reply. From my point of view, this is more related to the bt_navigator, you manage, that to slam_toolbox .

Why did you say you do not maintain those packages ?