ros-navigation/navigation2 #3685: Requested navigation from navigate_to_pose while another navigator is processing, rejecting request.

osama-z-salah, 2023-07-10:
**Required Info:**

- Operating System:
  -Ubuntu 22.04.2 
- ROS2 Version:
  - Humble binaries

#### Steps to reproduce issue

I have created my own custom navigator where I want to call NavigateToPose like this :

<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <Sequence name="ClearingActions">
        <NavigateToPose goal="{goal}"/>
    </Sequence>
  </BehaviorTree>
</root>

#### Expected behavior

#### Actual behavior
I am getting the following error 
  [bt_navigator_nanogrind_navigator_rclcpp_node]: NavigateToPoseAction: goal not provided
  [component_container_isolated-1] [ERROR] [1688970636.764230849] [bt_navigator]: Requested navigation from navigate_to_pose while another navigator is processing, rejecting request.
  [component_container_isolated-1] [WARN] [1688970636.764295919] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.
  [component_container_isolated-1] [ERROR] [1688970636.832460742] [bt_navigator]: Goal failed
  [component_container_isolated-1] [WARN] [1688970636.832561925] [bt_navigator]: [nanogrind_navigator] [ActionServer] Aborting handle.

SteveMacenski, 2023-07-10:
What is the problem? 

The navigators are mutually exclusive so that you don’t try planning / controlling via multiple BTs at once so you dont end up with resource contention problems and behavioral whiplash.

osama-z-salah, 2023-07-10:
I don't want to planning / controlling visa multiple navigator, I just want to create a simple navigator to call the NavigateToPose action node so that I can execute such tree :

<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <Sequence name="ClearingActions">
        <NavigateToPose goal="pose: {header: {frame_id: map}, pose: {position: {x: 6.0, y: 2.0, z: 0.0}, orientation:{x: 0.0, y: 0.0, z: 0, w: 1.0000000}}}"/>
        <NavigateToPose goal="pose: {header: {frame_id: map}, pose: {position: {x: 6.0, y: 2.0, z: 0.0}, orientation:{x: 0.0, y: 0.0, z: 0, w: 1.0000000}}}"/>
        <NavigateToPose goal="pose: {header: {frame_id: map}, pose: {position: {x: 6.0, y: 2.0, z: 0.0}, orientation:{x: 0.0, y: 0.0, z: 0, w: 1.0000000}}}"/>
    </Sequence>
  </BehaviorTree>
</root>

osama-z-salah, 2023-07-10:
`<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <Sequence name="ClearingActions">
        <NavigateToPose goal="pose: {header: {frame_id: map}, pose: {position: {x: 6.0, y: 2.0, z: 0.0}, orientation:{x: 0.0, y: 0.0, z: 0, w: 1.0000000}}}"/>
        <NavigateToPose goal="pose: {header: {frame_id: map}, pose: {position: {x: 6.0, y: 2.0, z: 0.0}, orientation:{x: 0.0, y: 0.0, z: 0, w: 1.0000000}}}"/>
        <NavigateToPose goal="pose: {header: {frame_id: map}, pose: {position: {x: 6.0, y: 2.0, z: 0.0}, orientation:{x: 0.0, y: 0.0, z: 0, w: 1.0000000}}}"/>
    </Sequence>
  </BehaviorTree>
</root>
`

osama-z-salah, 2023-07-10:
`<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <Sequence name="ClearingActions">
        <NavigateToPose goal={goal}/>
    </Sequence>
  </BehaviorTree>
</root>`

SteveMacenski, 2023-07-10:
A navigator plugin may not call another navigator plugin. That is a clear antipattern.

osama-z-salah, 2023-07-10:
Then how can I excute a simple behavior three that has access to all the nav2 behaviours and and nodes without the need to create a navigator plugin

SteveMacenski, 2023-07-10:
See other BT plugins for examples. There's no reason to create a navigator that simply just calls _another_ navigator type. The navigator takes a navigation request and processes the input for the BT blackboard and interprets the BT completion state for the result and feedback.

closed 2023-07-10 by SteveMacenski

deenqtt, 2024-04-29:
advertise: Unable to import msg class NavigateToPoseGoal from package nav2_msgs. Caused by module 'nav2_msgs.msg' has no attribute 'NavigateToPoseGoal'

hi, I have an error problem like above, I use ros2, and the nav2d.min.js that I use, I got it from another website, which contains the servername move_base, but I changed it to use nav2_msgs, so where is the peak of the problem??