ros-navigation/navigation2 #3699: velocity smoother deadband velocity broken since 1.1.7

Aposhian, 2023-07-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
  - 1.1.8
- DDS implementation:
  - N/A

#### Steps to reproduce issue

```
ros2 run nav2_velocity_smoother velocity_smoother --ros-args -p 'deadband_velocity:=[0.25,0.0,0.0]'
```
In a separate terminal:
```
ros2 lifecycle set /velocity_smoother configure && ros2 lifecycle set /velocity_smoother activate
ros2 topic pub -r 20 /cmd_vel geometry_msgs/msg/Twist '{ linear: { x: 0.5 }}'
```

In a separate terminal:
```
ros2 topic echo /cmd_vel_smoothed --field linear.x
```

#### Expected behavior
It works on 1.1.6. `/cmd_vel_smoothed.linear.x` ramps up to full speed (0.5 in this case).

#### Actual behavior
On 1.1.8. `/cmd_vel_smoothed.linear.x` stays at 0 forever.

#### Additional information
I believe this was broken in 1.1.7 because the `last_cmd_` is set after the deadband velocities are applied. This means that the last_cmd is clamped to 0 forever.

Workaround is to disable deadband_velocity.

BriceRenaudeau, 2023-07-26:
Sorry, the fix is here soon :https://github.com/ros-planning/navigation2/pull/3690

SteveMacenski, 2023-07-26:
Merged! Backporting to iron / humble shortly

closed 2023-07-26 by SteveMacenski