ros-navigation/navigation2 #3702: Nav2 dynamic composition not starting

Jayden-F, 2023-07-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
  
- ROS2 Version:
  - Humble Binary
- Version or commit hash:
  - ros-humble-navigation2	1.1.8-2jammy.20230624.064440
- DDS implementation:
  - Cyclone DDS

#### Steps to reproduce issue

run the following command `ros2 launch nav2_bringup navigation_launch.py use_composition:=True`

#### Expected behavior
The Nav2 stack will start up the same as using 'ros2 launch nav2_bringup navigation_launch.py' 

#### Actual behavior
```
[INFO] [launch]: All log files can be found below /home/jayden/.ros/log/2023-07-19-13-25-16-671107-dev-pc-129453
[INFO] [launch]: Default logging verbosity is set to INFO
```
Then nothing further happen or is logged.

SteveMacenski, 2023-07-22:
I just tested on Humble binaries and I see no issues. Use the proper parent launch file `ros2 launch nav2_bringup tb3_simulation_launch.py`. Navigation is an internal launch file which requires stimulation to launch the full system which other launch files provide.

closed 2023-07-22 by SteveMacenski

Jayden-F, 2023-07-24:
Hello Steve, I appreciate you looking into my issue. I will elaborate further.

I am using the internal navigation launch file to control a real robot. The plan was to run the nav2 stack on a raspberry pi. Hence, my interest in running the nodes in a composition setup. 

~~~
# Copyright 2022 Clearpath Robotics, Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
#
# @author Roni Kreinin (rkreinin@clearpathrobotics.com)

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, GroupAction, IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration, PathJoinSubstitution

from launch_ros.actions import PushRosNamespace

ARGUMENTS = [
    DeclareLaunchArgument('use_sim_time', default_value='false',
                          choices=['true', 'false'],
                          description='Use sim time'),
]

def generate_launch_description():
    pkg_turtlebot4_navigation = get_package_share_directory('turtlebot4_navigation')
    pkg_nav2_bringup = get_package_share_directory('nav2_bringup')

    nav2_params_arg = DeclareLaunchArgument(
        'params_file',
        default_value=PathJoinSubstitution(
            [pkg_turtlebot4_navigation, 'config', 'nav2.yaml']),
        description='Nav2 parameters')

    namespace_arg = DeclareLaunchArgument(
                        'namespace',
                        default_value='',
                        description='Robot namespace')

    namespace = LaunchConfiguration('namespace')
    use_sim_time = LaunchConfiguration('use_sim_time')

    nav2 = GroupAction([
        PushRosNamespace(namespace),

        IncludeLaunchDescription(
            PythonLaunchDescriptionSource(
                PathJoinSubstitution(
                    [pkg_nav2_bringup, 'launch', 'navigation_launch.py'])),
            launch_arguments={'use_sim_time': use_sim_time,
                              'use_composition': 'True',
                              'params_file': LaunchConfiguration('params_file')}.items()),
    ])

    ld = LaunchDescription(ARGUMENTS)
    ld.add_action(nav2_params_arg)
    ld.add_action(namespace_arg)
    ld.add_action(nav2)
    return ld
~~~

Here is the launch file I am using, it is slightly changed from the original created by clearpath robotics. I set use_composition to True. Otherwise, all is the same. To give you a minimally reproducible example, the issue I am experiencing is identical to the issue experienced when running `ros2 launch nav2_bringup navigation_launch.py use_composition:=True`

While running with composition I receive the following output:
~~~
[INFO] [launch]: All log files can be found below /home/jayden/.ros/log/2023-07-19-13-25-16-671107-dev-pc-129453
[INFO] [launch]: Default logging verbosity is set to INFO
~~~

When running without composition all nodes start up and output.

[log.txt](https://github.com/ros-planning/navigation2/files/12141404/log.txt)

Thank you for your time.

Benjamin-Tan, 2023-11-16:
@Jayden-F one thing missing from your launch file is that you would need to launch a component container node with container_name = 'nav2_container' by default. The navigation.launch.py would only load the components onto the 'nav2_container' node.

e.g. `ros2 run rclcpp_components component_container_isolated --ros-args --remap __node:=nav2_container` in a terminal prior to launching the navigation.launch.py should work.