ros-navigation/navigation2 #3703: Unable to run controller and planner server.

Rakshit-bit, 2023-07-20:
## Bug report

**Required Info:**

- Operating System:
  -Ubuntu 22.04
- ROS2 Version:
  Humble
- Version or commit hash:
  Binary
- DDS implementation:
  - 

#### Should run the Nav2 stack for my custom robot

#### [INFO] [launch]: All log files can be found below /home/computing/.ros/log/2023-07-20-16-09-12-279420-computing-System-Product-Name-2360157
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_server-1]: process started with pid [2360302]
[INFO] [planner_server-2]: process started with pid [2360304]
[INFO] [behavior_server-3]: process started with pid [2360306]
[INFO] [bt_navigator-4]: process started with pid [2360308]
[INFO] [lifecycle_manager-5]: process started with pid [2360310]
[planner_server-2] [INFO] [1689865752.728494178] [planner_server]: 
[planner_server-2] 	planner_server lifecycle node launched. 
[planner_server-2] 	Waiting on external lifecycle transitions to activate
[planner_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-2] [INFO] [1689865752.731469204] [planner_server]: Creating
[controller_server-1] [INFO] [1689865752.745888187] [controller_server]: 
[controller_server-1] 	controller_server lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-2] [INFO] [1689865752.774896712] [global_costmap.global_costmap]: 
[planner_server-2] 	global_costmap lifecycle node launched. 
[planner_server-2] 	Waiting on external lifecycle transitions to activate
[planner_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-2] [INFO] [1689865752.776202704] [global_costmap.global_costmap]: Creating Costmap
[planner_server-2] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[planner_server-2]   what():  parameter 'footprint_padding' has invalid type: Wrong parameter type, parameter {footprint_padding} is of type {double}, setting it to {string} is not allowed.
[controller_server-1] [INFO] [1689865752.778712582] [controller_server]: Creating controller server
[bt_navigator-4] [INFO] [1689865752.794330776] [bt_navigator]: 
[bt_navigator-4] 	bt_navigator lifecycle node launched. 
[bt_navigator-4] 	Waiting on external lifecycle transitions to activate
[bt_navigator-4] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-4] [INFO] [1689865752.798140487] [bt_navigator]: Creating
[lifecycle_manager-5] [INFO] [1689865752.839605679] [lifecycle_manager_navigation]: Creating
[controller_server-1] [INFO] [1689865752.842530627] [local_costmap.local_costmap]: 
[controller_server-1] 	local_costmap lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-1] [INFO] [1689865752.843217947] [local_costmap.local_costmap]: Creating Costmap
[controller_server-1] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[controller_server-1]   what():  parameter 'footprint_padding' has invalid type: Wrong parameter type, parameter {footprint_padding} is of type {double}, setting it to {string} is not allowed.
[lifecycle_manager-5] [INFO] [1689865752.843694789] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[behavior_server-3] [INFO] [1689865753.126791280] [behavior_server]: 
[behavior_server-3] 	behavior_server lifecycle node launched. 
[behavior_server-3] 	Waiting on external lifecycle transitions to activate
[behavior_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[lifecycle_manager-5] [INFO] [1689865753.127232918] [lifecycle_manager_navigation]: Starting managed nodes bringup...
[lifecycle_manager-5] [INFO] [1689865753.127270114] [lifecycle_manager_navigation]: Configuring controller_server
[ERROR] [planner_server-2]: process has died [pid 2360304, exit code -6, cmd '/opt/ros/humble/lib/nav2_planner/planner_server --ros-args --log-level info --ros-args -r __node:=planner_server -p use_sim_time:=True --params-file /tmp/tmpj0n34nmh -r /tf:=tf -r /tf_static:=tf_static'].
[ERROR] [controller_server-1]: process has died [pid 2360302, exit code -6, cmd '/opt/ros/humble/lib/nav2_controller/controller_server --ros-args --log-level info --ros-args -p use_sim_time:=True --params-file /tmp/tmpt0r8nam6 -r /tf:=tf -r /tf_static:=tf_static'].

**Additional  information**

- Using ROS2 Humble with Gazebo garden.
- Slam_toolbox is launching and detecting custom LiDAR.

- **My Launch file**

`import os
[... 480 lines cut ...]
    local_costmap_topic: local_costmap/costmap_raw
    local_footprint_topic: local_costmap/published_footprint
    global_costmap_topic: global_costmap/costmap_raw
    global_footprint_topic: global_costmap/published_footprint
    cycle_frequency: 10.0
    behavior_plugins: ["spin", "backup", "wait"]
    spin:
      plugin: "nav2_behaviors/Spin"
    backup:
      plugin: "nav2_behaviors/BackUp"
    
    wait:
      plugin: "nav2_behaviors/Wait"
    
    #local_frame: odom
    global_frame: odom
    robot_base_frame: base_link
    transform_timeout: 0.1
    simulate_ahead_time: 2.0
    max_rotational_vel: 1.0
    min_rotational_vel: 0.4
    rotational_acc_lim: 3.2
robot_state_publisher:
  ros__parameters:
    use_sim_time: True`

SteveMacenski, 2023-07-21:
> what(): parameter 'footprint_padding' has invalid type: Wrong parameter type, parameter {footprint_padding} is of type {double}, setting it to {string} is not allowed.

this is pretty clear. This is being set to a wrong type in your config files. See our example files / configuration docs

closed 2023-07-21 by SteveMacenski

Rakshit-bit, 2023-07-21:
Thanks Steve for replying. However, I have already figured that out and tried changing it.  In the config file above, under global costmap and loacl costmap params, I have provided footprint_padding as a double (as mentioned in NAV2 configurations) and footprint in quotes.
But it is still showing the same error. Therefore I tried to comment out this parameter, just to check but still the same issue.

SteveMacenski, 2023-07-21:
Did you rebuild your workspace?

Rakshit-bit, 2023-07-21:
Yes, I rebuild everything from fresh and now that errors are gone. However, another one coming:

[planner_server-2] [INFO] [1689954564.608912539] [planner_server]: Even sized heuristic lookup table size set 20000.000000, increasing size by 1 to make odd
[planner_server-2] [INFO] [1689954564.609201602] [planner_server]: Destroying plugin GridBased of type SmacPlannerHybrid
[planner_server-2] [ERROR] [1689954564.609298650] []: Caught exception in callback for transition 10
[planner_server-2] [ERROR] [1689954564.609305299] []: Original error: std::bad_alloc
[planner_server-2] [WARN] [1689954564.609332796] []: Error occurred while doing error handling.
[planner_server-2] [FATAL] [1689954564.609341349] [planner_server]: Lifecycle node planner_server does not have error state implemented
[lifecycle_manager-5] [ERROR] [1689954564.609757607] [lifecycle_manager_navigation]: Failed to change state for node: planner_server
[lifecycle_manager-5] [ERROR] [1689954564.609784215] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.

Rakshit-bit, 2023-07-21:
I think this might be caused by different RMWs versions as I am having this as well:
[RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7420: open_and_lock_file failed -> Function open_port_internal

SteveMacenski, 2023-07-22:
You could get a traceback to be sure, but that seems plausible

Rakshit-bit, 2023-07-26:
I have updated the RMWs version. Now this is coming and previous one  still comes. Sometimes the previous is coming while some times this one.

[bt_navigator-4] [ERROR] [1690388832.138353661] []: Original error: Could not load library: libnav2_are_error_codes_active_condition_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-4] [WARN] [1690388832.138802216] []: Error occurred while doing error handling.
[bt_navigator-4] [FATAL] [1690388832.139147730] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-5] [ERROR] [1690388832.140413933] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-5] [ERROR] [1690388832.140445460] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.

Rakshit-bit, 2023-07-27:
One thing more, I am launching navigation_launch.py and slam_toolbox. I am not providing the map file because I want slam toolbox to create the map while navigating. I have looked the params files it is still showing the same error.

Also, when I tried to run the default launch file (bringup_launch.py) from the naviagtion2 repo, it is launching but it is using default params. How to change the default params with my params as I am using Smac Plannar.

[planner_server-2] [ERROR] [1690468820.202704927] []: Caught exception in callback for transition 10
[planner_server-2] [ERROR] [1690468820.202711283] []: Original error: std::bad_alloc
[planner_server-2] [WARN] [1690468820.202725293] []: Error occurred while doing error handling.
[planner_server-2] [FATAL] [1690468820.202733757] [planner_server]: Lifecycle node planner_server does not have error state implemented
[lifecycle_manager-5] [ERROR] [1690468820.203080836] [lifecycle_manager_navigation]: Failed to change state for node: planner_server
[lifecycle_manager-5] [ERROR] [1690468820.203105628] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.

neerja123-gupta, 2024-07-09:
@Rak-r , Hi I am having similar issue , were you able to solve it ? Thanks.