ros-navigation/navigation2 #3707: How to improve the consistency of adjacent frame decision results in path planning?

xianglunkai, 2023-07-23:
I think this is very important. Currently, navigation systems have inconsistent output paths when making path decisions, which is very detrimental to controller execution. My initial idea is to change the planning starting point to the projection of the previous planned path, or to add continuous constraints between adjacent paths; Alternatively, use trajectory trees for planning instead of the current trajectory chain

SteveMacenski, 2023-07-23:
Certainly all of which are possible. Choosing how and when you plan is one key thing you can control in the behavior tree. If you would like to use a planning method with some memory of previous paths, that’s also easy to customize using the planner plugin APIs. I’m sure such methods would also make good open source contributions!

SteveMacenski, 2023-07-28:
Happy to keep chatting on this, but I think my explanation above gives some clear direction on the BT how to address this concern that is probably best and doesn't require follow up here in terms of code changes or new features in nav2.

closed 2023-07-28 by SteveMacenski