ros-navigation/navigation2 #3712: Planner quantize paths in one direction and not the other

alexanderjyuen, 2023-07-26:
- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Humble
- DDS implementation:
  - CycloneDDS

I am experiencing a behavior where when I try to generate a path, using either the `navigate_through_poses_w_replanning_and_recovery.xml` or `navigate_to_pose_w_replanning_and_recovery.xml
` behavior trees, I get a plan that is quantized at 0.5m in the Y direction of map frame but not in the X direction of map frame. The image below shows an example of this, where the grids are set to 0.5m:

[image].

I've tested this using the ThetaStarPlanner, Smac2DPlanner, and HybridA and the same behavior occurs for al lthree planners. Below is my global costmap parameters:

```
global_costmap:
  global_costmap:
    ros__parameters:
      
      footprint: "[[-1.0, -1.4], [-1.0, 1.4], [4.0, 1.4], [4.0, -1.4]]"
      footprint_padding: 0.0
      transform_tolerance: 0.3
      
      update_frequency: 10.0
      publish_frequency: 1.0

      global_frame: utm
      robot_base_frame: base_link

      width: 700
      height: 700
      origin_x: -350.0
      origin_y: -350.0
      resolution: 0.1
      
      rolling_window: false
  
      track_unknown_space: false
      unknown_cost_value: 25

      # plugins: ["obstacle_layer", "inflation_layer"]
      plugins: ["static_layer", "obstacle_layer", "inflation_layer"]

      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: true
        enabled: false
        subscribe_to_updates: true
        transform_tolerance: 0.1
        map_topic: "/map"
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: True
        observation_sources: "front_scan_filtered front_scan" # add front_scan when unlocking costmaps
        front_scan_filtered:
          data_type: LaserScan
          topic: /front_lidar/scan
          marking: false # Set this to True to start using it to generate obstacles
          clearing: true # Set this to True to make sure it is also used for clearing
          inf_is_valid: false # Leave this as False because we don't want infinites to be costmaps! 
          observation_persistence: 1.0
          min_obstacle_height: -10.0
          max_obstacle_height: 15.0
          obstacle_min_range: 0.0
          obstacle_max_range: 18.0
          raytrace_min_range: 0.0 
          raytrace_max_range: 20.0 # With a bigger raytrace than obstacle range we should clear infinites
        # Uncomment below to turn on infinite clearing
        front_scan: 
          data_type: LaserScan
          topic: /front_lidar/scan
          marking: false # Set this to False, We don't want to mark from infinite
          clearing: true # Set tihs to True, we want to use infinites to clear! 
          inf_is_valid: true # We want to use infinites for clearing
          min_obstacle_height: 0.0
          max_obstacle_height: 15.0
          obstacle_min_range: 0.0
          obstacle_max_range: 18.0
          raytrace_min_range: 0.0 
          raytrace_max_range: 20.0

      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 0.3
        inflation_radius: 0.01
        
      always_send_full_costmap: True
```
As shown above I have a global costmap resolution of 0.1, in the image provide it looks like in one direction the points are at spaced at about 0.1m (5 to a grid) but in the other direction it snaps to the grid at 0.5m. 

Has anyone seen this behavior and is there a known solution?

SteveMacenski, 2023-07-26:
As I mentioned in our Slack thread, I have no idea why you're seeing this and this isn't consistent from what I see on my side  (example) 
[image]

Given how NavFn/Smac/Theta* all operate differently with no code being shared, it makes me think it might be something you're post processing with -- or potentially I suppose something to do with your costmap configs, but I'm kind of doubting that. What algorithm are you showing in your example? There seems to be quite a few weird and overlapping frames on the left side

alexanderjyuen, 2023-07-27:
This happened for NavFn/Smac/Theta* with and without Smoother. We did some testing today and found something interesting. Looks like when we use `utm` instead of `map` for `global_frame` in `global_costmap` is when we get this situation, when we switch back to `map`, paths are smooth again. I'm not sure why this would effect it but could you give it a whirl and see if you get the same strange behavior

SteveMacenski, 2023-07-27:
That is interesting. Are you visualizing the plans in the utm frame (that the planners are using when this is happening) or the map frame when you see this distortion? Does it appear in both reference frames in rviz?

I don't have a gps enabled simulation model / environment to be able to test this easily unfortunately :( That's been something I've been trying to get contributions from GPS users for since... 2020. I can make something, but I'd feel alot more confident in it if it had the stamp of approval from an actual GPS user since I am not. 

Is it possible that the UTM coordinates aren't actually spatially consistent between the X and Y axes? While I couldn't find from about ~15 minutes of research on UTM an explicit mention of this being true or false, the graphics I found (e.x. https://www.maptools.com/tutorials/utm/details) heavily imply such a thing is true due to the curvature of the earth. The planners are searching differently, but all commonly have regular increments that relate to the resolution of the costmap - which assumes spatial consistency between the axes. So if the costmap is in a coordinate scheme which doesn't have `1 cell in X == 1 cell in Y`, it doesn't know anything about that. Its searching in the cell-space coordinate system. 

With that said, then if you visualized in the map frame, I would expect you to see that distortion. That distortion isn't necessarily problematic in it of itself as long as you're using a holonomic search-based planner (.e.g. theta* or navfn) since it doesn't really matter about spatial consistency. For Hybrid-A* and State Lattice however which are feasible, you would need that spatial consistency for the feasibility guarantees to be possible.  

While I wouldn't say theoretically that feasible planners couldn't be made to work on a non-spatially-consistent coordinate system, I would say in the practical sense it would be a substantial amount of complexity and difficulty that I don't think there's any reason to introduce for quite literally any application of ground-based mobility. Un-distorting that into some virtual but consistent coordinate system (e.g. `map` or something else, doesn't really matter) is a trivial conversation of coordinate systems and makes everything _far_ easier for feasibility guarantees from the high level planners to the low level controllers and safety systems.

SteveMacenski, 2023-08-17:
Conversation in person confirms this is the issue, closing. Resolution is to move to map frame, which the users did

closed 2023-08-17 by SteveMacenski