ros-navigation/navigation2 #3717: unable to use Smac planner based Nav2 params

Rakshit-bit, 2023-07-27:
## Bug report

**Required Info:**

- Operating System:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

Rakshit-bit, 2023-07-27:
Here the full nav2 params I am using.

```
amcl:
  ros__parameters:
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.1
    base_frame_id: "base_link"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 50.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 1000
    min_particles: 10000
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan
    map_topic: map
    set_initial_pose: false
    always_reset_initial_pose: false
    first_map_only: false
    initial_pose:
      x: 0.0
      y: 0.0
      z: 0.0
      yaw: 0.0
amcl_map_client:
  ros__parameters:
    use_sim_time: True

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: True
    
bt_navigator:
  ros__parameters:
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /model/podcar/odometry
    bt_loop_duration: 10
    default_nav_to_pose_bt_xml: will_be_handled_by_rewritten_yaml
    plugin_lib_names:
      - nav2_compute_path_to_pose_action_bt_node
      - nav2_compute_path_through_poses_action_bt_node
      - nav2_smooth_path_action_bt_node
[... 272 lines cut ...]
    local_footprint_topic: local_costmap/published_footprint
    global_costmap_topic: global_costmap/costmap_raw
    global_footprint_topic: global_costmap/published_footprint
    cycle_frequency: 10.0
    behavior_plugins: ["spin", "backup", "wait"]
    spin:
      plugin: "nav2_behaviors/Spin"
    backup:
      plugin: "nav2_behaviors/BackUp"
    
    wait:
      plugin: "nav2_behaviors/Wait"
    #local_frame: odom
    global_frame: odom
    robot_base_frame: base_link
    transform_timeout: 0.1
    simulate_ahead_time: 2.0
    max_rotational_vel: 1.0
    min_rotational_vel: 0.4
    rotational_acc_lim: 3.2
robot_state_publisher:
  ros__parameters:
    use_sim_time: True

```

Rakshit-bit, 2023-07-27:
This is my launch file. 

```
import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, SetEnvironmentVariable
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node
from nav2_common.launch import RewrittenYaml

def generate_launch_description():
    # Get the launch directory
    bringup_dir = get_package_share_directory('pod2_navigation')
    # nav2_params_path = os.path.join(
    #     get_package_share_directory('pod2_navigation'),
    #     'config/',
    #     'nav2_params.yaml')

    namespace = LaunchConfiguration('namespace')
    use_sim_time = LaunchConfiguration('use_sim_time')
    autostart = LaunchConfiguration('autostart')
    params_file = LaunchConfiguration('params_file')
    default_nav_to_pose_bt_xml = LaunchConfiguration('default_nav_to_pose_bt_xml')
    map_subscribe_transient_local = LaunchConfiguration('map_subscribe_transient_local')

    lifecycle_nodes = ['controller_server',
                       'planner_server',
                       'behavior_server',
                       'bt_navigator']

    # Map fully qualified names to relative ones so the node's namespace can be prepended.
    # In case of the transforms (tf), currently, there doesn't seem to be a better alternative
    # https://github.com/ros/geometry2/issues/32
    # https://github.com/ros/robot_state_publisher/pull/30
    # TODO(orduno) Substitute with `PushNodeRemapping`
    #              https://github.com/ros2/launch_ros/issues/56
    # remappings = [('/tf', 'tf'),
                #   ('/tf_static', 'tf_static')]

    # Create our own temporary YAML files that include substitutions
    param_substitutions = {
        'use_sim_time': use_sim_time,
        'autostart': autostart,
        # 'default_nav_to_pose_bt_xml': default_nav_to_pose_bt_xml,
        # 'map_subscribe_transient_local': map_subscribe_transient_local
        }

    configured_params = RewrittenYaml(
            source_file=params_file,
            root_key=namespace,
            param_rewrites=param_substitutions,
            convert_types=True)

    return LaunchDescription([
        # Set env var to print messages to stdout immediately
        SetEnvironmentVariable('RCUTILS_LOGGING_BUFFERED_STREAM', '1'),

        DeclareLaunchArgument(
            'namespace', default_value='',
            description='Top-level namespace'),

        DeclareLaunchArgument(
            'use_sim_time', default_value='true',
            description='Use simulation (Gazebo) clock if true'),

        DeclareLaunchArgument(
            'autostart', default_value='true',
            description='Automatically startup the nav2 stack'),

        DeclareLaunchArgument(
            'params_file',
            default_value=os.path.join(bringup_dir, 'config', 'nav2_params.yaml'),
            description='Full path to the ROS2 parameters file to use'),

        DeclareLaunchArgument(
            'default_nav_to_pose_bt_xml',
            default_value=os.path.join(bringup_dir, 'behavior_trees', 'default.xml'),
            description='Full path to the behavior tree xml file to use'),

        DeclareLaunchArgument(
            'map_subscribe_transient_local', default_value='true',
            description='Whether to set the map subscriber QoS to transient local'),

        Node(
            package='nav2_controller',
            executable='controller_server',
            output='screen',
            parameters=[configured_params],
            # remappings=remappings
            ),

        Node(
            package='nav2_planner',
            executable='planner_server',
            name='planner_server',
            output='screen',
            parameters=[configured_params],
            # remappings=remappings
            ),

        Node(
            package='nav2_behaviors',
            executable='behavior_server',
            name='behavior_server',
            output='screen',
            parameters=[configured_params],
            # remappings=remappings
            ),

        Node(
            package='nav2_bt_navigator',
            executable='bt_navigator',
            name='bt_navigator',
            output='screen',
            parameters=[configured_params],
            # remappings=remappings
            ),

        Node(
            package='nav2_lifecycle_manager',
            executable='lifecycle_manager',
            name='lifecycle_manager_navigation',
            output='screen',
            parameters=[{'use_sim_time': use_sim_time},
                        {'autostart': autostart},
                        {'node_names': lifecycle_nodes}]),

       
    ])
```

SteveMacenski, 2023-07-27:
Please also grab that traceback

SteveMacenski, 2023-07-27:
Off the top of my head, this resolution `resolution: 0.001` is kind of insane. Why do you need 1mm level costmap representations? That seems nonsensical. That lookup table generation for that low of a resolution would be 800000000 bytes, or 0.8GB of your RAM alone. You may quite literally have run out of memory for use.

Rakshit-bit, 2023-07-27:
Apologies for out of topic question.
Could tell how to copy this text from gdb, I have got the backtrace but its not copying.

Rakshit-bit, 2023-07-27:
This I know. However, there nothing in the terminal. Its the xtrem which is open and I cannot copy the content

SteveMacenski, 2023-07-27:
Send a screen shot I guess

SteveMacenski, 2023-07-27:
Did you compile with debug flags? It should give you a line somewhere which I don't see in those screen shots. Basically #14 in that traceback that is `??` is what is the useful information.

SteveMacenski, 2023-07-27:
You have to do that for the smac planner package, not your highest level package.

Rakshit-bit, 2023-07-27:
I have done this for the navigation package only which contains nav2 params file

Rakshit-bit, 2023-07-27:
In the params file?

Rakshit-bit, 2023-07-27:
Could you give a simple demo of it

SteveMacenski, 2023-07-27:
Look at your cmake above, add the -g in the Smac Planner version. This is straight forward stuff, check out the tutorial & your own code blob above. Compile it with debug flags.

Rakshit-bit, 2023-07-27:
Thanks a lot. I will do and will post the output

Rakshit-bit, 2023-07-27:
One more thing, if we look at thd screenshots, it is mentioning fastdds but I downloaded the cycloneDDS using

Sudo apt-get install ros-humble-cycloneDDS

And then changed it using export
I checked using 
ros2 doctor --report | grep middleware

It returned CycloneDDS

Is it strange

SteveMacenski, 2023-07-27:
That's not really relevant to this issue (probably).

Rakshit-bit, 2023-07-28:
Ok. Just want to give the possibe information which I have. Using Gazebo garden's Ackermann plugin for twist messages and ground truth odometry from Gazebo using ros_gz_bridge.

SteveMacenski, 2023-07-28:
Did you get a traceback? And/or did you try reducing the resolution to something more reasonable like 2.5-10cm?

Rakshit-bit, 2023-07-28:
I tried changing the resolution and it seems the planner server error is gone. But this is coming, I am attaching the backtrace as well.

[image]

Rakshit-bit, 2023-07-28:
> Exactly. You have to add `-g` to `nav2_smac_planner`.

I tried changing the resolution, planner server error is showing now and it seems other error this time. But this is coming, I am attaching the backtrace as well. Also, i don't have any specific package of nav2_smac_planner in my workspace. The package contains launch files, config for nav2 and slam-toolbox. 

[image]

[image]

SteveMacenski, 2023-07-28:
> planner server error is not showing now

sounds like the fix. Without you providing the traceback, that’s the best we can do. Closing.

closed 2023-07-28 by SteveMacenski

Rakshit-bit, 2023-07-28:
What this other message means? I have provided the bt-navgator params

Rakshit-bit, 2023-07-28:
I tried to look at this error previously but still couldn't figure out the root. I have provided the backtrace for this as well in the above screenshots.

lourencorcc, 2024-04-19:
Hey man, getting the same error here! Did you manage to fix it? Thanks

Rak-r, 2024-04-19:
I  believe just go through the parameters again, set with defaults and also copy the default one from humble branch not from the ros2 website. See if that works.@lourencorcc

lourencorcc, 2024-04-19:
Where on github?

Rak-r, 2024-04-20:
On this nav2 github page ->select the branch name as per your ROS2 version and go to nav2_smac_planner. Find the params there and try it out.@lourencorcc

phanhoc1902, 2025-06-13:
> Hey man, getting the same error here! Did you manage to fix it? Thanks

Hey man, getting the same error here! Did you manage to fix it? Thanks

Rak-r, 2025-06-13:
> On this nav2 github page ->select the branch name as per your ROS2 version and go to nav2_smac_planner. Find the params there and try it out. @phanhoc1902