ros-navigation/navigation2 #3721: Allow inflation_layer inheritance

BriceRenaudeau, 2023-07-28:
The `obstacles_critics` of MPPI and SMAC_planner are tightly related to the `nav2_costmap_2d::InflationLayer`.

https://github.com/ros-planning/navigation2/blob/2f1f9e4a225f912c4c1ae9821fb7eb31ac192c16/nav2_mppi_controller/src/critics/obstacles_critic.cpp#L52-L55

To use a custom version of the inflation layer with the MPPI/SMAC, the custom layer needs to inherit from the inflation layer.
But the `inflation_layer` isn't adapted to inheritance, there are private members and methods.

**Required Info:**
- Operating System:
   - Ubuntu 22.04
- ROS2 Version:
  - Iron
- Version or commit hash:
  - main
- DDS implementation:
  - cycloneDDS

#### Expected behavior
When using a child class of `inflation_layer`, all the methods should work.

#### Actual behavior
When using a child class of `inflation_layer`, the attributes are not shared, so some methods, like getCostScalingFactor(), don't return the real value.

#### Implementation considerations
We fixed this by removing the `private:` keyword in the `inflation_layer` header file.

https://github.com/ros-planning/navigation2/blob/2f1f9e4a225f912c4c1ae9821fb7eb31ac192c16/nav2_costmap_2d/include/nav2_costmap_2d/inflation_layer.hpp#L185-L191C9

SteveMacenski, 2023-07-28:
Perhaps just make private protected so derived classes can use them? Seems very reasonable to me!

SteveMacenski, 2023-07-28:
https://github.com/ros-planning/navigation2/pull/3722

closed 2023-07-28 by SteveMacenski