ros-navigation/navigation2 #3725: Nothing published on topic /global_costmap/costmap_updates

Juhyung-L, 2023-07-29:
I am very new to ROS2 and Nav2.

I am developing a program that needs the costmap. But instead of subscribing to the /global_costmap/costmap topic and getting the full costmap every time it is published, I want to get the full costmap just once then subscribe to the /global_costmap/costmap_updates topic and update the initial costmap. The benefit of this method is that you can keep the same pointer to the costmap for the entirety of the program instead of getting a new pointer to the full costmap everytime. I don't know if this method is better than just getting the full costmap, but I want to try it for the sake of learning. I got the idea from here https://github.com/robo-friends/m-explore-ros2/blob/main/explore/src/costmap_client.cpp

The problem is that nothing gets published to the /global_costmap/costmap_updates topic. When I run ros2 topic echo /global_costmap/costmap_updates, nothing gets printed. I also made a custom node that listens to the topic and print log messages every time the node received a message, but nothing got printed there as well. Additionally, I tried driving my robot around while echoing the topic to see if it only receives messages when the costmap is updated. That also did not work.

When I run ros2 node info /global_costmap/global_costmap, it says that the node publishes to the /global_costmap/costmap_updates topic.
[image]

Other than this problem, everything else seems to be working fine: I can echo /global_costmap/costmap and /global_costmap/costmap_raw and I can visualize the costmap updating properly in RVIZ.

I am using a launch file to launch navigation_launch.py from nav2 and online_async_launch.py from slam-toolbox.

Could it be that the variation of SLAM that I am using (online async) just doesn't publish to this topic? Maybe this is just supposed to work this way and I am clueless.

SteveMacenski, 2023-07-31:
Please read our configuration guide: https://navigation.ros.org/configuration/packages/configuring-costmaps.html#costmap2d-ros-parameters. The very first parameter on this page is probably why you're not seeing it; its set to true in your config file most likely so we send the full costmap every time instead of sending just the diff updates of a small patch that was updated.

SteveMacenski, 2023-08-01:
I just tested this - in the default config file we have `always_send_full_costmap` as `True` which sends the full costmap without using the updates API. I just set it to `false` and tested and I do see it being properly published. I think that's your only issue.

closed 2023-08-01 by SteveMacenski

Juhyung-L, 2023-08-01:
I'm able to receive costmap updates now. Thank you!