ros-navigation/navigation2 #3726: Boost packages are missing in cmake

felixf4xu, 2023-07-31:
## Bug report

**Required Info:**

- Operating System:
   Ubuntu 22.04
- ROS2 Version:
  - rolling
- Version or commit hash:
  - main branch 
- DDS implementation:
  - ros2 default

#### Steps to reproduce issue

Boost is installed by `apt`
 
Build `ompl` from source : https://github.com/ompl/ompl , main branch (using `colcon`)

Build this project from source 

#### Expected behavior

cmake should configure ok.

#### Actual behavior

cmake error:
>--- stderr: nav2_smac_planner
CMake Error at CMakeLists.txt:75 (add_library):
  Target "nav2_smac_planner" links to target "Boost::serialization" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

#### Additional information

this project uses `ompl` which exports Boost as dependency at https://github.com/ompl/ompl/blob/main/omplConfig.cmake.in#L43
but `ompl` forgets to `find_dependency` of Boost.

So when "nav2_smac_planner" links to `ompl`, it knows it links to Boost but Boost is not found by find_package, which causes the error as above.

Strictly speaking this seems to be an upstream issue, but for the time being, to remove the cmake error, I need to add this line:

    find_package(Boost COMPONENTS serialization filesystem system)

SteveMacenski, 2023-07-31:
Can you submit this bug / the simple PR to OMPL to fix it there?

felixf4xu, 2023-08-01:
sure I'm planning to.

Just to confirm, does this issue exist in `navigation2`? I'm just curious, anyone else has this issue or just me?

SteveMacenski, 2023-08-01:
I'm trusting in what you're saying that OMPL needs to properly export its dependencies :-) We don't use boost in Smac outside of OMPL -- especially not the serialization module, so I think its a good assessment. Given the build time of OMPL and the extremely simple utilities of OMPL we actually use here, I rarely build OMPL from source to have first-hand experience. I always just use the distribution provided binaries! I did however just test and I do see this behavior when I try to build OMPL and smac planner in the same build call. We have had OMPL work through in source builds for a long time with OMPL in an underlay workspace that we sourced.

It builds using raw cmake with a couple of lines for helping with ament for underlay workspaces, so it wouldn't surprise me if there was just a little gap for building and working with OMPL in a workspace with packages that use it directly.

closed 2023-08-01 by SteveMacenski

felixf4xu, 2023-08-02:
thanks.

Just for anyone's information, https://github.com/ompl/ompl/pull/1085 is created in the `ompl` repo.