ros-navigation/navigation2 #3737: Nav2 BT navigator issue

Rakshit-bit, 2023-08-03:
ROS2 Humble
Cyclone DDS

Expected behaviour- should run NAV2. I am using default plugins of bt navigator but still I am not able to figure out the issue.

Actual- bt navigator node not launching.

bt_navigator-4] [ERROR] [1690388832.138353661] []: Original error: Could not load library: libnav2_are_error_codes_active_condition_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-4] [WARN] [1690388832.138802216] []: Error occurred while doing error handling.
[bt_navigator-4] [FATAL] [1690388832.139147730] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-5] [ERROR] [1690388832.140413933] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-5] [ERROR] [1690388832.140445460] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.

SteveMacenski, 2023-08-03:
> libnav2_are_error_codes_active_condition_bt_node

This isn't a BT node in Humble, you're probably using a parameter file you took from the `main` branch instead of appropriately from the `humble` branch. The `main` branch has stuff `humble` doesn't.

Rakshit-bit, 2023-08-07:
After I made changes it seems that issue is removed but now this another one coming. I have provided use_sim_true everywhere but I don't why this is coming.

I have set the fixed frame to map and tried setting and then Nav2goal in rviz. Seems it is recieving the commands but then this appearing.

**RVIZ terminal**

`[INFO] [1691402475.158808423] [rviz]: Message Filter dropping message: frame 'odom' at time 203715.705 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[INFO] [1691402491.001204592] [rviz2]: Trying to create a map of size 497 x 690 using 1 swatches
[INFO] [1691403291.902107796] [rviz2]: Setting estimate pose: Frame:odom, Position(-0.821431, 1.34686, 0), Orientation(0, 0, 0.018375, 0.999831) = Angle: 0.0367521
Start navigation
[INFO] [1691403309.404173685] [_]: NavigateToPose will be called using the BT Navigator's default behavior tree.
[INFO] [1691403362.477397942] [rviz2]: Setting estimate pose: Frame:odom, Position(-2.79619, 1.52737, 0), Orientation(0, 0, -0.04183, 0.999125) = Angle: -0.0836845
[INFO] [1691403379.841732149] [rviz2]: Setting estimate pose: Frame:map, Position(-1.91526, 1.05117, 0), Orientation(0, 0, 0.0183008, 0.999833) = Angle: 0.0366036
Start navigation
[INFO] [1691403394.445131898] [_]: NavigateToPose will be called using the BT Navigator's default behavior tree.
`

**NAV2 Terminal**

`[lifecycle_manager-5] [INFO] [1691402339.134456682] [lifecycle_manager_navigation]: Creating bond timer...
[bt_navigator-4] [INFO] [1691403309.406489630] [bt_navigator]: Begin navigating from current location to (-0.14, 0.89)
[controller_server-1] [INFO] [1691403309.427583877] [controller_server]: Received a goal, begin computing control effort.
[controller_server-1] [WARN] [1691403309.427635044] [controller_server]: No goal checker was specified in parameter 'current_goal_checker'. Server will use only plugin loaded goal_checker . This warning will appear once.
[controller_server-1] [INFO] [1691403310.477902490] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403311.527901091] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403312.527904601] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403313.577903805] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403314.577903308] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403315.627901908] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403316.627899250] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403317.678021554] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403318.677899074] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403319.727899434] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403320.727953078] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403321.777901604] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403322.777903251] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403323.827966203] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403324.877900784] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403325.878036082] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403326.927905560] [controller_server]: Passing new path to controller.
[controller_server-1] [ERROR] [1691403327.777962110] [controller_server]: Failed to make progress
[controller_server-1] [WARN] [1691403327.778007259] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[controller_server-1] [INFO] [1691403327.796813018] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[controller_server-1] [INFO] [1691403327.797815007] [controller_server]: Received a goal, begin computing control effort.
[controller_server-1] [INFO] [1691403328.298091881] [controller_server]: Passing new path to controller.
[controller_server-1] [WARN] [1691403328.302328468] [controller_server]: Control loop missed its desired rate of 20.0000Hz
[controller_server-1] [INFO] [1691403328.999699842] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403330.002393869] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403331.002394935] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403332.052529859] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403333.102393086] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403334.102410978] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403335.152452918] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403336.154618877] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403337.202393806] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1691403338.202399498] [controller_server]: Passing new path to controller.
[... cut ...]
[planner_server-2] [WARN] [1691403412.568870234] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[behavior_server-3] [INFO] [1691403412.587503545] [behavior_server]: Running backup
[behavior_server-3] [WARN] [1691403422.587650930] [behavior_server]: Exceeded time allowance before reaching the DriveOnHeading goal - Exiting DriveOnHeading
[behavior_server-3] [WARN] [1691403422.587701511] [behavior_server]: backup failed
[behavior_server-3] [WARN] [1691403422.587716740] [behavior_server]: [backup] [ActionServer] Aborting handle.
`

Rakshit-bit, 2023-08-07:
I have looked at different topics and the time is sim time, then from where this error is coming from?

Rakshit-bit, 2023-08-07:
I tried changing the params and it seems the issue is no longer there. Thanks a lot.

Cristian-wp, 2023-11-24:
Hi @Rak-r I have your exact same problem:
```
lifecycle_manager-14] [INFO] [1700820547.300933970] [lifecycle_manager_navigation]: Starting managed nodes bringup...
[lifecycle_manager-14] [INFO] [1700820547.301005364] [lifecycle_manager_navigation]: Configuring controller_server
[controller_server-7] [WARN] [1700820547.301310911] [rcl_lifecycle]: No transition matching 1 found for current state inactive
[controller_server-7] [ERROR] [1700820547.301329534] [controller_server]: Unable to start transition 1 from current state inactive: Transition is not registered., at ./src/rcl_lifecycle.c:355
[lifecycle_manager-14] [INFO] [1700820547.301832985] [lifecycle_manager_navigation]: Configuring smoother_server
[smoother_server-8] [WARN] [1700820547.302026701] [rcl_lifecycle]: No transition matching 1 found for current state inactive
[smoother_server-8] [ERROR] [1700820547.302045041] [smoother_server]: Unable to start transition 1 from current state inactive: Transition is not registered., at ./src/rcl_lifecycle.c:355
[lifecycle_manager-14] [INFO] [1700820547.302364218] [lifecycle_manager_navigation]: Configuring planner_server
[planner_server-9] [WARN] [1700820547.302519899] [rcl_lifecycle]: No transition matching 1 found for current state inactive
[planner_server-9] [ERROR] [1700820547.302537385] [planner_server]: Unable to start transition 1 from current state inactive: Transition is not registered., at ./src/rcl_lifecycle.c:355
[lifecycle_manager-14] [INFO] [1700820547.302776746] [lifecycle_manager_navigation]: Configuring behavior_server
[behavior_server-10] [WARN] [1700820547.302911425] [rcl_lifecycle]: No transition matching 1 found for current state inactive
[behavior_server-10] [ERROR] [1700820547.302927600] [behavior_server]: Unable to start transition 1 from current state inactive: Transition is not registered., at ./src/rcl_lifecycle.c:355
[lifecycle_manager-14] [INFO] [1700820547.303191023] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-11] [INFO] [1700820547.303381089] [bt_navigator]: Configuring
[bt_navigator-11] [ERROR] [1700820547.337725521] [bt_navigator]: Caught exception in callback for transition 10
[bt_navigator-11] [ERROR] [1700820547.337762428] [bt_navigator]: Original error: Could not load library: libnav2_are_error_codes_active_condition_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-11] [WARN] [1700820547.337784641] [bt_navigator]: Error occurred while doing error handling.
[bt_navigator-11] [ERROR] [1700820547.337798562] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-14] [ERROR] [1700820547.338181273] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-14] [ERROR] [1700820547.338208415] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
[lifecycle_manager-14] [INFO] [1700820593.618853682] [lifecycle_manager_navigation]: Resetting managed nodes...
[lifecycle_manager-14] [INFO] [1700820593.618894026] [lifecycle_manager_navigation]: Deactivating velocity_smoother
[velocity_smoother-13] [WARN] [1700820593.619228674] [rcl_lifecycle]: No transition matching 4 found for current state unconfigured
```
Can you please link the branch you used?
I have try to use these ones:

1. git clone --recursive https://github.com/ros-planning/navigation2.git -b humble
2. git clone --recursive https://github.com/ros-planning/navigation2.git -b 1.1.9

But nothing has changed.

Rak-r, 2023-11-24:
In the main NAV2 repo just switch to humble branch and look for nav2_params.yaml in nav2_bringup package. I just took that configuration for BT navigator and it seemed to solved the issue.

Cristian-wp, 2023-11-24:
I will try it now and let you know.
Sorry for the dummy question but I do not understand this thing, if I clone the report as I do (git clone --recursive https://github.com/ros-planning/navigation2.git -b humble) it should be already correct right?

Rak-r, 2023-11-24:
I am not sure about because I haven't done it in that way. However as long as you are cloning the humbe branch it should work. Are you having NAV2 installed through binary?

Cristian-wp, 2023-11-24:
In both way, I have install with apt-get and clone the source package in my workspace.
but all two are humble version, I am sure 100%.
I get that error doing the [Navigating Using GPS Localization tutorial](https://navigation.ros.org/tutorials/docs/navigation2_with_gps.html#setup-navigation-system) testing the setup of the navigation system with the command:
`ros2 launch nav2_gps_waypoint_follower_demo gps_waypoint_follower.launch.py use_rviz:=True
` and sending goal to the robot with RViz

Vasishthbhatt, 2023-11-27:
> In both way, I have install with apt-get and clone the source package in my workspace. but all two are humble version, I am sure 100%. I get that error doing the [Navigating Using GPS Localization [...]

hello sir but i faced the same issue and solved it can you please share your colcon build output? because my problem was connected  to nav2_sms_behavior  package after removing that package my gps package was built correctly.

Cristian-wp, 2023-11-27:
Hi @Vasishthbhatt I had the same problem of yours, but in my case I was missing some dependancy.
I resolve installing them

ZaBy10, 2023-12-01:
> In both way, I have install with apt-get and clone the source package in my workspace. but all two are humble version, I am sure 100%. I get that error doing the [Navigating Using GPS Localization [...]

Hii @Cristian-wp , I have the same problem and could not find any solution to this........Could you provide me the entire solution to this,please it is urgent!!!

Thanks in advance....

ZaBy10, 2023-12-01:
Hii @Cristian-wp  , Will  the Navigating Using GPS Localization tutorial work on ros2 foxy??

Cristian-wp, 2023-12-01:
@ZaBy10 no, as you can read [here](https://github.com/ros-planning/navigation2/pull/2814#issuecomment-1760312685) only Iron or newer

ZaBy10, 2023-12-01:
@Cristian-wp ,Thanks for taking your time and helping ,Aprreciate it a lot.....

Do you think is there any other way to Navigate using gps localisation on ros2 humble or ros2 foxxy