ros-navigation/navigation2 #3741: Extrapolation Error for base_link to map transform when attempting to navigate

alexanderjyuen, 2023-08-04:
- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - humble
- DDS implementation:
  - CycloneDDS

I am currently getting an issue where when I send a goal and the robot executes the `navigate_to_pose_w_replanning_and_recovery.xml` behavior tree, the navigation stack will plan for a little bit and fail after a few seconds stating  that there is an extrapolation error going from base_link to map:

```
[bt_navigator-11] [INFO] [1691190455.879684352] [bt_navigator]: Begin navigating from current location to (32.05, 0.66)
[controller_server-7] [INFO] [1691190455.902021074] [controller_server]: Received a goal, begin computing control effort.
[controller_server-7] [WARN] [1691190455.902143493] [controller_server]: No goal checker was specified in parameter 'current_goal_checker'. Server will use only plugin loaded goal_checker . This warning will appear once.
[controller_server-7] [INFO] [1691190456.202368477] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190456.402344674] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190456.602343989] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190456.902363239] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190457.102360323] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190457.302356449] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190457.502344812] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190457.802352587] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190458.002359233] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190458.202346588] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190458.502353400] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190458.702350913] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190458.902364877] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190459.202360847] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190459.402344543] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190459.602350023] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190459.802350522] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190460.102342919] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190460.302358895] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190460.502361328] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190460.802409924] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190461.002351572] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190461.202346480] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190461.502369100] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190461.702352011] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190461.902365604] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190462.102344006] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190462.402346614] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190462.602355085] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190462.802356741] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190463.102346333] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190463.302399887] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190463.502343559] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190463.802374738] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190464.002364775] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190464.202352729] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190464.402341106] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190464.702344989] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190464.902370424] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190465.102360291] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190465.402376914] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190465.602354681] [controller_server]: Passing new path to controller.
[controller_server-7] [INFO] [1691190465.802349595] [controller_server]: Passing new path to controller.
[planner_server-9] [ERROR] [1691190470.997915737] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the past.  Requested time 1691190455.798790 but the earliest data is at time 1691190460.996669, when looking up transform from frame [base_link] to frame [map]
[planner_server-9] 
[planner_server-9] [WARN] [1691190470.998032704] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-9] [WARN] [1691190470.998157375] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-9] [INFO] [1691190471.010150866] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[controller_server-7] [INFO] [1691190471.102362349] [controller_server]: Passing new path to controller.
[controller_server-7] [ERROR] [1691190471.103126134] [controller_server]: Invalid path, Path is empty.
```
In our setup we have a global EKF publishing that base_link to map transform at 50Hz and the local EKF publishing the base_link to odom transform at 60Hz, with both EKFs being `robot_localization`. So it seems strange that we would be without a base_link to map transform for more than 5 seconds. The pose is sent to Navigation 2 via RVIZ and the parameter_bridge by opening up rviz with `rosrun rviz rviz /move_base_simple/goal:=/goal_pose`. 

I've also done a `ros2 topic hz` on the outputs of `navsat`, `global EKF,` and `local EKF`, to check if nothing is skipping a beat and it looks like they are publishing at the specified rates

Has anyone experienced this issue before? The following are what I believe to be relevant configurations:

[... 152 lines cut ...]

      map_frame: map
      odom_frame: odom
      base_link_frame: base_link
      world_frame: map

```
**Local EKF Settings**
```
odom_filter_node:
    ros__parameters:
      # For parameter descriptions, please refer to the template parameter files for each node.
      frequency: 60.0
      sensor_timeout: 0.1
      two_d_mode: true
      transform_time_offset: 0.0
      transform_timeout: 0.0
      print_diagnostics: true
      debug: false

      map_frame: map
      odom_frame: odom
      base_link_frame: base_link
      world_frame: odom
```

SteveMacenski, 2023-08-05:
> So it seems strange that we would be without a base_link to map transform for more than 5 seconds.

It seems to be saying the exact opposite; `Requested time 1691190455.798790 but the earliest data is at time 1691190460.996669` -- it wants something from the past which is not available because the earliest data in its cache is 5 seconds after the requested time. 

> In our setup we have a global EKF publishing that base_link to map transform at 50Hz and the local EKF publishing the base_link to odom transform at 60Hz, with both EKFs being robot_localization.

You... shouldn't have a global EKF publishing `base_link` to `map` **and** a local EKF publishing `base_link` to `odom`, I'm hoping that's just a typo, but if its not, that could be part of the problem. See [REP 105](https://www.ros.org/reps/rep-0105.html). A local state estimate system should be providing `odom` to `base_link` and the global localization system should be providing `map` to `odom`. If you're having a disjointed tree like that, its possible that that is the cause of your issues. 

I don't know anything about this `parameter_bridge` you mention, but if you're bridging stuff, I'd also be very suspicious and/or interested in looking into that bridge.

You can omit the Nav2 configs, there's nearly zero chance its something in the codebase causing this issue - it sounds like a system level issue either w.r.t. your bridging technologies, use of TF2, or I suppose also your filters and/or inputs to the filters.

alexanderjyuen, 2023-08-06:
> You... shouldn't have a global EKF publishing `base_link` to `map` **and** a local EKF publishing `base_link` to `odom`, I'm hoping that's just a typo, but if its not, that could be part of the [...]

Yes I mis-spoke, what you stated is corrected

> I don't know anything about this `parameter_bridge` you mention, but if you're bridging stuff, I'd also be very suspicious and/or interested in looking into that bridge.
>
> You can omit the Nav2 [...]

Also correct here, looks like when I'm not bridging the the goal pose from ROS1 Rviz to the Nav2 Stack I don't get this issue, closing this issue

closed 2023-08-06 by alexanderjyuen