ros-navigation/navigation2 #3744: Collision monitor docs, and default params misleading

nakai-omer, 2023-08-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - ROS2 Humble - binaries
- Version or commit hash:
  - ros-humble-navigation2 1.1.6-1jammy.20230426.153445
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
By default, collision monitor publishes to `cmd_vel`, but current nav2 stacks controller servers publish to `cmd_vel_nav` to allow the smoother node to work on the cmd before publishing it `cmd_vel`. So by using default params, and by working by the docs, we basically cancel out the smoother server. 

#### Expected behavior
Adding collision monitor node should integrate seamlessly into the nav2 pipeline.
 
#### Actual behavior
Current collision monitor default params and docs will cause an integration which circumvents the smoother server.

#### Additional information
Working on a PR

nakai-omer, 2023-08-07:
I have added two PRs one for docs and one for default collision monitor param.
https://github.com/ros-planning/navigation.ros.org/pull/454
https://github.com/ros-planning/navigation2/pull/3745

The only thing I am not so sure about is whether collision monitor's default `cmd_vel_out_topic` should be changed from `cmd_vel` to `cmd_vel_nav`.
I went with changing it, since I presume most users use nav2_bringup to launch the nav2 stack (or should at least do so). In nav2 bringup the controller node outputs into `cmd_vel_nav`. So to allow for integrating collision monitor without having to override `cmd_vel_out_topic` config, I have changed the default `cmd_vel_out_topic` config to `cmd_vel_nav`, just as nav2 bringup expects.

nakai-omer, 2023-08-07:
Another option is to integrate the collision monitor after the velocity smoother, if you think this is a better solution, I will update the PRs.

SteveMacenski, 2023-08-07:
I think the documentation makes it clear how to add it in the launch system using topic remaps, no? Ref: https://navigation.ros.org/tutorials/docs/using_collision_monitor.html. Anything here that should just be updated instead?

nakai-omer, 2023-08-08:
@SteveMacenski Currently in nav2_bringup the default flow is: controller -> cmd_vel_nav -> smoother -> cmd_vel.
Currently the tutorial will cause the flow to be: controller -> cmd_vel_raw -> collision monitor -> cmd_vel, leaving out the smoother.
The proposed change will cause the flow to be: controller -> cmd_vel_raw -> collision monitor -> cmd_vel_nav -> smoother -> cmd_vel.
Another possible easier change can be: controller -> cmd_vel_nav -> smoother -> cmd_vel_raw -> collision monitor -> cmd_vel.
I am just not sure if the smoother should also be after the collision monitor or not.

nakai-omer, 2023-08-09:
@SteveMacenski Got it, will update the PR asap

AlexeyMerzlyakov, 2023-08-10:
Hi, sorry for being late. @nakai-omer, thank you for the problem notice.

Yes, in the CM docs we are indeed using twist, going out from Controller and bypassing Velocity Smoother. That was for historical reasons, since CM and VS were developing in parallel. However, the nodes should be ordered in the same sequence as @SteveMacenski [noted](https://github.com/ros-planning/navigation2/issues/3744#issuecomment-1670232663) above: CM should be the last chain of this scheme going after VS, as it makes final safety decisions to stop/slowdown the robot.

Waiting for the PR-s update and then I'll ready to check them.

nakai-omer, 2023-08-10:
@AlexeyMerzlyakov @SteveMacenski What do you think should be the default flow, using the smoother or not? In other words, should I change the default collision monitor's "in" topic to `cmd_nav_smoothed`, or leave it at `cmd_vel_nav`?

SteveMacenski, 2023-08-10:
> What do you think should be the default flow, using the smoother or not? 

Using it! Since its in the default bringup now and its a really low-compute operation which can add guarantees.

The current default is `cmd_vel_raw`which doesn't appear to be used anywhere (but that's OK, its pretty much always going to be remapped anyway). Changing to something else would be fine, but note anyway that we often do remappings at launch time. No matter how this is done, there's going to need to be remapping for the controller server and optionally velocity smoother / collision monitor, which ever depends on the defaults we set. 

I'd say tentatively either `cmd_vel_nav` or `cmd_vel_smoothed` are fine options - as long as the docs are appropriately updated in the tutorial / configuration guides to point to that change. I lean towards `_nav` since that's "generic", even though default `_smoothed` is outputted. So... I don't have a strong preference.

Also, why did you close your PR? Should the docs PR as well or are you going to update that?

nakai-omer, 2023-08-16:
@SteveMacenski @AlexeyMerzlyakov I mistakenly closed the PR when I thought it should only be a configuration change. Anyways, I have updated both PRs to include the discussed changes. By default the collision monitor will listen on `cmd_vel_smoothed`. I have updated the docs to also state the needed changes in navigation_launch.py
https://github.com/ros-planning/navigation.ros.org/pull/454
https://github.com/ros-planning/navigation2/pull/3745

SteveMacenski, 2023-08-16:
Great! The biggest thing is to make sure its consistent and documented in all the appropriate places and tutorials updated that use the collision monitor / velocity smoother so that folks can have a smooth copy-paste experience

nakai-omer, 2023-09-03:
@SteveMacenski Sorry it took a little time. I have updated the documentation [PR](https://github.com/ros-planning/navigation.ros.org/pull/454) with the requested changes.

SteveMacenski, 2023-09-05:
Looked over it, seems good to me now for the docs PR. Please make sure to follow up on the code change PR. There are a couple of quick things that are missing there to be merge-able.