ros-navigation/navigation2 #3746: Not able to navigate the robot

Rakshit-bit, 2023-08-07:
## Bug report

**Required Info:**

- Operating System:
 Ubuntu 22.04
- ROS2 Version:
 ROS2 Humble Binary

- DDS implementation:
  - CycloneDDS

#### Expected behavior 

Should run NAV2

#### Actual behavior

**The launch file seems ok because the last output I am getting is
[lifecycle_manager-5] [INFO] [1691409212.195861492] [lifecycle_manager_navigation]: Creating bond timer...

I have provided use_sim_time true everywhere but I don't why it is coming.**

`[bt_navigator-4] [INFO] [1691424773.213829933] [bt_navigator]: Begin navigating from current location to (-3.35, 2.02)
[planner_server-2] [ERROR] [1691424774.006121032] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the future.  Requested time 1691424773.212782 but the latest data is at time 3617.600000, when looking up transform from frame [map] to frame [odom]
[planner_server-2] 
[planner_server-2] [WARN] [1691424774.006158189] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-2] [WARN] [1691424774.006191148] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1691424774.024193653] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-2] [ERROR] [1691424774.864641914] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the future.  Requested time 1691424773.212782 but the latest data is at time 3618.100000, when looking up transform from frame [map] to frame [odom]
[planner_server-2] 
[planner_server-2] [WARN] [1691424774.864672393] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-2] [WARN] [1691424774.864695616] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[controller_server-1] [INFO] [1691424774.884121835] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[planner_server-2] [INFO] [1691424774.884386162] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-2] [ERROR] [1691424775.871126413] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the future.  Requested time 1691424773.212782 but the latest data is at time 3618.700000, when looking up transform from frame [map] to frame [odom]
[planner_server-2] 
[planner_server-2] [WARN] [1691424775.871156354] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-2] [WARN] [1691424775.871178511] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1691424775.885008689] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-2] [ERROR] [1691424776.872278416] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the future.  Requested time 1691424773.212782 but the latest data is at time 3619.400000, when looking up transform from frame [map] to frame [odom]
[planner_server-2] 
[planner_server-2] [WARN] [1691424776.872311456] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-2] [WARN] [1691424776.872334640] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[behavior_server-3] [INFO] [1691424776.884494541] [behavior_server]: Running spin
[behavior_server-3] [INFO] [1691424776.884605657] [behavior_server]: Turning 1.57 for spin behavior.
[behavior_server-3] [WARN] [1691424786.884706643] [behavior_server]: Exceeded time allowance before reaching the Spin goal - Exiting Spin
[behavior_server-3] [WARN] [1691424786.884743251] [behavior_server]: spin failed
[behavior_server-3] [WARN] [1691424786.884756924] [behavior_server]: [spin] [ActionServer] Aborting handle.
[behavior_server-3] [INFO] [1691424786.904300441] [behavior_server]: Running wait
[behavior_server-3] [INFO] [1691424791.904446638] [behavior_server]: wait completed successfully
[planner_server-2] [ERROR] [1691424792.730907641] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the future.  Requested time 1691424773.212782 but the latest data is at time 3629.300000, when looking up transform from frame [map] to frame [odom]
[planner_server-2] 
[planner_server-2] [WARN] [1691424792.730940504] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-2] [WARN] [1691424792.730967524] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1691424792.744965117] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-2] [ERROR] [1691424793.884840781] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the future.  Requested time 1691424773.212782 but the latest data is at time 3630.100000, when looking up transform from frame [map] to frame [odom]
[planner_server-2] 
[planner_server-2] [WARN] [1691424793.884876311] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-2] [WARN] [1691424793.884901487] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[behavior_server-3] [INFO] [1691424793.904321399] [behavior_server]: Running backup
[behavior_server-3] [WARN] [1691424803.904854927] [behavior_server]: Exceeded time allowance before reaching the DriveOnHeading goal - Exiting DriveOnHeading
[behavior_server-3] [WARN] [1691424803.904900950] [behavior_server]: backup failed
[behavior_server-3] [WARN] [1691424803.904915892] [behavior_server]: [backup] [ActionServer] Aborting handle.
[controller_server-1] [INFO] [1691424803.924166316] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[planner_server-2] [INFO] [1691424803.925834499] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-2] [ERROR] [1691424804.953690738] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the future.  Requested time 1691424773.212782 but the latest data is at time 3636.800000, when looking up transform from frame [map] to frame [odom]
[planner_server-2] 
[planner_server-2] [WARN] [1691424804.953725100] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-2] [WARN] [1691424804.953756433] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[... 535 lines cut ...]

        Node(
            package='nav2_bt_navigator',
            executable='bt_navigator',
            name='bt_navigator',
            #prefix=['xterm -e gdb -ex run --args'],
            output='screen',
            parameters=[configured_params],
            remappings=remappings
            ),

        Node(
            package='nav2_lifecycle_manager',
            executable='lifecycle_manager',
            name='lifecycle_manager_navigation',
            #prefix=['xterm -e gdb -ex run --args'],
            output='screen',
            parameters=[{'use_sim_time': use_sim_time},
                        {'autostart': autostart},
                        {'node_names': lifecycle_nodes}]),

       
    ])`

Also, in rviz localization shows inactive. But I launched Slam_toolbox pror to launching NAV2. I want to create map and localize simulatneously, that's why I have not provided pre-build map. Do I have to provide pre-built map for slam as well? I am bit new to this, so that's why.

SteveMacenski, 2023-08-07:
> how to configure slam_toolbox to do both mapping and localization? without pre-built map?

Read the docs and on the basics of SLAM. Respectfully, there is a litany of documentation on this that I shouldn’t need to repeat individually 🙂.

Rakshit-bit, 2023-08-07:
> What is your actual problem? The ticket never explains what your concern or issue is, just a set of (long) longs and configs with a title saying ‘its not working’. Please provide a detailed description of what’s not working and what you’ve tried. 

After launching the launch file and hitting the 2DPoseEstimate its sets the estimate but whe I do Nav2Goal it throws the above provided error and robot do not move. 

I looked at the topics and nodes using ros2 param get bt_navigator use_sim_time n its True (same with other nodes) However, the error seems transforms are not getting published on using simulation time

SteveMacenski, 2023-08-07:
See our default launch files where sim time works fine as a launch configuration -- you might find your error by comparing them. Since our simulation launch files work properly to set sim time, its something to do with your config / launch files, not Nav2 related. As such with a user setup error, closing. Unfortunately, I don't have the cycles to personally debug every user's errors, but there are many clear examples in place of working systems you can base off of or reference.

SteveMacenski, 2023-08-07:
I have closed a large volume of tickets from you with this exact same comment - please do take this advice. We're not here to fix all your configuration or misunderstanding issues. There is a wide body of documentation at navigation.ros.org and in the community available that you should look into first before filing here (let alone trying to fix an issue yourself, knowing that the default system is working fine).

Please do try to stop and think and spend some time to fix your own issue before filing a ticket of this type

closed 2023-08-07 by SteveMacenski

Rakshit-bit, 2023-08-08:
> See our default launch files where sim time works fine as a launch configuration -- you might find your error by comparing them. Since our simulation launch files work properly to set sim time, its [...]

I have looked for similar errors and tried therr solutions but still the issue persists that's why I created the issue.

Combinatrix, 2024-03-15:
Hello

No clue if someone still cares.

Ran into the same issue.
"requested time 1691424773.212782 but the latest data is at time 3617.600000"
requested time is current time while latest data is sim_time. 
**Solution**: run **rviz** with "use_sim_time" set to true, not just your nodes...

Rak-r, 2024-03-16:
> I have closed a large volume of tickets from you with this exact same comment - please do take this advice. We're not here to fix all your configuration or misunderstanding issues. There is a wide body of documentation at navigation.ros.org and in the community available that you should look into first before filing here (let alone trying to fix an issue yourself, knowing that the default system is working fine).
> 
> 
> 
> Please do try to stop and think and spend some time to fix your own issue before filing a ticket of this type. Please don't abuse the issue tracker. If you have this many issues and need professional support, you're welcome to email us at info@opennav.org and we could discuss there. Though, most of your issues are really trivial solutions if you read the documentation and set things up // googled your issues. 

Yes. Otherwise I just made gazebo to work on wall time. Better to have all physical and simulation robot to work on same time.

Anfil033RC, 2024-09-26:
thanks @Combinatrix 
$ ros2 run rviz2 rviz2 --ros-args -p use_sim_time:=true
instead of just $ rviz2 worked for me