ros-navigation/navigation2 #3747: Nav2 Accuracy

osama-z-salah, 2023-08-08:
I am wondering what is the maximum accuracy that can be achieved using normal navigation, in other words to what extent I can improve the navigation accuracy by tuning and using high-quality hardware (wheel odometry and Lidar's)?
because I have seen a lot of robots using different approaches for precise docking they navigate until a point close to the docking stations, and then use the accurate final approach to get into the location exactly

SteveMacenski, 2023-08-08:
That's a difficult question to answer. The framework in it of itself has no theoretical bounds on accuracy. The question then becomes between (A) your robot's hardware, calibration, responsiveness etc, (B) the planning, control algorithms available theoretical accuracy, and (C) the specific tuning and behaviors you're requiring and how those impact accuracy along path tracking or in the final goal pose achievement. 

Since A is dependent on you as the user, and C is dependent on the behaviors you find adventitious to tune the algorithms (which then might impact various accuracy metrics) -- its really not a number I can spit out to you with any real level of confidence. The planning and control algorithms themselves (B) have no theoretical limitations creating inaccuracies in it of themselves. 

In "practical" terms, I know for a fact that within 0.125m, ~4.7 inches (or a 0.25m radius) is our standard default and that has no problems with it at all. I know as well that there are a couple of users that tune and use Nav2 with a requirement to be within 5cm (~2 inches) due to their requirements and have done so successfully for final pose achievement. 

I would fully expect even better is possible, but that would be really dependent on your Point (A) and (C) being exceptionally well setup and tuned. Localization accuracy would be your biggest hurtle, beyond the obvious odometry accuracy and motor responsiveness from the base. 

If trying to dock so you're not using global localization but a specific and live-detected object that you're trying to do something relative to, then far more accurate navigation is pretty easy to achieve in the odometric frame to make that happen. Many have done this, so this isn't exactly unknown territory. That point your sources of error are primarily (A) laser scanner, (B) odometry error and hardware responsiveness, and (C) whatever detector you have running. After that, tracking to get to a pose relative to that is pretty much a solved problem for docking, interfacing with another vehicle, etc.

osama-z-salah, 2023-08-08:
Thanks for your reply, that is answering my question

closed 2023-08-08 by SteveMacenski