ros-navigation/navigation2 #3762: controller_server crashed with calling /lifecycle_manager/manage_nodes service

AmmarAlbakri, 2023-08-17:
## Bug report

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble binary
- DDS implementation:
  - FastRTPS

#### Steps to reproduce the issue
- Launch the navigation stack 
- Call the `manage_nodes` service of the navigation lifecycle manager with a value of 3 -> Reset
- Call the `manage_nodes` service of the navigation lifecycle manager with a value of 0 -> Bringup

#### Expected behavior
The Navigation stack should transition into the active state with no problem

#### Actual behavior
`controller_server` crashes with the error:
`[controller_server-1] terminate called after throwing an instance of 'std::runtime_error'
[controller_server-1]   what():  Node '/local_costmap/local_costmap' has already been added to an executor.`

and the node is dead and must be restarted.

SteveMacenski, 2023-08-17:
Can you verify that it works in `main` and/or if you [apply this patch](https://github.com/ros-planning/navigation2/pull/3630/files) onto Humble that it works? 

I believe we have handled this but it just hasn't been released to Humble binaries yet as a relatively recent fix

SteveMacenski, 2023-08-18:
But anyhow this is the answer:

> I removed the nav2_mppi_controller and can confirm with [this patch](https://github.com/ros-planning/navigation2/pull/3630/files) it works on humble.

That patch is actually already on humble, its just not in the released binaries. If you build humble from source, you're good to go! It will be released into binary format next time there is a release. I ran a release on August 4th and it looks like that release is happening today / tomorrow https://discourse.ros.org/t/preparing-for-humble-sync-2023-08-18/32950 so you should have humble binaries with this fix by the latest end of this week.

SteveMacenski, 2023-08-19:
https://discourse.ros.org/t/new-packages-for-humble-hawksbill-2023-08-18/32971

Update now and it should be patched