ros-navigation/navigation2 #3769: Failed to transform initial pose in time (Lookup would require extrapolation into the future.

Rak-r, 2023-08-23:
## Bug report

**Required Info:**

- Operating System:
    - Ubuntu 22.04
- ROS2 Version:
  - ROS2 Humble desktop
- Version or commit hash:
  - binary installation
- DDS implementation:
  -CycloneDDS

**terminal log**

`[component_container_isolated-1] [INFO] [1692792734.270110837] [amcl]: initialPoseReceived
[component_container_isolated-1] [WARN] [1692792734.270208549] [amcl]: Failed to transform initial pose in time (Lookup would require extrapolation into the future.  Requested time 1692792734.270147 but the latest data is at time 21.420000, when looking up transform from frame [base_link] to frame [odom])
[component_container_isolated-1] [INFO] [1692792734.270224747] [amcl]: Setting pose (1692792734.270225): -1.130 4.947 -0.206
[component_container_isolated-1] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-1]          at line 93 in ./src/buffer_core.cpp
[component_container_isolated-1] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-1]          at line 93 in ./src/buffer_core.cpp
[component_container_isolated-1] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-1]          at line 93 in ./src/buffer_core.cpp
[component_container_isolated-1] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-1]          at line 93 in ./src/buffer_core.cpp
[component_container_isolated-1] [INFO] [1692792734.570301587] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Lookup would require extrapolation into the past.  Requested time 21.620000 but the earliest data is at time 22.500000, when looking up transform from frame [base_link] to frame [map]
[component_container_isolated-1] [INFO] [1692792735.070242067] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Lookup would require extrapolation into the past.  Requested time 22.000000 but the earliest data is at time 22.500000, when looking up transform from frame [base_link] to frame [map]
[component_container_isolated-1] [INFO] [1692792735.570252799] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Lookup would require extrapolation into the past.  Requested time 22.340000 but the earliest data is at time 22.500000, when looking up transform from frame [base_link] to frame [map]
[component_container_isolated-1] [INFO] [1692792736.070299366] [global_costmap.global_costmap]: start
[component_container_isolated-1] [INFO] [1692792737.121162644] [planner_server]: Activating plugin GridBased of type SmacPlannerHybrid
[component_container_isolated-1] [INFO] [1692792737.122428944] [planner_server]: Creating bond (planner_server) to lifecycle manager.
[component_container_isolated-1] [INFO] [1692792737.227553321] [lifecycle_manager_navigation]: Server planner_server connected with bond.
[component_container_isolated-1] [INFO] [1692792737.227596198] [lifecycle_manager_navigation]: Activating behavior_server
[component_container_isolated-1] [INFO] [1692792737.227859574] [behavior_server]: Activating
[component_container_isolated-1] [INFO] [1692792737.227881872] [behavior_server]: Activating spin
[component_container_isolated-1] [INFO] [1692792737.227901715] [behavior_server]: Activating backup
[component_container_isolated-1] [INFO] [1692792737.227913208] [behavior_server]: Activating wait
[component_container_isolated-1] [INFO] [1692792737.227925670] [behavior_server]: Creating bond (behavior_server) to lifecycle manager.
[component_container_isolated-1] [INFO] [1692792737.332385261] [lifecycle_manager_navigation]: Server behavior_server connected with bond.
[component_container_isolated-1] [INFO] [1692792737.332426300] [lifecycle_manager_navigation]: Activating bt_navigator
[component_container_isolated-1] [INFO] [1692792737.332614698] [bt_navigator]: Activating
[component_container_isolated-1] [INFO] [1692792737.404508297] [bt_navigator]: Creating bond (bt_navigator) to lifecycle manager.
[component_container_isolated-1] [INFO] [1692792737.509052499] [lifecycle_manager_navigation]: Server bt_navigator connected with bond.
[component_container_isolated-1] [INFO] [1692792737.509089712] [lifecycle_manager_navigation]: Activating waypoint_follower
[component_container_isolated-1] [INFO] [1692792737.509524831] [waypoint_follower]: Activating
[component_container_isolated-1] [INFO] [1692792737.509566366] [waypoint_follower]: Creating bond (waypoint_follower) to lifecycle manager.
[component_container_isolated-1] [INFO] [1692792737.615100946] [lifecycle_manager_navigation]: Server waypoint_follower connected with bond.
[component_container_isolated-1] [INFO] [1692792737.615138690] [lifecycle_manager_navigation]: Activating velocity_smoother
[component_container_isolated-1] [INFO] [1692792737.615414054] [velocity_smoother]: Activating
[component_container_isolated-1] [INFO] [1692792737.615456755] [velocity_smoother]: Creating bond (velocity_smoother) to lifecycle manager.
[component_container_isolated-1] [INFO] [1692792737.720938414] [lifecycle_manager_navigation]: Server velocity_smoother connected with bond.
[component_container_isolated-1] [INFO] [1692792737.720974572] [lifecycle_manager_navigation]: Managed nodes are active
[component_container_isolated-1] [INFO] [1692792737.721008458] [lifecycle_manager_navigation]: Creating bond timer...`

I checked my launch file and it is correctly launching the nodes from the default nav2_bringup.
When providing the pre built map for AMCL and setting the initial pose , the robot is expected to navigate.
After launching all the nodes and setting the intial pose through rviz, it is givng error - 
`[amcl]: Failed to transform initial pose in time (Lookup would require extrapolation into the future.  Requested time 1692792734.270147 but the latest data is at time 21.420000, when looking up transform from frame [base_link] to frame [odom])`

The similar error is then given by global and local costmaps. It seems that something behind is using ros time which is expecting transforms. However, I checked every node is using sim time parameter set to true. This seems a similar issue to [https://github.com/ros-planning/navigation2/issues/3586] and I added `use-sim-time` argument in rviz node but seems it is not using. Moreover tried changing with the nav2 parameters transform tolerance and update frequencies, some of them got resolved for planner and controller server but now for other nodes it coming. 

I know this might not related to Nav2 but if someone could suggest the solution, it might be useful.

SteveMacenski, 2023-08-23:
Please ask on robotics stack exchange and provide more info about your platform

closed 2023-08-23 by SteveMacenski

taylerkubins, 2025-11-21:
have it solved? i just encountered the same problem