ros-navigation/navigation2 #3771: Rotational Shim controller that only activates when robot has reached the goal xy position

JslBotBuilder, 2023-08-23:
----
## Feature request

I'm pretty sure this functionality isn't implemented yet, but I may very well just be missing it in my search. 

#### Feature description
I'm in need of a controller that allows an omnidirectional robot to use it's lateral motion capabilities while it approaches the goal position and then commands it to rotate to the correct orientation once it has reached that position. 

#### Implementation considerations
My thought was to take the Rotational Shim controller and modify it so that it only computes a rotate to heading command when it has reached within the goal tolerance given by the goal checker. The only thing I don't is how to get the controller information about the goal_position or where that information is stored if the controller indeed has it.

SteveMacenski, 2023-08-28:
Hi,

That seems like a straight-forward contribution. You can use the `goal_checker` given in `computeVelocityCommands` to check if you've reached the XY tolerances you'd like and then have it rotate to the `current_path_`'s final marker orientation. All of that should only happen if a parameter `rotate_to_goal` is set to `true` and set the default to `false` to not adjust default behavior.

The main issue here is that most controllers will try to do that themselves, so why aren't the current controllers good enough for you at that? I think DWB and RPP have the logic built in to do that automatically (and MPPI will always approach the goal at the right angle whenever possible). So I think a real question is what is it that you're doing that this is separately necessary?

JslBotBuilder, 2023-08-29:
Yeah, I actually ended up digging out that information. Just took me a bit to realize what the getTolerances method from the goal checker did. But ended up pulling the tolerances from that and comparing it to the last element of the current path like you said. I only wanted the behavior to happen when the robot reached it's goal and not every time path updates so I just made it a separate controller instead of a modification the current (That was going to be the plan for me anyway but realize that's not how I worded it at all in my first post).

To address the real concern, the reason I'm even making this separate controller is because I was unable to tune the DWB controller to allow my robot to move in the y-direction (I need to approach all pickup points sideways) without completely breaking it's ability to rotate in place. I couldn't get the MPPI model, using the Omni motion model, to approach any of it's goal positions at the goal orientation or move in the y direction at all even when there was no obstacles or obstructions. As for the RPP controller, I was under the impression that it was not suited to applications that have various dynamic obstacles that need to be navigated around so I didn't try using it.

This seemed to me like the easiest solution and actually ended up working exactly the way we needed it to.

SteveMacenski, 2023-08-29:
OK, glad you found a solution that works for you!