ros-navigation/navigation2 #3772: Looking for recommendation on scan topic remapping

hilary-luo, 2023-08-24:
### The Setup
ROS 2 Humble on Ubuntu 22.04 running nav2 for Clearpath Husky / Jackal robots. 

### Problem
Currently we are setting the scan topic in the nav2 config yaml to `sensors/lidar2d_0/scan` with the expectation that this topic will be resolved as the global topic `/namespace/sensors/lidar2d_0/scan`. Instead this is being resolved as `/namespace/global_costmap/sensors/lidar2d_0/scan` and `/namespace/local_costmap/sensors/lidar2d_0/scan`.

As a result, nav2 is not able to access the scan data.

### The Question
We do not want to hardcode the namespace into the config, so what is the recommended method for remapping the scan topic? or should this be necessary?

### Additional Details
The full nav2 yaml can be found here: https://github.com/clearpathrobotics/clearpath_nav2_demos/blob/5ecd0116296cbfce053455dc69505a050323d4b9/config/a200/nav2.yaml#L200 
and corresponding launch file here: https://github.com/clearpathrobotics/clearpath_nav2_demos/blob/main/launch/nav2.launch.py

I did try adding SetRemap commands when launching nav2 unsuccessfully:
```
    nav2 = GroupAction([
        PushRosNamespace(namespace),
        SetRemap('global_costmap/sensors/lidar2d_0/scan', 'sensors/lidar2d_0/scan'),
        SetRemap('local_costmap/sensors/lidar2d_0/scan', 'sensors/lidar2d_0/scan'),

        IncludeLaunchDescription(
            PythonLaunchDescriptionSource(launch_nav2),
            launch_arguments=[
                ('use_sim_time', use_sim_time),
                ('params_file', file_parameters),
                ('use_composition', 'False'),
                ('namespace', namespace)
              ]
        ),
    ])
```

StetroF, 2023-08-26:
I want to know this solution as well. Currently once there is a new robot need to work  . Except change the namespace ,i have to change the loaded yaml's global_cost_map and local_cost_map's topic manually. I have a think that maybe we can use nav2_common.launch.RewrittenYaml function ,to rewrite the scan topic from /sensors/lidar2d_0/scan to /namespace/sensors/lidar2d_0/scan rather than change this topic in the config file.

SteveMacenski, 2023-08-28:
Interestingly LG added something that might help recently, which allows things like https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/params/nav2_multirobot_params_all.yaml#L213 to be parsed at launch time via https://github.com/ros-planning/navigation2/blob/6ef3d7bfd0f617bd0751cdbe7a58c3f9f66fe882/nav2_bringup/launch/bringup_launch.py#L61-L64.

Remapping would also generally work if you did it on a per-node basis for just the nodes taking in a scan in the costmaps.

SteveMacenski, 2023-08-28:
Adding the prefix `/` makes the topic absolute. Without it its relative to the namespace, which is what you see happening with the added `global_costmap` etc since that's the node's name. The global costmap you see there **is** the namespace, since the node is under `global_costmap/global_costmap` as you can see if you do `ros2 node list`.

NachtaktiverHalbaffe, 2023-08-28:
Yeah, but do the people expect this behaviour? What I try to say is that I, my fellow students and supervisor expect(ed) when we set the topic  of the observation sources to `scan` inside the configuration file, then the namespace defined by the user/developer gets appended. So if the user sets the whole nav2 stack under the namespace e.g. `robot1` that the observation source topic would change from `scan` to `/robot1/scan` (or just `/scan` if we dont pass a namespace). 

What we get then is instead is `/robot1/local_costmap/scan` (or `/local_costmap/scan`) where cleary no roboter is publishing Laserdata. The reason for that is now clear to us like you mentioned (were basically kinda applying two namespaces) and as far as I see/experienced the costmaps are the only place in nav2 where the action server run under own namespaces. As a consequence the obstacle layer is the only place were we cant use relative topics inside the yaml configuration file as observation sources and we have to use absolute topics and/or remappings on launch file layer. My question/proposal is that we change the obstacle layer under the hood (basically remapping the topic name in the code instead of remapping in launch files) so the yaml-configuration behaves like I described/expected and relative topics inside yaml-configurations can be used as observation sources in the obstacle layer. 

Then it would be the same behaviour like in other parts of the nav2 configuration. Then no custom remappings in (own) launch files would be needed and the ones provided by nav2 can be used. Then the obstacle layer works no matter whether I use relative topics in the yaml configuration like I described (my usecase) or remappings via launch files like the solution you described which makes this all less prone no misconfiguration errors. Currently we have to use absolute topics and do remappings via launch files

Maybe I have to mention that @hilary-luo and I don't work on the same project. I use the turtlebot 4 as a robot and opened the issue originally on his repo and then we forwarded onto here. Hence we have the same problem, but maybe different needs/goals regarding the solution because of different use cases

SteveMacenski, 2023-08-28:
The namespace isn't just what you set though, its also the namespaces already applied to the nodes. I don't think its a good idea to hack out a namespace for one unique situation. 

See https://github.com/ros-planning/navigation2/issues/3541 for more discussion about why the namespaces exist for the local/global costmaps. If we can find a way around that, I suppose we could just remove the namespace wrapping the costmaps as a real solution to this behavior. 

Otherwise if a generic solution cannot be found, adding the `<robot_namespace>/scan` ReplaceString in the launch file seems like a good option.

NachtaktiverHalbaffe, 2023-08-28:
Ah thanks for the link, have looked into #3541. Although I have one point to dig deeper into the discussion:

The namespace of the costmaps have the purpose of seperating the topics and services in order for the costmap and their API to work when being utilized in planner/controller server at same time. Here the namespace of the costmap has a responsibility like discussed in #3541 and is used inside the nav2-stack to seperate global and local costmap. Their purpose is to namespace the costmap topics etc. and nothing else. They provide reusability of the costmaps and also ensuring unique topics at same times. And they separate within the nav2-stack, but not on a system wide level. Out of the box I dont know a solution other than hardcoding these namespaces into the topics, so maybe its the best to continue using these "subnamespaces" like described in #3541 

The topics for the observation sources however reference the topic on which a completly other node outside the nav2 stack publishes. It hasn't directly something todo with the inner workings of nav2, the obstacle layer only reads from it. Further it can be the same for global and local costmap, they dont need to be seperated inside the nav2 stack. As a consequence the topic from the obsaervation sources fall under the namespace of the publishing robot and not of the costmap/nav2 stack. So the namespace under which the observation sources fall is from my point of view completly decoupled from the namespace of the costmaps. The namespace of the observation nodes doesn't need the be assigned to these topic(s) and are basically wrongly assigned in this case because of technical reasons when using relative topics. 

However, in multi-roboter-scenarios the user-defined namespaces are usually used to differentiate the individual robots and their running stacks to enable multi-roboter-support on the same ROS-Domain (at least from my point of view). They have a different concern and responsibility than the namespace of the costmap. So its the closest to tie the observation topic to the user-defined namespace responsibility-wise and thus setting it into the user-defined namespace and ignoring the costmap namespace. And finally like I described before in my comments I think the expected way of relative topics inside the yaml-configuration files (not to mix up with the inner workings of the nav2-stack) is that the user-defined namespaces are appended and not both the internal one and the userdefined one (which is how it basically works now). 

To sum up treating the namespace of the observation topic different like I described is justified from my point of view because of these reasons. In my opinion this doesnt hack out a namespace, it basically just re-assigns the observation topic to the right, user-defined namespace instead to the automatically added one which is the internal costmap namespace with another responsibility. And the obstacle layer would be the only place which needed to be changes as far as I can see

However, if your offered solution is still the preffered one, then maye it would be good to write a hint/warning it in the configuration documentation for the topics for the ones who don't dig as deep as we do to prohibit future misconfiguration errors.

SteveMacenski, 2023-08-29:
> So the namespace under which the observation sources fall is from my point of view completly decoupled from the namespace of the costmaps

That's nice in theory, look at the code and tell me how to do that. In reality, those topics belong to a node, specifically the costmap node, which is under its own namespace. I'm happy to agree with your philosophical position, but if that's not paired with a technically generic solution, there's not much to be done :shrug:

NachtaktiverHalbaffe, 2023-08-29:
Does this need a generic solution? The observation topic in the obstacle layer is the only place I found which needed to be fixed, so my draft should be feasible if placed in https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/plugins/obstacle_layer.cpp#L140 and looking something like:
```cpp
node->get_parameter(name_ + "." + source + "." + "topic", topic); // topic is observation topic
if(topic[0] !="/" && node -> get_namespace() != "/"){  
   topic = node -> get_namespace() + topic;
}
```

Sure it's a draft and needs testing and polishing (and seeing if I ported correctly, my mind is currently in python mode), but It should achive what I am talking about. All the other parts work already like I mentioned, so I dont know why this kind of specific solution/change to the obstacle layer hurts. But maybe we are talking past each other

SteveMacenski, 2023-08-29:
> The observation topic in the obstacle layer is the only place I found which needed to be fixed

It would be every sensor topic of every costmap layer

NachtaktiverHalbaffe, 2023-08-29:
Oh I did oversee some layers which I arent using. But then we can create a "re-assign-namespace function" in a utils package or in the layer class itself and have to only call a single line in each plugin when a topic parameter from sensors are read?

I get now that it is a manual thing the developer of new costmap plugins have to implement, its a non-automatic solution which only works if the devs know and implement it. The question underlying here is if the responsibility for detecting and handling this namespacing problems lies on the developer to call that reassigning funtion if needed or the user/integrator to detect and handle it with remapping in the launch files. Im obviously see the responsibility more in the developer and if the problem reoccurs on new plugins then its a one line fix. After that, both solutions would work and thus lesser errors and issues should occur (especially with users/integrators which blackbox the layers and arent aware of this problem) 

At the end i get that it's a decision between bad luck and brimstone, so I can understand why the decision could fall to just prefer launch file remappings

hilary-luo, 2023-08-29:
> Interestingly LG added something that might help recently, which allows things like https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/params/nav2_multirobot_params_all.yaml#L213 to [...]

@SteveMacenski Any chance on this change being ported to / released on humble? It sounds exactly like what I am looking for both for the turtlebot4 as well as the clearpath robots. 

In the meantime I will look at remapping the nodes. @NachtaktiverHalbaffe I will update the turtlebot4 repo and then let you know on your ticket when that is done.

SteveMacenski, 2023-08-29:
> But then we can create a "re-assign-namespace function" in a utils package

This is why I think this is starting to border a hack. Its subtly hiding a behavior that every other sub/pub/service has in those nodes other than the few that are hand selected to not use the namespace (and error prone for new interfaces being added). Then all new layers need to implement this very subtle util that they may or may not be aware of to get analog behavior to the default plugins. But, from your comment it sounds like you get it :-)

@hilary-luo if you open a PR to backport the ReplaceString nav2_common API, I'll merge that. Then adding 
```
    # '<robot_namespace>' keyword shall be replaced by 'namespace' launch argument
    # User defined config file should contain '<robot_namespace>' keyword for the replacements.
    params_file = ReplaceString(
        source_file=params_file,
        replacements={'<robot_namespace>': ('/', namespace)},
        condition=IfCondition(use_namespace))
```

To your launch file will do that substitution (for namespaces or anything else you like)

SteveMacenski, 2023-08-30:
Closing ticket with that answer. @hilary-luo I'll count on you for the Humble PR :smile:

closed 2023-08-30 by SteveMacenski

meetgandhi-dev, 2025-03-17:
Hello Everyone,
I'm facing the same issue while using the [navigation_launch.py](https://github.com/ros-navigation/navigation2/blob/humble/nav2_bringup/launch/navigation_launch.py). Do you believe the fix should be part of this launch file as well?