ros-navigation/navigation2 #3773: Integrating Nav2 with Ardupilot

pedro-fuoco, 2023-08-25:
Hi! I'm a GSoC student for Ardupilot. My project involves using Cartographer and Nav2 to control a copter in Gazebo Garden. So far the Gazebo and Cartographer bits have been solved and integrated into Ardupilot's native port for ROS 2, AP_DDS, with the help of @Ryanf55.

- Operating System:
  - Ubuntu 22
- ROS2 Version:
  - Humble
- Version or commit hash:
  - ros-humble-navigation2	1.1.9-1jammy.20230814.171753
- DDS implementation:
  - Micro-XRCE-DDS

I've remmaped `/cmd_vel` to `/ap/cmd_vel`, which made it possible to control the copter through the `/goal_pose` topic (which I tested with rviz). But the costmaps are not being published correctly and, because of that, the copter isn't avoiding obstacles, as you can see in the video below:

[image]

This was done launching `ros2 launch nav2_bringup navigation_launch.py use_sim_time:=true` after launching the Copter simulation in Gazebo Garden and Cartographer.

Additional information:
- Nav2 does subscribe to the map topic provided by cartographer

- Here's the tf tree from `robot_state_publisher` and cartographer. It `map -> odom -> base_link`
[image]

- The laser scan is being published at `/scan` as the default parameters for nav2 suggest

- The log from the `nav2` launch gets stuck as follows:
```
[INFO] [launch]: All log files can be found below /home/fuoco/.ros/log/2023-08-25-01-38-40-740637-fuoco-desk-116858
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_server-1]: process started with pid [117384]
[INFO] [smoother_server-2]: process started with pid [117386]
[INFO] [planner_server-3]: process started with pid [117388]
[INFO] [behavior_server-4]: process started with pid [117390]
[INFO] [bt_navigator-5]: process started with pid [117392]
[INFO] [waypoint_follower-6]: process started with pid [117394]
[INFO] [velocity_smoother-7]: process started with pid [117396]
[INFO] [lifecycle_manager-8]: process started with pid [117398]
[lifecycle_manager-8] [INFO] [1692938321.547333614] [lifecycle_manager_navigation]: Creating
[lifecycle_manager-8] [INFO] [1692938321.549701976] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[bt_navigator-5] [INFO] [1692938321.549969914] [bt_navigator]: 
[bt_navigator-5] 	bt_navigator lifecycle node launched. 
[bt_navigator-5] 	Waiting on external lifecycle transitions to activate
[bt_navigator-5] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-5] [INFO] [1692938321.550069553] [bt_navigator]: Creating
[controller_server-1] [INFO] [1692938321.550973482] [controller_server]: 
[controller_server-1] 	controller_server lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[velocity_smoother-7] [INFO] [1692938321.551451257] [velocity_smoother]: 
[velocity_smoother-7] 	velocity_smoother lifecycle node launched. 
[velocity_smoother-7] 	Waiting on external lifecycle transitions to activate
[velocity_smoother-7] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [INFO] [1692938321.558875814] [planner_server]: 
[planner_server-3] 	planner_server lifecycle node launched. 
[planner_server-3] 	Waiting on external lifecycle transitions to activate
[planner_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-1] [INFO] [1692938321.561262477] [controller_server]: Creating controller server
[waypoint_follower-6] [INFO] [1692938321.561540913] [waypoint_follower]: 
[waypoint_follower-6] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-6] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-6] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-6] [INFO] [1692938321.562945670] [waypoint_follower]: Creating
[planner_server-3] [INFO] [1692938321.563979479] [planner_server]: Creating
[smoother_server-2] [INFO] [1692938321.564302573] [smoother_server]: 
[smoother_server-2] 	smoother_server lifecycle node launched. 
[smoother_server-2] 	Waiting on external lifecycle transitions to activate
[smoother_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[smoother_server-2] [INFO] [1692938321.567915433] [smoother_server]: Creating smoother server
[behavior_server-4] [INFO] [1692938321.574240440] [behavior_server]: 
[behavior_server-4] 	behavior_server lifecycle node launched. 
[... 100 lines cut ...]
[lifecycle_manager-8] [INFO] [1692938323.298581490] [lifecycle_manager_navigation]: Activating behavior_server
[behavior_server-4] [INFO] [1692938323.298737594] [behavior_server]: Activating
[behavior_server-4] [INFO] [1692938323.298757467] [behavior_server]: Activating spin
[behavior_server-4] [INFO] [1692938323.298766257] [behavior_server]: Activating backup
[behavior_server-4] [INFO] [1692938323.298772321] [behavior_server]: Activating drive_on_heading
[behavior_server-4] [INFO] [1692938323.298778313] [behavior_server]: Activating assisted_teleop
[behavior_server-4] [INFO] [1692938323.298783653] [behavior_server]: Activating wait
[behavior_server-4] [INFO] [1692938323.298792649] [behavior_server]: Creating bond (behavior_server) to lifecycle manager.
[lifecycle_manager-8] [INFO] [1692938323.400977077] [lifecycle_manager_navigation]: Server behavior_server connected with bond.
[lifecycle_manager-8] [INFO] [1692938323.401038759] [lifecycle_manager_navigation]: Activating bt_navigator
[bt_navigator-5] [INFO] [1692938323.401198534] [bt_navigator]: Activating
[bt_navigator-5] [INFO] [1692938323.422932591] [bt_navigator]: Creating bond (bt_navigator) to lifecycle manager.
[lifecycle_manager-8] [INFO] [1692938323.525302904] [lifecycle_manager_navigation]: Server bt_navigator connected with bond.
[lifecycle_manager-8] [INFO] [1692938323.525337624] [lifecycle_manager_navigation]: Activating waypoint_follower
[waypoint_follower-6] [INFO] [1692938323.525538661] [waypoint_follower]: Activating
[waypoint_follower-6] [INFO] [1692938323.525565539] [waypoint_follower]: Creating bond (waypoint_follower) to lifecycle manager.
[lifecycle_manager-8] [INFO] [1692938323.628096670] [lifecycle_manager_navigation]: Server waypoint_follower connected with bond.
[lifecycle_manager-8] [INFO] [1692938323.628133169] [lifecycle_manager_navigation]: Activating velocity_smoother
[velocity_smoother-7] [INFO] [1692938323.628342224] [velocity_smoother]: Activating
[velocity_smoother-7] [INFO] [1692938323.628403318] [velocity_smoother]: Creating bond (velocity_smoother) to lifecycle manager.
[lifecycle_manager-8] [INFO] [1692938323.730793485] [lifecycle_manager_navigation]: Server velocity_smoother connected with bond.
[lifecycle_manager-8] [INFO] [1692938323.730822686] [lifecycle_manager_navigation]: Managed nodes are active
[lifecycle_manager-8] [INFO] [1692938323.730829778] [lifecycle_manager_navigation]: Creating bond timer...

```

mamariomiamo, 2023-08-28:
Just wondering if UAV would require a 3D costmap?

Ryanf55, 2023-08-28:
Yea we will need full 3D support eventually, but the scope right now is using NAV2 on a copter but only moving in 2D.

SteveMacenski, 2023-08-28:
Please ask for configuration help on Robotics Stack Exchange. 

But generally speaking, I don't see a costmap in your video, so you may want to look into that + configuring it for your application.

closed 2023-08-28 by SteveMacenski

Karthik03B, 2024-10-14:
It will help alot if you can push this project to your GitHub repositories.
Thank you