ros-navigation/navigation2 #3774: Cannot install gazebo on RPI4 8Gb - Humble - Ubuntu 22.04

phil123456, 2023-08-25:
Ubuntu 22.04
ROS Humble / RPI4 8gb

Steps to reproduce issue

```
source /opt/ros/humble/setup.bash

sudo apt install ros-humble-ros-base python3-colcon-common-extensions python3-rosdep2 ~nros-humble-rqt* ros-humble-example-interfaces git ros-humble-tf2-tools ros-humble-tf-transformations tree ros-humble-navigation2 ros-humble-nav2-bringup ros-humble-slam-toolbox ros-humble-rviz2 ros-humble-turtlebot3* ros-humble-desktop
```

gazebo should be available , yet

```
pi@pi-ros2:~/WORK$ gazebo
gazebo: command not found
```
searhing for it :
```
pi@pi-ros2:~/WORK$ sudo find / -name gazebo
[sudo] password for pi: 
/opt/ros/humble/share/turtlebot3_manipulation_description/gazebo
/home/pi/WORK/bookros2_ws/src/ThirdParty/tiago_robot/tiago_description/gazebo
/home/pi/WORK/bookros2_ws/src/ThirdParty/pmb2_robot/pmb2_description/gazebo
/home/pi/WORK/bookros2_ws/src/ThirdParty/pal_gripper/pal_gripper_description/gazebo
/home/pi/WORK/bookros2_ws/src/ThirdParty/hey5_description/gazebo
```

trying to install ros-humble-turtlebot3-gazebo

```
sudo apt install ros-humble-turtlebot3-gazebo
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
**E: Unable to locate package ros-humble-turtlebot3-gazebo**
```

so I cannot follow basic navigation tutorials

https://navigation.ros.org/getting_started/index.html
https://roboticsbackend.com/ros2-nav2-tutorial/
https://www.youtube.com/watch?v=idQb2pB-h2Q&t=4s

.bashrc log:
```
-----------------------------------------
ROS_HOSTNAME: rosmaster
ROS_IP: 192.168.1.84
ROS_MASTER_URI: http://rosmaster:11311
ROS_DOMAIN_ID: 1
ROS_LOCALHOST_ONLY: 0
RMW_IMPLEMENTATION: 
TURTLEBOT3_MODEL: waffle
GAZEBO_MODEL_PATH: :/opt/ros/humble/share/turtlebot3_gazebo/models
-----------------------------------------
ROS_VERSION=2
SESSION_MANAGER=local/pi-ros2:@/tmp/.ICE-unix/3084,unix/pi-ros2:/tmp/.ICE-unix/3084
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=1
ROS_IP=192.168.1.84 
VNCDESKTOP=pi-ros2:1 (pi)
AMENT_PREFIX_PATH=/opt/ros/humble
PYTHONPATH=/opt/ros/humble/lib/python3.10/site-packages:/opt/ros/humble/local/lib/python3.10/dist-packages
ROS_MASTER_URI=http://rosmaster:11311
ROS_HOSTNAME=rosmaster
LD_LIBRARY_PATH=/opt/ros/humble/opt/rviz_ogre_vendor/lib:/opt/ros/humble/lib/aarch64-linux-gnu:/opt/ros/humble/lib
ROS_LOCALHOST_ONLY=0
GAZEBO_MODEL_PATH=:/opt/ros/humble/share/turtlebot3_gazebo/models
PATH=/opt/ros/humble/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
ROS_DISTRO=humble

```

SteveMacenski, 2023-08-25:
See the pinned ticket https://github.com/ros-planning/navigation2/issues/3766

closed 2023-08-25 by SteveMacenski

phil123456, 2023-08-26:
do you know how to build from source the gazebo packages ?

phil123456, 2023-08-28:
> See the pinned ticket #3766

I use Humble, not Iron

how come these packages are not available ?
how can I build them ?

brianlmerritt, 2024-02-23:
On aarch64 as it is called these days, I tried 

`rosdep check --from-paths src --ignore-src --rosdistro=humble` but that failed on the missing ros-humble-gazebo-ros-pkgs

An alternate method is to apt install the below
```
ros-humble-slam-toolbox
ros-humble-ompl
nlohmann-json3-dev
lcov
ros-humble-diagnostic-updater
libgraphicsmagick++1-dev
graphicsmagick-libmagick-dev-compat
libxtensor-dev
```
You might depending upon your build need more stuff, so in the root workspace you can always type after installing the above to see if anything else is missing

`rosdep check --from-paths src --ignore-src --rosdistro=humble`

chatgpt gave me this auto install if all the packages are apt

`rosdep check --from-paths src --ignore-src --rosdistro=humble 2>&1 | grep '^apt' | grep -v 'ros-humble-gazebo-ros-pkgs' | sed 's/apt\s\+//' | xargs -I {} sudo apt install -y {}
`

You then run

`colcon build --symlink-install --cmake-args -Dbondcpp_DIR=/opt/ros/humble/share/bondcpp/cmake --packages-skip nav2_system_tests`

This fixes a build error where cmake can't find bondcpp and skips nav2_system_tests which has the gazebo requirement

I have yet to test this actually running apart from bringing it up without the robot running - if I can find this issue again I will update success or not