ros-navigation/navigation2 #3779: Rotate shim controller and MPPI controller with foxy distribution possible?

NamTruongTran, 2023-08-28:
Hey,

first I would like to thank you for your amazing contributions. 

Is it possible to downgrade both controllers to the Foxy distribution? I rely on Foxy and can't use Humble.

Thank you :)

SteveMacenski, 2023-08-29:
No, Foxy is EOL

closed 2023-08-29 by SteveMacenski

NamTruongTran, 2023-08-29:
Hey Steve,

thank you for your fast reply.
I know but I depend on foxy that's why I want to use both planner in foxy.
In General it is possible to downgrade these two planners to be compatible with foxy? (Change syntax etc.).

SteveMacenski, 2023-08-29:
Its probably possible, yes! I think it would be relatively low effort but would require some effort

Namo98, 2023-09-04:
Hey Steve,

soryy for bother you again. 
I successfully converted the Rotate shim controller to foxy. 
Unfortuneately i can't convert the mppi controller to foxy. It seems like xtensor is missing:
This is the output of my terminal:
```
--- stderr: nav2_mppi_controller                                           
CMake Error at CMakeLists.txt:12 (find_package):
  By not providing "Findxtensor.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "xtensor", but
  CMake did not find one.

  Could not find a package configuration file provided by "xtensor" with any
  of the following names:

    xtensorConfig.cmake
    xtensor-config.cmake

  Add the installation prefix of "xtensor" to CMAKE_PREFIX_PATH or set
  "xtensor_DIR" to a directory containing one of the above files.  If
  "xtensor" provides a separate development package or SDK, be sure it has
  been installed.
```

Do you know how i can fix that?
Thank you in advance.