ros-navigation/navigation2 #3780: nav2_behavior Wait is not using sim time

doisyg, 2023-08-29:
## Bug report

The Wait behavior https://github.com/ros-planning/navigation2/blob/main/nav2_behaviors/plugins/wait.cpp is not using the ROS clock and hence will not have a consistant behavior when used in sim if the real time factor is not maintained to 1

#### Expected behavior

Wait based on sim time when in sim

#### Actual behavior

Wait based on the wall clock

#### Additional information

Would it be okay to switch to using the ROS clock ? I can make a PR

SteveMacenski, 2023-08-29:
This is covered by https://github.com/ros-planning/navigation2/issues/3303, but sure a PR to resolve this particular instance would be nice! We're waiting on some other things in rclcpp itself to update Rate to be able to use ROS time, but this particular instance doesn't require it

doisyg, 2023-08-29:
https://github.com/ros-planning/navigation2/pull/3782