ros-navigation/navigation2 #3781: Hardcoded timeout for action client in BtActionNode

fmros, 2023-08-29:
Hy there,

the BtActionNode has a hardcoded timeout [here](https://github.com/ros-planning/navigation2/blob/6ef3d7bfd0f617bd0751cdbe7a58c3f9f66fe882/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L96C55-L96C55) to wait for the server to be active. I have seen a system sporadically exceed the 1 second for the client to connect. I would propose to use the `server_timeout` parameter from the `input_port` to be able to customize this timeout. Let me know and I can propose a PR.

SteveMacenski, 2023-08-29:
What setup do you have that it takes longer than 1s for something to detect a running server? That seems like there are other issues afoot, I view that `1s` as such an excessively long time that no one should require longer.

But if you have a good reason / explanation, then a parameter would make sense. I'd also update the BT action cancel & BT service node as well that has the same analog structure as the code you point to with a `1s` wait timeout

fmros, 2023-08-30:
We are running the nav2 stack on a mobile robot onboard PC, having approximately 50% CPU laod. 

We are using FastRTPS and following profile:
````xml
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles" >
<transport_descriptors>
        <transport_descriptor>
        <transport_id>CustomUdpTransport</transport_id>
        <type>UDPv4</type>
        </transport_descriptor>
</transport_descriptors>

<participant profile_name="participant_profile" is_default_profile="true">
        <rtps>
        <userTransports>
                <transport_id>CustomUdpTransport</transport_id>
        </userTransports>

        <useBuiltinTransports>false</useBuiltinTransports>
        </rtps>
</participant>
</profiles>
````
Any hints are very much appreciated. Thank you!

SteveMacenski, 2023-08-30:
This is a good question, I can't say I've played too much with the DDS configs - you may want to file a ticket with eProsima if discovery on-computer is taking > 1 second or review the Fast-DDS documentation. This is not something I'm deeply familiar with. It should definitely not be taking that long. That really is what sounds like is the root of your issues. I assumed if you were taking > `1s` that you were trying to connect to a cloud server on flaky internet or something. If on the same computer or LAN, 1s is outrageously high (and that's why its set to 1s) 

Have you also tried Cyclone-DDS? Its what I and many in the Nav2 community use, but not to disuade you from Fast-DDS if it otherwise is your preferred solution due to some specific integrations or features.

MiguelCompany, 2023-08-31:
@fmros @SteveMacenski 

Discovery in ROS 2 involves UDP traffic both to the discovery port and the user data port since it involves the `ros_discovery_info` topic that conveys the node graph information.

This is a hard guess, but if there is a lot of traffic there might be a lot of drops due to the socket buffers getting overflowed, and that could explain the issue.

The default configuration makes Fast DDS use the system's default values for the socket buffer sizes.
You can set different values either [on the XML](https://fast-dds.docs.eprosima.com/en/latest/fastdds/xml_configuration/transports.html) or using `sysctl` calls.

An updated version of your XML file, increasing the buffer sizes to 12M, would be the following

```xml
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles" >
    <transport_descriptors>
        <transport_descriptor>
            <transport_id>CustomUdpTransport</transport_id>
            <type>UDPv4</type>
            <sendBufferSize>12582912</sendBufferSize>
            <receiveBufferSize>12582912</receiveBufferSize>
        </transport_descriptor>
    </transport_descriptors>

    <participant profile_name="participant_profile" is_default_profile="true">
        <rtps>
            <userTransports>
                <transport_id>CustomUdpTransport</transport_id>
            </userTransports>

            <useBuiltinTransports>false</useBuiltinTransports>
        </rtps>
    </participant>
</profiles>
```

fmros, 2023-08-31:
Hy @MiguelCompany, we tried to apply the proposed profile but see no improvement. Can you recommend a technique to investigate? Thanks a lot.

BriceRenaudeau, 2023-09-05:
Related to this one:  https://github.com/ros-planning/navigation2/issues/3682

SteveMacenski, 2023-09-11:
Closing https://github.com/ros-planning/navigation2/issues/3781 as duplicate of https://github.com/ros-planning/navigation2/issues/3682. Please lets continue the discussion over there that has some more discussions that have already taken place!

closed 2023-09-11 by SteveMacenski

BriceRenaudeau, 2023-09-12:
It is not an issue anymore. We made is more robust to start the BT when everything is ready.