ros-navigation/navigation2 #3784: [RPP] Carrot at end of path does not respect look ahead distance

james-ward, 2023-08-30:
## Feature request

#### Feature description
At the moment the `getLookAheadPoint()` function takes the last point on the path if the look ahead distance would put it past the last point. This has the effect of increasing the gain of the steering inputs (ie the look ahead distance is reduced).
https://github.com/ros-planning/navigation2/blob/210977790bfb6e809f2c2ef3b7e1660662d630cc/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L323-L325

We are using an Ackermann steered vehicle and having the carrot be so close to the robot makes for very unstable motion. When the robot pose is close to the final point in the path small deviations create large steering efforts, and inevitably full-lock steering just before the path is completed.

#### Implementation considerations
Project carrot off the end of the last path segment in RPP controller. There is various collision checking, etc that is based on the carrot position, so this "augmented carrot" would have to be implemented such that those functions are unchanged.

If this sounds like the thing to do, I'll get on to writing something.

SteveMacenski, 2023-08-30:
Sounds like a nice feature addition, I think that would be valuable to have. Making sure we still slow down / stop at the goal pose is important though!

@doisyg at Dexory has thought about something similar to speed the robot towards the goal with this commit which might be a useful starting point (or actually just be what you need [but perhaps broken into a new function]!) https://github.com/ros-planning/navigation2/commit/a9d8c4c9889c7d8684296c5d7733f507255520c1

james-ward, 2023-08-31:
Heh! That commit is pretty much what I was thinking of/needed!
I'll use that as a starting point and put in a PR.

doisyg, 2023-08-31:
Glad it is useful!