ros-navigation/navigation2 #3790: nav2 devcontainer on humble branch uses rolling image

Ryanf55, 2023-09-01:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 23
- ROS2 Version:
  - Docker (Ros humble)
- Version or commit hash:
  - humble
- DDS implementation:
  - n/a
#### Steps to reproduce issue

SteveMacenski, 2023-09-01:
@ruffsl

Ryanf55, 2023-09-01:
Please assign this to me; I can take the recommended route to add compose support to nav2. Do you want a `docker-compose.yml` file at the root of the repo, or in the `.devcontainer` folder?

ruffsl, 2023-09-01:
> I would expect `FROM_IMAGE` is set to `humble` when using VSCode on the humble branch.

How would you expect to set the build arg dynamically, while remaining agnostic of the docker client? The dev container could be started from Mac or windows, so use of scripts would be out of scope. We could just change/commit the default value for that build arg per branch.

Ryanf55, 2023-09-01:
> How would you expect to set the build arg dynamically, while remaining agnostic of the docker client? The dev container could be started from Mac or windows, so use of scripts would be out of scope. [...]

That's what I did; take a look at the draft PR?

Ryanf55, 2023-10-25:
I tried a very specific change to just change the `build.cacheFrom` argument, however the dockerfile still has an ARG for the ROS distro, and VSCode doesn't support build arguments in devcontainer. 
https://github.com/microsoft/vscode-remote-release/issues/3545

I don't have a good path forward on this.

Ryanf55, 2024-01-17:
We have a path forward. I'd say once the next VSCode version is out of beta and released, we can add build arguments to specify a ros distro.
https://github.com/microsoft/vscode-remote-release/issues/3545#issuecomment-1892274755

tonynajjar, 2024-04-22:
This should now be possible @Ryanf55 right?

Ryanf55, 2024-04-22:
> This should now be possible @Ryanf55 right?

Correct, feel free to submit a PR. I have other priorities right now.

SteveMacenski, 2025-05-03:
Working on this right now as a Friday afternoon toy project. I just enabled Jazzy in https://github.com/ros-navigation/navigation2/pull/5119 and https://github.com/ros-navigation/navigation2/pull/5121. Working on Humble next https://github.com/ros-navigation/navigation2/pull/5122

closed 2025-05-03 by SteveMacenski