ros-navigation/navigation2 #3797: Errors installing nav2_bringup and nav2_system_tests on and and arm63 due missing Gazebo Class packages

ARLunan, 2023-09-05:
Required Info:
Operating System:
Ubuntu 22.04.3 amd64
ROS2 Version: ROS Iron binary
Version or commit hash:
1.2.2-1jammy.20230804.210746

Steps to reproduce issue
On Ubuntu 22.04.3 Intel i7 PC (amd64) 
Try to install ros-iron-nav2-bringup
Try to install ros-iron-nav2-system-tests

Expected behavior
ros-iron-nav2-bringup should be available for amd64 (and arm64 also)
ros-iron-nav2-system-tests should be available for amd64 (and arm64 also)

Actual behavior
ubuntu@7400-ub22i-r2:~$ sudo apt install ros-iron-nav2-system-tests
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help resolve the situation:

The following packages have unmet dependencies:
 gz-tools2 : Conflicts: gazebo (>= 11.0.0) but 11.10.2+dfsg-1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
ubuntu@7400-ub22i-r2:

Same behavior with ubuntu@7400-ub22i-r2:~$ sudo apt install ros-iron-nav2-bringup

Feature request
Provide a Binary install of ROS 2 Iron nav2_bringup and nav2_system_tests packages that are compatible with Gazebo Harmonic for amd64 and arm64 machines.

closed 2023-09-06 by SteveMacenski

ARLunan, 2023-09-06:
#3766 which referred only to arm64 is closed and I could not Comment, so I posted my slightly different #3797 that referred to amd64 and arm64. I am additionally reporting that the Gazebo Classic problem is relevant to both amd64 and arm64 machines. Agreed the common issue is that in navigation2 packages, current Gazebo Classic functions need to be removed and changed out for Gazebo Harmonic.

SteveMacenski, 2023-09-06:
That's odd that you couldn't comment, its not locked. This seems incorrect from what I can see. 

```
steve@mo:/opt/ros$ apt show ros-iron-nav2-bringup
Package: ros-iron-nav2-bringup
Version: 1.2.2-1jammy.20230821.231231
Priority: optional
Section: misc
Maintainer: Michael Jeronimo <michael.jeronimo@intel.com>
Installed-Size: 347 kB
Depends: ros-iron-launch-ros, ros-iron-nav2-common, ros-iron-navigation2, ros-iron-slam-toolbox, ros-iron-turtlebot3-gazebo, ros-iron-ros-workspace
Download-Size: 22.6 kB
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: Bringup scripts and configurations for the Nav2 stack
```

```
steve@mo:/opt/ros$ sudo apt install ros-iron-nav2-bringup
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ros-iron-nav2-bringup is already the newest version (1.2.2-1jammy.20230821.231231).
```

It appears to have info about my installed version and has the ability to be installed. 

Further those binaries are working:
```
source /opt/ros/iron/setup.bash
ros2 launch nav2_bringup tb3_simulation_launch.py
```

The amd64 binaries look good to me. I looked over the jobs too in build.ros2.org and they are succeeding, which implies the gazebo dependencies are also succeeding.

ARLunan, 2023-09-07:
Tnx, I get the same result as you on my amd64 machine (Intel i7 x86-64) from "apt show ros-iron-nav2-bringup", that I note it has a ros-iron-turtlebot-gazebo dependency, which in turn has a ros-iron-gazebo-ros-pkgs dependency .

However I cannot install that because I believe there is a conflict due further missing dependencies. My Synaptic Package Manager issues loads of warnings!

***@***.***:/opt/ros$ sudo apt install ros-iron-nav2-bringup
[sudo] password for ubuntu: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help resolve the situation:

The following packages have unmet dependencies:
 gz-tools2 : Conflicts: gazebo (>= 11.0.0) but 11.10.2+dfsg-1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

***@***.***:/opt/ros$ sudo apt install ros-iron-gazebo-ros-pkgs
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help resolve the situation:

The following packages have unmet dependencies:
 gz-tools2 : Conflicts: gazebo (>= 11.0.0) but 11.10.2+dfsg-1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

Noting the reference to Gazebo 11 i.e. “Classic” and I understand should not be installed along with Gazebo Sim 8.0, which is the whole point of this current Gazebosim Migration to Harmony. Gazebo 11 can be installed on Ubuntu 22.04 but I would rather not.

So based on what as I’ve read & discussed elsewhere , to try out a functional amd64 Desktop (and possibly future arm64) Iron navigation2 and Gazebo 8 to use with my current “bots” (currently Humble "Neato Turtle" and Create Turtle”) that I want to upgrade to Iron and gz harmony, I need to develop and compile a “gazebo free” source of the nav2-bringup and nav2-system-test” packages? Then with more future work, add back in Gazebo 8 functionality. Or maybe others may do that!

I appreciate  any corrections to any misunderstandings I have and your perspective advancing functionality of navigation2 and gazebo8 ?
Regards
Ross

SteveMacenski, 2023-09-07:
Every ros distribution has a supported gazebo version and that is the only version we officially support for that distribution. It sounds like you’ve installed another version and your issues are related to that. Unfortunately I can’t help unbreak non-standard setups, your best bet is to have a clean install or ask on robotics stack exchange for general assistance. This isn’t a problem with Nav2.