ros-navigation/navigation2 #3798: The robot has collision when faced with an obstacle in front direction and the goal is set behind the obstacle

GiantSeaweed, 2023-09-06:
## Bug report

**Required Info:**

- Operating System: 
  - Ubuntu 20.04 
- ROS2 Version:
  - Foxy source 
- Version or commit hash:
  - Foxy-devel `ca482808a7a7c52ce01ae3c662dc2b980968fc16` 

#### Steps to reproduce issue

I kept every configuration of `foxy-devel` branch.

Step1. Launch the nodes and simulation.
 
(Terminal 1) Started turtlebot3_world with: 
```ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py```

(Terminal 2) Start slam_toolbox async mode: 
```ros2 launch slam_toolbox online_async_launch.py```

(Terminal 3) Start nav2 navigation: 
```ros2 launch nav2_bringup navigation_launch.py```

(Terminal 4) Start rviz2: 
```ros2 launch nav2_bringup rviz_launch.py```

Step2.  I found when the robot is _**faced with an obstacle in front direction and the destination is set behind the obstacle**_, the robot cannot successfully navigate around the obstacle to reach the goal. I attached the Terminal 3 console log and video below.

#### Expected behavior
The robot may spin to a direction with more free room ahead and bypass the obstacle.

#### Actual behavior
The robot just got stuck, could not follow the planned path, leading to a failed navigation task.

I am not sure if it is related to configuration problems. I can understand that if the tolerance is too narrow the robot cannot bypass the obstacle to reach the goal. But I am confused why it collides into the obstacle. Thanks!!

https://github.com/ros-planning/navigation2/assets/28821472/e3a1a6e1-2100-4ea0-aaa5-171e82c5608d

```
[bt_navigator-4] [INFO] [1693968008.686269475] [bt_navigator]: Begin navigating from current location to (-1.27, -0.75)
[planner_server-2] [WARN] [1693968008.686876370] [planner_server]: GridBased: failed to create plan with tolerance 0.50.
[planner_server-2] [WARN] [1693968008.686906395] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.27, -0.75)
[planner_server-2] [WARN] [1693968008.686918807] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1693968008.706985920] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[controller_server-1] [INFO] [1693968008.726921680] [controller_server]: Received a goal, begin computing control effort.
[controller_server-1] [INFO] [1693968008.777180826] [controller_server]: Passing new path to controller.
[planner_server-2] [WARN] [1693968009.737098910] [planner_server]: GridBased: failed to create plan with tolerance 0.50.
[planner_server-2] [WARN] [1693968009.737130425] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.27, -0.75)
[planner_server-2] [WARN] [1693968009.737137312] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1693968009.756714712] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[controller_server-1] [INFO] [1693968009.777152883] [controller_server]: Passing new path to controller.
[controller_server-1] [ERROR] [1693968010.529579780] [DWBLocalPlanner]: No valid trajectories out of 419! 
[controller_server-1] [ERROR] [1693968010.529606501] [DWBLocalPlanner]: 1.00: BaseObstacle/Trajectory Hits Obstacle.
[controller_server-1] [ERROR] [1693968010.529754903] [controller_server]: No valid trajectories out of 419! 
[controller_server-1] [WARN] [1693968010.529788008] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
[controller_server-1] [INFO] [1693968010.547144978] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[controller_server-1] [INFO] [1693968010.547818969] [controller_server]: Received a goal, begin computing control effort.
[planner_server-2] [WARN] [1693968010.777193461] [planner_server]: GridBased: failed to create plan with tolerance 0.50.
[planner_server-2] [WARN] [1693968010.777241375] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.27, -0.75)
[planner_server-2] [WARN] [1693968010.777250261] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1693968010.796811538] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[controller_server-1] [INFO] [1693968010.848096875] [controller_server]: Passing new path to controller.
[controller_server-1] [ERROR] [1693968010.951130298] [DWBLocalPlanner]: No valid trajectories out of 419! 
[controller_server-1] [ERROR] [1693968010.951168582] [DWBLocalPlanner]: 1.00: BaseObstacle/Trajectory Hits Obstacle.
[controller_server-1] [ERROR] [1693968010.951326140] [controller_server]: No valid trajectories out of 419! 
[controller_server-1] [WARN] [1693968010.951359895] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
[controller_server-1] [INFO] [1693968010.966944051] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[... 92 lines cut ...]
[controller_server-1] [INFO] [1693968032.237175118] [controller_server]: Passing new path to controller.
[planner_server-2] [WARN] [1693968033.216979295] [planner_server]: GridBased: failed to create plan with tolerance 0.50.
[planner_server-2] [WARN] [1693968033.217011864] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.27, -0.75)
[planner_server-2] [WARN] [1693968033.217017793] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1693968033.226898727] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[controller_server-1] [INFO] [1693968033.287195106] [controller_server]: Passing new path to controller.
[planner_server-2] [WARN] [1693968034.256921845] [planner_server]: GridBased: failed to create plan with tolerance 0.50.
[planner_server-2] [WARN] [1693968034.256951739] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.27, -0.75)
[planner_server-2] [WARN] [1693968034.256958531] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1693968034.276888081] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[controller_server-1] [INFO] [1693968034.287185919] [controller_server]: Passing new path to controller.
[planner_server-2] [WARN] [1693968035.296949653] [planner_server]: GridBased: failed to create plan with tolerance 0.50.
[planner_server-2] [WARN] [1693968035.296986206] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.27, -0.75)
[planner_server-2] [WARN] [1693968035.296992984] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1693968035.306826429] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[controller_server-1] [INFO] [1693968035.337221371] [controller_server]: Passing new path to controller.
[controller_server-1] [ERROR] [1693968035.937211680] [controller_server]: Failed to make progress
[controller_server-1] [WARN] [1693968035.937366058] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
[controller_server-1] [INFO] [1693968035.956797897] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[planner_server-2] [INFO] [1693968035.957056211] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[recoveries_server-3] [INFO] [1693968035.957307656] [recoveries_server]: Attempting spin
[recoveries_server-3] [INFO] [1693968035.957343306] [recoveries_server]: Turning 1.57 for spin recovery.
[recoveries_server-3] [INFO] [1693968036.957427276] [recoveries_server]: spin running...
[recoveries_server-3] [INFO] [1693968037.957513273] [recoveries_server]: spin running...
```

SteveMacenski, 2023-09-06:
Please display the local costmap's footprint in another gif and zoom in on it (I don't care about the global costmap or its footprint or the TF frames, disable those so we can see clearly)

I assume you didn't disable DWB's obstacle critics - did you do any retuning of configurations of the controller? 

Something doesn't seem right here in the first place - why is the planner going straight through an obstacle? This vaguely reminds me of a synchronization issue with costmap clearing we resolved years ago. Given you're on Foxy, I suspect that may be the cause. If you upgrade to Humble or newer, does that work? You may need to compile from source and make a few superficial API updates from foxy->humble codebase changes, but it shouldn't take long. 

Usually I'd close with "foxy's not supported" but collisions shouldn't be happening and I want to make sure this isn't still an issue in a modern version.

GiantSeaweed, 2023-09-07:
I disable the global costmap and TF and keep the local costmap. I did not do any retuning of controller configs, just the config in `foxy-devel` branch. And I attach a new video here. 

You can see at 3s, the robot successfully finished the previous navigation task and stopped quite close to the obstacle. 

Then, seven seconds later, I set the goal at 10s. Then the trajectory seems to have overlap with obstacle and the collision happened.

As for the planner, the console shows the info like `Planning algorithm GridBased failed to generate a valid path to xxx`.

https://github.com/ros-planning/navigation2/assets/28821472/f1cf6f0f-37fe-4ef5-8f8f-4ca1a92e3ddf

I haven't try Humble source build yet, could you give me a pointer on how to make API updates from foxy->humble? Thanks!

SteveMacenski, 2023-09-07:
I asked to see the footprint in rviz, I don't see any footprint at all. That may be part of your issue unless you didn't enable it. 

I see the pipeline 
- Going to goal
- Failed to plan
- Clearign costmap
- Planning successful
- Going

That aligns with the bug that we resolved long ago w.r.t. costmap clearing mechanics and threading issues, I believe. Try a modern ROS 2 distribution and it very well likely is gone. You can try also using Docker with the Humble image and install Nav2 from binaries in the docker image (will be much faster than building) and running in the image. That may be faster.

Else, if you want to compile locally, start building and working through API issues. I can't tell you much more than that without doing it myself.

GiantSeaweed, 2023-09-07:
Thanks for your response! I found you said this issue is a wont-fix problem for Foxy at here https://github.com/ros-planning/navigation2/issues/2452. I will try on Humble.

I just attached a video with local costmap's footprint here. I changed the Fixed Frame in Rviz from `map` to `odom`, as instructed by [here](https://navigation.ros.org/setup_guides/footprint/setup_footprint.html#visualizing-footprint-in-rviz). I think it confirms your speculation.

https://github.com/ros-planning/navigation2/assets/28821472/40d01ad3-0d7b-4dea-873a-8e65fbe71e5a

closed 2023-09-08 by SteveMacenski