ros-navigation/navigation2 #3800: Bug in Behavior Tree XML Nodes Plugins

rscova, 2023-09-08:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04
- ROS2 Version: Humble
- Version or commit hash:  1.1.8
- Package: nav2_behavior_tree

### Description of the bug
Behavior Tree XML Nodes Plugins (Action, Condition, Control and Decorator) are trying to get parameters from *bt_navigator* when they can't.  For instance, DistanceController from `distance_controller.cpp` is performing this code:
```
node_ = config().blackboard->get<rclcpp::Node::SharedPtr>("node");
node_->get_parameter("transform_tolerance", transform_tolerance_);
```
When **node_** is in fact from `bt_action_server_impl.hpp`:
```
client_node_ = std::make_shared<rclcpp::Node>("_", options);
blackboard_->set<rclcpp::Node::SharedPtr>("node", client_node_);
``` 

#### Expected behavior
transform_tolerance_ value has the value set in navigation params yaml:
```
bt_navigator:
  ros__parameters:
    transform_tolerance: 0.1
```
Or from nav2_bt_navigator, bt_navigator.cpp:
`  declare_parameter("transform_tolerance", rclcpp::ParameterValue(0.1)); `

#### Actual behavior
transform_tolerance_ is not getting the value from node_->get_parameter. When you print the value, it is 0.0, instead of 0.1.

#### Additional information
Use `navigate_w_replanning_distance.xml` behavior tree.
It seems that everything works fine, but in reality, the value is always 0.
The problem seems to be because the fake node, client_node_, is sent instead of the actual node that is  given in used in `bt_action_server_impl.hpp` because adding to this code: 
```
node->get_parameter("transform_tolerance", transform_tolerance_);
```
where `node` is `auto node = node_.lock();` you have the current value of transform_tolerance.

SteveMacenski, 2023-09-09:
This is a good point. The client node is not the BT Navigator node itself. While useful to separate the BT Navigator's execution from BT internal interfaces (e.g. topic, service, and action calls) it cannot be used to obtain parameters from the BT navigator, except when explicitly forwarded (ex https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server_impl.hpp#L132-L135).

For transform tolerance, we can explicitly forward it like we do the global frame and base frame IDs as general params. 

```
plugins/action/remove_passed_goals_action.cpp:  node->get_parameter("transform_tolerance", transform_tolerance_);
plugins/decorator/speed_controller.cpp:  node_->get_parameter_or("odom_topic", odom_topic, std::string("odom"));
plugins/decorator/goal_updater_node.cpp:  node_->get_parameter_or<std::string>("goal_updater_topic", goal_updater_topic, "goal_update");
plugins/decorator/distance_controller.cpp:  node_->get_parameter("transform_tolerance", transform_tolerance_);
plugins/condition/goal_reached_condition.cpp:  node_->get_parameter_or<double>("goal_reached_tol", goal_reached_tol_, 0.25);
plugins/condition/goal_reached_condition.cpp:  node_->get_parameter("transform_tolerance", transform_tolerance_);
plugins/condition/distance_traveled_condition.cpp:  node_->get_parameter("transform_tolerance", transform_tolerance_);
```

However we also have a few instances of other parameters like odom_topic, goal_reached_tol, goal_updater_topic. Just manually setting those, which are far less general than frames and tolerances, seems to beg for another solution. Either to expose these as blackboard ports instead of parameters, or another solution.

Another solution option is to potentially pass in the main node's `NodeParametersInterface` to get parameters from. That would work, though clunky and require users making their own BT nodes have to know that subtle thing.

A better solution I like is to update the client node's `NodeParametersInterface` to the main node's `NodeParametersInterface`. We can `[get_node_parameters_interface](https://docs.ros2.org/dashing/api/rclcpp/classrclcpp_1_1Node.html#ab6272f2facdfb35f0ef9ce01ffedeec9)` in the API, but I'm not sure if we can set it. There is no `set_` methods for the interfaces and the interface variables are set as private (as they should be). 

Let me inquire on some internal channels if a `set_` set of methods would be acceptable for rclcpp. If so, then I can add that in and we can set the client's node interface as a copy of the main node's parameter interface and then it should just work

SteveMacenski, 2023-09-11:
@rscova please test this PR branch. @doisyg I don't think you need to test, but just FYI in this related update. This is going to globally copy all parameters from the BT Navigator over to the client node so that any BT node can access any parameter setup in the BT Navigator configs. 

https://github.com/ros-planning/navigation2/pull/3804

SteveMacenski, 2023-09-13:
I personally tested, it works. Closing and merging.

closed 2023-09-13 by SteveMacenski