ros-navigation/navigation2 #3801: Unable to use binary filter and keepout filter at the same time

osama-z-salah, 2023-09-11:
- Operating System:
  - 
- ROS2 Version:
  - 

#### Expected behavior
- I expect all the costmap filters such as keepout mask filter, speed mask filter and binary mask filter can be work at same time.

#### Actual behavior
- When my robot enters both keepout zones and binary zones, one of filters straight away missing.

SteveMacenski, 2023-09-11:
Be precise about the nature of your problem. This report has no actionable details about what is going wrong or what you’re attempting to do and how you’ve configured it.

osama-z-salah, 2023-09-13:
I apologize for not providing sufficient details earlier. Here is a more comprehensive description:

**Required Info:**

- Operating System:
  - Ubutnu 20.04
- ROS2 Version:
  - Ubutnu 20.04

#### Steps to reproduce issue
I am attempting to run two costmap filters simultaneously in a specific location on the map. These filters are primarily the "keepout" and "binary" filters. When my robot is located outside the designated area on the map, I can still observe the presence of the "keepout" filter on the global costmap. However, as the robot approaches the area and enters the zone covered by the "binary" filter, the "keepout" filter disappears from the global costmap.
```
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.008
    alpha2: 0.040
    alpha3: 0.004
    alpha4: 0.025
    alpha5: 0.2
    base_frame_id: "base_link"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 24.0
    laser_min_range: 0.050
    laser_model_type: "likelihood_field"
    set_initial_pose: true
    initial_pose: 
      x: 0.0
      y: 0.0
      yaw: 0.0
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.10
    pf_z: 0.5
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.9
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: /scan1

bt_navigator:
  ros__parameters:
    use_sim_time: True
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /diffbot_base_controller/odom
    bt_loop_duration: 10
    default_server_timeout: 20
    # 'default_nav_through_poses_bt_xml' and 'default_nav_to_pose_bt_xml' are use defaults:
    # nav2_bt_navigator/navigate_to_pose_w_replanning_and_recovery.xml
    # nav2_bt_navigator/navigate_through_poses_w_replanning_and_recovery.xml
    # They can be set here or via a RewrittenYaml remap from a parent launch file to Nav2.
[... 727 lines cut ...]
            namespace=namespace,
            output='screen',
            emulate_tty=True,  # https://github.com/ros2/launch/issues/188
            parameters=[configured_params])

    ld = LaunchDescription()

    ld.add_action(declare_namespace_cmd)
    ld.add_action(declare_use_sim_time_cmd)
    ld.add_action(declare_autostart_cmd)
    ld.add_action(declare_params_file_cmd)
    ld.add_action(declare_mask_yaml_file_cmd)

    ld.add_action(start_lifecycle_manager_cmd)
    ld.add_action(start_map_server_cmd)
    ld.add_action(start_costmap_filter_info_server_cmd)

    return ld
```

#### Expected behavior
The ideal behavior would be for both the "keepout" and "binary" filters to work simultaneously without one affecting the other. When the robot is outside the designated area, the "keepout" filter should be active on the global costmap. As the robot enters the "binary" filter zone, the "keepout" filter should remain in place while the "binary" filter is applied independently.

#### Actual behavior
when the robot is outside the designated area on the map, the "keepout" filter is visible on the global costmap. However, as the robot enters the area covered by the "binary" filter, the "keepout" filter disappears from the global costmap.

SteveMacenski, 2023-09-13:
This perhaps is more in @AlexeyMerzlyakov 's wheelhouse. 

@osama-z-salah provide a video and logs. I reviewed the relevant code and see no reason that the keepout filter or the binary filter would in any way interact or the process of triggering one would stop another from performing its independent action.

osama-z-salah, 2023-09-25:
hi sorry i was bussy, 
[image]
[image]
The first image illustrates that I am using both the keepout filter and the speed filter in my global planner. However, when I run RViz, I can only visualize the keepout zone, and the speed zone disappears. I have noticed that these two filters cannot be used together in the global costmap.
[0.045s] Invoking command in '/home/nanoamr/ros2_ws/build/nav2_bringup': CMAKE_PREFIX_PATH=/opt/ros/humble /usr/bin/cmake --build /home/nanoamr/ros2_ws/build/nav2_bringup -- -j24 -l24
[0.248s] Invoked command in '/home/nanoamr/ros2_ws/build/nav2_bringup' returned '0': CMAKE_PREFIX_PATH=/opt/ros/humble /usr/bin/cmake --build /home/nanoamr/ros2_ws/build/nav2_bringup -- -j24 -l24
[0.271s] Invoking command in '/home/nanoamr/ros2_ws/build/nav2_bringup': CMAKE_PREFIX_PATH=/opt/ros/humble /usr/bin/cmake --install /home/nanoamr/ros2_ws/build/nav2_bringup
[0.275s] -- Install configuration: "Release"
[0.276s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/launch
[0.276s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/launch/costmap_filter_info_keepout.launch.py
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/launch/bringup_launch.py
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/launch/slam_launch.py
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/launch/localization_launch.py
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/launch/costmap_filter_info_speed.launch.py
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/launch/rviz_launch.py
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/launch/costmap_filter_info_binary.launch.py
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/launch/navigation_cal_launch.py
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/launch/navigation_launch.py
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/maps
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/maps/speed_second.pgm
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/maps/keepout_second.pgm
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/maps/binary_mask.yaml
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/maps/speed_mask.pgm
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/maps/binary_mask.pgm
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/maps/speed_second.yaml
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/maps/sixth_second.pgm
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/maps/binary_second.pgm
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/maps/keepout_second.yaml
[0.277s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/maps/sixth_second.yaml
[0.278s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/maps/binary_second.yaml
[0.278s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/maps/speed_mask.yaml
[0.278s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/rviz
[0.278s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/rviz/nav2_default_view.rviz
[0.278s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/rviz/nav2_namespaced_view.rviz
[0.278s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/worlds
[0.278s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/worlds/world_only.model
[0.278s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/worlds/waffle.model
[0.278s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/params
[0.278s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/params/slam.yaml
[0.278s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/params/calibration_params.yaml
[0.278s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/params/nav2_params (another copy).yaml
[0.279s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/params/binary_params.yaml
[0.279s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/params/keepout_params.yaml
[0.279s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/params/nav2_params (copy).yaml
[0.279s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/params/nav2_params.yaml
[0.279s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/params/speed_params.yaml
[0.279s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/urdf
[0.279s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/urdf/turtlebot3_waffle.urdf
[0.280s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/ament_index/resource_index/package_run_dependencies/nav2_bringup
[... cut ...]
[0.282s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/cmake/nav2_bringupConfig.cmake
[0.282s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/cmake/nav2_bringupConfig-version.cmake
[0.282s] -- Up-to-date: /home/nanoamr/ros2_ws/install/nav2_bringup/share/nav2_bringup/package.xml
[0.305s] Invoked command in '/home/nanoamr/ros2_ws/build/nav2_bringup' returned '0': CMAKE_PREFIX_PATH=/opt/ros/humble /usr/bin/cmake --install /home/nanoamr/ros2_ws/build/nav2_bringup

SteveMacenski, 2023-09-25:
I don't understand, I don't think that proves anything. Are you actually seeing that the speed / keepout behavior aren't working or are you just basing this on your interpretation of rviz?

dines1111, 2023-09-26:
Hi, I came across this question as well. I need the steps on how to apply binary filters. I followed this link https://navigation.ros.org/tutorials/index.html for the keepout filter and speed filter, but it did not contain information regarding binary filters. 

For example, when i use speed filter;

1. For the speed_mask.yaml, the content should be:

image: turtlebot3_world.pgm
->
image: speed_mask.pgm

mode: trinary
->
mode: scale

occupied_thresh: 0.65
free_thresh: 0.196
->
occupied_thresh: 1.0
free_thresh: 0.0

2. For the speed_mask param file:

costmap_filter_info_server:
  ros__parameters:
    use_sim_time: true
    type: 1
    filter_info_topic: "/costmap_filter_info"
    mask_topic: "/speed_filter_mask"
    base: 100.0
    multiplier: -1.0
filter_mask_server:
  ros__parameters:
    use_sim_time: true
    frame_id: "map"
    topic_name: "/speed_filter_mask"
    yaml_filename: "speed_mask.yaml"

3. For nav2_params.yaml, i need add:

global_costmap:
  global_costmap:
    ros__parameters:
      ...
      plugins: ["static_layer", "obstacle_layer", "inflation_layer"]
      filters: ["speed_filter"]
      ...
      speed_filter:
        plugin: "nav2_costmap_2d::SpeedFilter"
        enabled: True
        filter_info_topic: "/costmap_filter_info"
        speed_limit_topic: "/speed_limit"

These are the configurations I need in order to apply and run the speed filter. Returning to my question, I also require a similar configuration setup for the binary filter. Do you have any information or suggestions regarding this?

SteveMacenski, 2023-09-26:
@dines1111 those parameters are arrays, you can add multiple to them ` filters: ["speed_filter", "binary_filter", "my_filter"]` to run multiple. Read the configuration guide docs on navigation.ros.org for the specific parameters of the binary filter. We don't have a tutorial for every different costmap plugin since it all becomes largely redundant - instead just follow the steps to setup the filter and checkout the configuration guide for the specific parameters and knobs.

SteveMacenski, 2023-09-26:
@osama-z-salah your `/costmap_filter_info` conflicts for both layers. How do you even know you're getting the right information to the right layer? Remap that topic to specialized ones for each layer to make sure they cannot conflict. 

> KeepoutFilter: New costmap filter info arrived from /costmap_filter_info topic. Updating old filter info.

The logs are clearly showing this as well. 

> KeepoutFilter: For proper use of keepout filter base and multiplier in CostmapFilterInfo message should be set to their default values (0.000000 and 1.000000)

And its also clearly indicating that your parameters for the Keepout are invalid, so that's why its not being used.

closed 2023-09-26 by SteveMacenski