ros-navigation/navigation2 #3802: RQT reconfigure not working with controller_server

BriceRenaudeau, 2023-09-11:
## Bug report

We cannot see the controller_server parameters using RQT_reconfigure. 
[image]

**Required Info:**
- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Iron
- Version or commit hash:
  - main
- DDS implementation:
  - cyclone

#### Steps to reproduce issue
1- start example with navigation
2- open rqt_reconfigure
3- try to change `controller_server` params

#### Expected behavior
Parameters should be configurable like other nodes.

#### Actual behavior
The parameters doesn't appear.

#### Additional information
The parameters are visible using `ros2 param list /controller_server` CLI.

BriceRenaudeau, 2023-09-11:
Tuning the MPPI is quite complicated, without rqt_reconfigure it takes forever.

SteveMacenski, 2023-09-11:
https://github.com/ros-planning/navigation2/blob/main/nav2_controller/src/controller_server.cpp#L786

You may only change server params when not currently executing a control task for safety. You may not use dynamic parameters mid-execution. Once you’ve halted the robot, then it is permissible. 

Wrt mppi, I’m pretty sure that works, that’s how I’ve played with things in the past. I haven’t used RQT before for the task but CLI. Can you see if that works for you?

BriceRenaudeau, 2023-09-11:
Using CLI, the parameters are correctly set (even while navigating).

But it's not really easy and it works perfectly for the planner_server. I wasn't able to see why the controller_server isn't compatible with RQT.

BriceRenaudeau, 2023-09-11:
Do you use another tool for tuning MPPI?

SteveMacenski, 2023-09-11:
CLI, param files, and my notebook with metrics :-)

SteveMacenski, 2023-09-11:
Did you try refreshing? I really don't know much about RQT since I don't use it, but perhaps it just needed some time to get the params if it was launched in parallel with Nav2 due to delays with lifecycle transitioning

SteveMacenski, 2023-09-13:
This is a good question. I can see that the planner server seems to work fine but the controller server doesn't. Those servers are both generally formatted the same way, so I'm not sure why one would work and the other would not. 

RQT uses the standard get/set/list parameter services https://github.com/ros-visualization/rqt_reconfigure/blob/rolling/src/rqt_reconfigure/param_api.py. I just tested and I can use the list/get/describe parameter services in both servers with no problem. 

I also tested literally every other node in RQT in Nav2 and they all worked perfectly. Its only the  controller server. So I think there is something wrong there. The best I can tell from doing a line-by-line comparison with the planner sersver is that it could be related to the odom sub or the speed limit sub (?) since those are the only real differences (well, beyond the unique interfaces the planner server has, but _adding_ servers isn't going to be fixing the issue) beyond minor reordering of the configure function of costmap relative to algorithm loading of plugins.

SteveMacenski, 2023-09-13:
I isolated it to 

```
  try {
    nav2_util::declare_parameter_if_not_declared(
      node, "progress_checker_plugin", rclcpp::PARAMETER_STRING);
    std::string progress_checker_plugin;
    progress_checker_plugin = node->get_parameter("progress_checker_plugin").as_string();
    progress_checker_ids_.clear();
    progress_checker_ids_.push_back(progress_checker_plugin);
    RCLCPP_WARN(
      get_logger(),
      "\"progress_checker_plugin\" parameter was deprecated and will be removed soon. Use "
      "\"progress_checker_plugins\" instead to specify a list of plugins");
  } catch (const std::exception &) {
    // This is normal situation: progress_checker_plugin parameter should not being declared
  }
```

https://github.com/ros-planning/navigation2/pull/3815

It looks like RQT has a silent error that is `list`s parameters and then it tries to `get` them then they're not populated it just returns nothing. That would explain why it breaks since there is no value for `progress_checker_plugin` in a well made system

closed 2023-09-13 by SteveMacenski

BriceRenaudeau, 2023-09-14:
Did you succeed in viewing the parameters using RQT after this PR?