ros-navigation/navigation2 #3805: Nav2 Lifecycle Manager Diagnostic Publishing

Plaqueoff, 2023-09-12:
## Bug report

**Required Info:**

Operating System: Ubuntu 22.04
ROS2 Version: Humble
Version or commit hash: ead48a9039c80cadb18dcfbb21fff0bdd496ef03
Package: nav2_lifecycle_manager

#### Description of the Bug

When the lifecycle manager's pause method is called, a diagnostic status of ERROR is published with the message "Nav2 is inactive". We have a robot with a few states, where we want nav2 to be paused, and I don't see why the paused state would be an error state, so I wished to ask if this is the wanted behavior for publishing diagnostics.

#### Expected behavior

Navigation nodes could be paused through lifecycle manager without error diagnostics.

#### Actual behavior

Error diagnostic is published when navigation nodes are paused through lifecycle manager.

SteveMacenski, 2023-09-12:
Seems reasonable! This is low on the priority list for Open Navigation to work on, so if this is something you need on a reasonable timeline, I suggest you contribute a PR that we can review and merge!

Plaqueoff, 2023-09-18:
I'll try to get a PR out if not today by the end of the week!

SteveMacenski, 2023-10-03:
Imminently to merge!

closed 2023-10-03 by SteveMacenski